Edzelf/ESP32-Radio #392: Bad month for LCD2004.

hevet, 2020-10-17:
The wrong date is showing on the LCD2004, it is one month earlier. If today is 17/10, it displays 17/09.
In the LCD2004.h file, change this solution in line 475 of the code:
timeinfo.tm_mon to timeinfo.tm_mon + 1.

h1aji, 2020-10-18:
I found same issue here: https://github.com/Edzelf/ESP32-Radio/issues/382

Edzelf, 2020-10-18:
Fixed.

hevet, 2020-10-18:
> 
> 
> Fixed.

By mistake you subtracted another month instead of adding :)

Edzelf, 2020-10-18:
Yes.  Will be fixed too....

hevet, 2020-10-20:
There is one more problem. When we listen to the radio at midnight, the date and day of the week will not change. Only restarting the radio will refresh the day of the week and date. I don't know how it is on other displays.

Edzelf, 2020-10-20:
Yes, the day is only updated once in 7200 seconds by connecting to an NTP server.
It was never my intention to display the date on a display (what's the point?), but this part was added by a user who wants to use the LCD2004 and display the date.  The preferred display is a 128x160 SPI display.

hevet, 2020-10-20:
OK I understand.