Edzelf/ESP32-Radio #393: Communication error with the LCD2004 display.

hevet, 2020-10-19:
After uploading the latest version, there is a problem with the LCD2004 display.  There is an error comunication error with the display in the terminal.  The radio tries to initialize it, after an unsuccessful attempt, it restarts with the network and tries to initialize the display again.  After several attempts, it finally succeeds, sometimes it takes a few minutes, but it can also start without a working display.  Checked on two displays.  I have downloaded an older version and there is no problem, the radio starts up quickly and there is no problem with initializing the display.

Edzelf, 2020-10-19:
Please show the logging.

hevet, 2020-10-19:
`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Sun, 18 Oct 2020 10:15:00 GMT.  Free memory 287572
D: Display type is LCD2004
D: Partition nvs found, 20480 bytes
D: Read 66 keys from NVS
D: pin_ir set to 0
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to 16
D: pin_tft_dc set to 17
D: pin_tft_scl set to 22
D: pin_tft_sda set to 21
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_ch376_cs set to -1
D: pin_ch376_int set to -1
D: pin_vs_cs set to 27
D: pin_vs_dcs set to 14
D: pin_vs_dreq set to 26
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is LOW, probably no PULL-UP
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: gpio_15 will execute downpreset = 1
D: Enable pin 0 for IR
D: Start display
D: Init LCD2004, I2C pins 21,22
D: LCD2004 communication error!
D: Create list with acceptable WiFi networks
D: Added HUAWEI to list of networks
D: Added SSID2 to list of networks
[... 1056 lines cut ...]
D: Connect to stream14.radioagora.pl on port 80, extension /tuba9006-1.mp3
D: Connected to server
D: Duration mp3loop 1
D: Switch to HEADER
D: Headerline: Server: Icecast 2.4.4
D: Headerline: Connection: Close
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: Pragma: no-cache
D: Headerline: icy-br:128
D: Headerline: ice-audio-info: ice-samplerate=44100;ice-bitrate=128;ice-channels=2
D: Headerline: icy-br:128
D: Headerline: icy-genre:Rock
D: Headerline: icy-name:Rock Radio Warszawa 
D: Headerline: icy-private:0
D: Headerline: icy-pub:1
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 128, metaint is 16000
D: Duration mp3loop 46
D: Metadata block 48 bytes
D: Streamtitle found, 47 bytes
D: StreamTitle='RAINBOW / SINCE YOU'VE BEEN GONE';
`

h1aji, 2020-10-19:
@hevet comment  displaytime part in LCD2004.h. Works for me.
I think latest commit is breaking it

Edzelf, 2020-10-19:
Pin 21 and 22 are also used for I2C.  Please remove them in the config page of the web interface.

hevet, 2020-10-19:
> 
> 
> Pin 21 and 22 are also used for I2C. Please remove them in the config page of the web interface.

I don't really understand, for me the pins 21 and 22 are I2C, I can't remove them, because then there will be no display support.

Edzelf, 2020-10-19:
Yes, you are right.  LCD2004 is I2C.  I forgot.
I will try to reproduce the error.

Edzelf, 2020-10-19:
Fixed.

hevet, 2020-10-19:
Cool, looks like it's working fine.

Unfortunately, the volume level has disappeared.
In line 475 it is,

dline[3].str += "\x7F" ; // Add block character

and from what I can see it's a blank character, the dash is x2D.

Unless it is meant to be?

Edzelf, 2020-10-19:
I guess it depends on the display.  You may use any character you like.

hevet, 2020-10-19:
I did, thanks for the quick fixes.

h1aji, 2020-10-19:
Correct. It depends on i2c controller. I couldnt find block and used \x23 (hash char)

w1up, 2021-05-05:
Hello - where do the I2C pins get set?  My debug output has:
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1

and I believe it should be:
D: pin_tft_scl set to 22
D: pin_tft_sda set to 21

Edzelf, 2021-05-06:
You can set the pins in the config page of the webinterface.  Read the documentation, page 19.