Edzelf/ESP32-Radio #394: Failed to start

macwojs, 2020-10-19:
Hi
I have similar issue as below but I don't use display, only esp and VS1003. I get error in similar part:

> 19:49:06.802 -> D: Scan Networks
> 19:49:09.092 -> D: Scan completed
> 19:49:09.092 -> D: Number of available networks: 9
> 19:49:09.092 -> D:  1 - Grzegorz                  Signal: -56 dBm, Encryption WPA2_PSK, Acceptable
> 19:49:09.092 -> D:  2 - Alfred                    Signal: -58 dBm, Encryption WPA2_PSK, 
> 19:49:09.125 -> D:  3 - tvksmp_1426               Signal: -77 dBm, Encryption WPA2_PSK, 
> 19:49:09.125 -> D:  4 - tvksmp_3105               Signal: -78 dBm, Encryption WPA2_PSK, 
> 19:49:09.125 -> D:  5 - FunBox3-9D10              Signal: -82 dBm, Encryption WPA2_PSK, 
> 19:49:09.125 -> D:  6 - moto x4 5219              Signal: -83 dBm, Encryption WPA2_PSK, 
> 19:49:09.125 -> D:  7 - Orange_Swiatlowod_5452_2.4GHz Signal: -83 dBm, Encryption WPA2_PSK, 
> 19:49:09.158 -> D:  8 - pieskajtek                Signal: -87 dBm, Encryption WPA_WPA2_PSK, 
> 19:49:09.158 -> D:  9 - Xiaomi_5FF1               Signal: -92 dBm, Encryption WPA_WPA2_PSK, 
> 19:49:09.158 -> D: End of list
> 19:49:09.158 -> D: Command: clk_dst with parameter 1
> 19:49:09.158 -> D: Command: clk_offset with parameter 1
> 19:49:09.158 -> D: Command: clk_server with parameter pool.ntp.org
> 19:49:09.158 -> D: Command: gpio_00 with parameter uppreset = 1
> 19:49:09.191 -> D: Command: gpio_12 with parameter upvolume = 2
> 19:49:09.191 -> D: Command: gpio_13 with parameter downvolume = 2
> 19:49:09.191 -> D: Command: gpio_14 with parameter stop
> 19:49:09.191 -> D: Command: gpio_17 with parameter resume
> 19:49:09.191 -> D: Command: gpio_34 with parameter station = icecast.omroep.nl:80/radio1-bb-mp3
> 19:49:09.191 -> D: Command: ir_40bf with parameter upvolume = 2
> 19:49:09.191 -> D: Command: ir_c03f with parameter downvolume = 2
> 19:49:09.225 -> D: Command: mqqprefix with parameter none
> 19:49:09.225 -> D: Command: mqttbroker with parameter none
> 19:49:09.225 -> D: Command: mqttpasswd with parameter *******
> 19:49:09.225 -> D: Command: mqttport with parameter 1883
> 19:49:09.225 -> D: Command: mqttuser with parameter none
> 19:49:09.225 -> D: Command: pin_enc_clk with parameter 25
> 19:49:09.225 -> D: Command: pin_enc_dt with parameter 26
> 19:49:09.225 -> D: Command: pin_enc_sw with parameter 27
> 19:49:09.225 -> D: Command: pin_ir with parameter 35
> 19:49:09.258 -> D: Command: pin_sd_cs with parameter 21
> 19:49:09.258 -> D: Command: pin_tft_cs with parameter 15
> 19:49:09.258 -> D: Command: pin_tft_dc with parameter 2
> 19:49:09.258 -> D: Command: pin_vs_cs with parameter 5
> 19:49:09.258 -> D: Command: pin_vs_dcs with parameter 32
> 19:49:09.258 -> D: Command: pin_vs_dreq with parameter 4
> 19:49:09.258 -> D: Command: preset with parameter 6
> 19:49:09.258 -> D: Command: preset_00 with parameter 109.206.96.34:8100
> 19:49:09.258 -> D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
> 19:49:09.291 -> D: Command: preset_02 with parameter us2.internet-radio.com:8050
> 19:49:09.291 -> D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
> 19:49:09.291 -> D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
> 19:49:09.291 -> D: Command: preset_05 with parameter us1.internet-radio.com:8105
> 19:49:09.291 -> D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
> 19:49:09.324 -> D: Command: preset_07 with parameter 205.164.62.15:10032
> 19:49:09.324 -> D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
> 19:49:09.324 -> D: Command: preset_09 with parameter 94.23.66.155:8106
> 19:49:09.324 -> D: Command: preset_10 with parameter ihr/IHR_IEDM
> 19:49:09.324 -> D: Command: preset_11 with parameter ihr/IHR_TRAN
> 19:49:09.324 -> D: Command: toneha with parameter 0
> 19:49:09.324 -> D: Command: tonehf with parameter 0
> 19:49:09.357 -> D: Command: tonela with parameter 0
> 19:49:09.357 -> D: Command: tonelf with parameter 0
> 19:49:09.357 -> D: Command: volume with parameter 72
> 19:49:09.457 -> D: Slow SPI, Testing VS1053 read/write registers...
> 19:49:09.623 -> D: This is not a VS1053, but a VS1003 instead!
> 19:49:09.623 -> D: Connect to WiFi
> 19:49:12.776 -> D: Connected to Grzegorz
> 19:49:12.776 -> D: IP = 192.168.1.166
> 19:49:15.765 -> D: Start server for commands
> 19:49:15.765 -> D: Network found. Starting mqtt and OTA
> 19:49:15.765 -> D: MDNS responder started
> 19:49:16.794 -> D: Rotary encoder is enabled
> 19:49:16.794 -> D: STOP requested
[... 525 lines cut ...]
> 19:49:51.347 -> D:   9 -     0
> 19:49:51.347 -> D:   A -     0
> 19:49:51.347 -> D: SPI semaphore not taken within 350 ticks by CPU 0, id hspec
> 19:49:51.347 -> D: Semaphore is claimed by stopsong
> 19:49:51.347 -> D:   B -  F8F8
> 19:49:51.380 -> D:   C -     0
> 19:49:51.380 -> D: SPI semaphore not taken within 360 ticks by CPU 0, id hspec
> 19:49:51.380 -> D: Semaphore is claimed by stopsongD:   D -     0
> 19:49:51.380 -> 
> 19:49:51.380 -> D:   E -     0
> 19:49:51.380 -> D: SPI semaphore not taken within 370 ticks by CPU 1, id hspec
> 19:49:51.380 -> D: Semaphore is claimed by stopsong
> 19:49:51.380 -> D:   F -     0
> 19:49:51.380 -> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
> 19:49:51.380 -> Core 1 register dump:
> 19:49:51.380 -> PC      : 0x400d1f9e  PS      : 0x00060d30  A0      : 0x800d1ffe  A1      : 0x3ffd7f00  
> 19:49:51.413 -> A2      : 0x00000000  A3      : 0x000000d4  A4      : 0x00000020  A5      : 0x00000001  
> 19:49:51.413 -> A6      : 0x00060d20  A7      : 0x00000000  A8      : 0x800d1f9e  A9      : 0x3ffd7ee0  
> 19:49:51.413 -> A10     : 0x3ffd8314  A11     : 0x00000020  A12     : 0x00000020  A13     : 0x80000020  
> 19:49:51.413 -> A14     : 0x00000008  A15     : 0x00000001  SAR     : 0x00000010  EXCCAUSE: 0x0000001d  
> 19:49:51.446 -> EXCVADDR: 0x00000018  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  
> 19:49:51.446 -> 
> 19:49:51.446 -> Backtrace: 0x400d1f9e:0x3ffd7f00 0x400d1ffb:0x3ffd7f20 0x400d2021:0x3ffd7f40 0x400d2054:0x3ffd7f60 0x400d208f:0x3ffd7f80 0x400d2111:0x3ffd7fb0 0x400d230e:0x3ffd7fe0 0x400d47e1:0x3ffd8000 0x400d480e:0x3ffd8020 0x4008aba9:0x3ffd8040
> 19:49:51.479 -> 
> 19:49:51.479 -> Rebooting...

Edzelf, 2020-10-19:
If you don't a display, use the dummytft, not the 2004.  Make a choice in the main sketch for the display.

macwojs, 2020-10-19:
Yes, sorry, my bad.
Now semetimes I get "Rotation count -1" but is disapear after plug D25 to GND, but I still have another issue. 
Ip where is my ESP is Unreachable. I tried with ping and in browser. What can I do? It's possible to restart preferences to default?

macwojs, 2020-10-19:
Ok, It's working. Just like that. Sorry.

closed 2020-10-19 by macwojs