Edzelf/ESP32-Radio #396: ch376 wiring

clear-sky, 2020-10-22:
hi dear ed and thanks for your great work
i have made this radio and now would like to add ch376 to my project i studied ch376 datasheet 
https://www.mpja.com/download/ch376ds1.pdf
in page 26 cs pin is connected to vcc(3v3)
but in page 13 of your manual(https://github.com/Edzelf/ESP32-Radio/tree/master/doc/ESP32-radio.odt) dont see this connection
are you sure that pin does not need vcc connection?

clear-sky, 2020-10-22:
also see here 
https://github.com/djuseeq/Ch376msc
cs pin is connected to vcc

Edzelf, 2020-10-23:
The CH376 uses the SPI bus.  But the CH376 is not the only one on this bus.  The VS1053 and the SPI display use the bus as well.
Now if you have just ONE device on the SPI bus, you can afford to tie the CS signal to +VCC.  The device will always react on SPI activity.
But if you have more than one devices on the SPI bus, you need to select the intended device before starting a SPI transfer.  The CS (Chip Select) causes the device to be addressed.
So for one device on the bus you need 3 I/O pins, for two devices 5 I/O pins, for three devices 6 I/O pins. .......

clear-sky, 2020-10-23:
hi dear ed
thanks for the time took for me 
 it seems pin d3 of ch376 is named "SS"(slave select)
is d3 chip select pin? or another pin named CS?!
which of them must be used  as cs pin?
thanks again

Edzelf, 2020-10-23:
SS is just another name for CS.

clear-sky, 2020-10-24:
this is my experience of ch376s module

WR > GND
RD > GND
CS > 3V3 (note that datasheet says  this pin is spi cs pin page 3 pin 18 but  [djuseeq](url=https://github.com/djuseeq/Ch376msc) have used d3 as spi cs pin)
D7 > pin_spi_miso
D6 > pin_spi_mosi
D5 > pin_spi_sck
D3 > pin_ch376_cs
INT > pin_ch376_int

remember to add this line to your config line 

fs = USB

now ch376 works fine!

clear-sky, 2020-10-24:
big thanks for ed for his great work

closed 2020-10-24 by clear-sky