Edzelf/ESP32-Radio #400: No Sound from 3.5mm jack

bl-ack-bi-rd, 2020-10-28:
Hi There, 
I use the bare minimum: an ESP32 + VS1053B 
The webinterface is running and responding, connects to different radio stations.

But there is no Sound if i Plug in headphones into the 3.5mm jack.

Log:

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 25 Mar 2020 10:45:00 GMT.  Free memory 293936
D: Display type is DUMMYTFT
D: Partition nvs found, 20480 bytes
D: Read 44 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 21
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is LOW, probably no PULL-UP
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Enable pin 35 for IR
D: Start display
D: SD Card Mount Failed!
D: Create list with acceptable WiFi networks
D: Added Die_Breznsoiza to list of networks
D: Added DieBreznsoiza to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 9
D:  1 - Die_Breznsoiza            Signal: -54 dBm, Encryption WPA_WPA2_PSK, Acceptable
D:  2 - DieBreznsoiza             Signal: -59 dBm, Encryption WPA2_PSK, Acceptable
D:  3 - KabelBox-E728             Signal: -66 dBm, Encryption WPA2_PSK, 
D:  4 - Vodafone Hotspot          Signal: -88 dBm, Encryption OPEN, 
D:  5 - Vodafone Homespot         Signal: -88 dBm, Encryption OPEN, 
D:  6 - WuTangLan                 Signal: -89 dBm, Encryption WPA2_PSK, 
D:  7 - HITRON-48D0               Signal: -89 dBm, Encryption WPA2_PSK, 
D:  8 - KabelBox-70BC             Signal: -92 dBm, Encryption WPA2_PSK, 
D:  9 - Internetz                 Signal: -94 dBm, Encryption WPA_WPA2_PSK, 
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: gpio_00 with parameter uppreset = 1
D: Command: gpio_12 with parameter upvolume = 2
D: Command: gpio_13 with parameter downvolume = 2
D: Command: ir_40bf with parameter upvolume = 2
D: Command: ir_c03f with parameter downvolume = 2
D: Command: lstmods with parameter Tue, 14 Jul 2020 09:44:13 GMT
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttprefix with parameter none
D: Command: mqttuser with parameter none
D: Command: pin_enc_clk with parameter 25
D: Command: pin_enc_dt with parameter 26
D: Command: pin_enc_sw with parameter 27
D: Command: pin_ir with parameter 35
D: Command: pin_sd_cs with parameter 21
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 16
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 0
D: Command: preset_00 with parameter br-b5aktuell-live.cast.addradio.de/br/b5aktuell/live/mp3/mid
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 80
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Connect to WiFi
D: Connected to Die_Breznsoiza
D: IP = 192.168.188.41
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MDNS responder started
D: Rotary encoder is enabled
D: STOP requested
D: New preset/file requested (0/0) from br-b5aktuell-live.cast.addradio.de/br/b5aktuell/live/mp3/mid
D: Connect to new host br-b5aktuell-live.cast.addradio.de/br/b5aktuell/live/mp3/mid
D: Connect to br-b5aktuell-live.cast.addradio.de on port 80, extension /br/b5aktuell/live/mp3/mid
D: Connected to server
D: Duration mp3loop 1
D: Switch to HEADER
D: Headerline: Server: nginx
D: Headerline: Content-Length: 0
D: Headerline: Connection: close
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Headerline: Location: http://br-edge-2006-fra-lg-cdn.cast.addradio.de/br/b5aktuell/live/mp3/mid?_art=dj0yJmlwPTE4OC4xOTQuMjQwLjIzMSZpZD1pY3NjeGwtZG5x
D: Switch to DATA, bitrate is 0, metaint is 0
D: Duration mp3loop 51
D: Connect to new host br-edge-2006-fra-lg-cdn.cast.addradio.de/br/b5aktuell/live/mp3/mid?_art=dj0yJmlwPTE4OC4xOTQuMjQwLjIzMSZpZD1pY3NjeGwtZG5xZmFtbm9iJ
D: Connect to br-edge-2006-fra-lg-cdn.cast.addradio.de on port 80, extension /br/b5aktuell/live/mp3/mid?_art=dj0yJmlwPTE4OC4xOTQuMjQwLjIzMSZpZD1pY3NjeGw
D: REG   Contents
D: REG   Contents
D: ---   -----
D:   0 -   808
D:   1 -    40
D:   2 -     0
D:   3 -  6000
D:   4 -     0
D:   5 -  AC44
D:   6 -     0
D:   7 -  1E06
D:   8 -     0
D:   9 -     0
D:   A -     0
D:   B -  F8F8
D:   C -     0
D:   D -     0
D:   E -     0
D:   F -     0
D: Connected to server
D: Switch to HEADER
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-br:128
D: Headerline: ice-audio-info: bitrate=128
D: Headerline: icy-br:128
D: Headerline: icy-description:B5 aktuell
D: Headerline: icy-genre:news
D: Headerline: icy-name:B5 aktuell
D: Headerline: icy-pub:1
D: Headerline: Server: dg-br-http_fra-lg_edge_bfedadf0e8ef58af041090739a045e6c
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Headerline: Connection: Close
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 128, metaint is 16000

anyone any ideas?

Edzelf, 2020-10-28:
Pin 18 and 25 should be high under normal circumstances.

bl-ack-bi-rd, 2020-10-29:
I Pulled Pin 25 to ground since i had the issue #357. (reassining to -1 oder 26 does not work) 
there is no sound even if i turn it up.

and Pin 18 is the SCK pin for the VS1053. can i just pull it up? or is something wrong with my VS1053 ?

Edzelf, 2020-10-29:
Pin18 should normallly be HIGH.  Could be a wiring problem.  But it is strange that the VS1053 tests work correctly.

bl-ack-bi-rd, 2020-10-29:
maybe a power-supply problem? I use an 220µF capacitor (not an 3300 μF one like in your documentation)

Edzelf, 2020-10-29:
One ampere powersupply should be enough.  Can you remove the VS1053 and see if pin 18 will stay high?  Do you have a voltmeter?

bl-ack-bi-rd, 2020-10-29:
Pin 18 is 1mV to GND when the VS1053 is connected, if i pull it off (while staying connected to the power source) it is at 50mV to GND
it stays low at all times

Edzelf, 2020-10-29:
Mybe you can try a different pin.  Change pin_spi_sck accordingly in the config page of the web interface.

bl-ack-bi-rd, 2020-10-29:
ok, i solved it.
-tried Pin 22 and the result does not change.
The right move was to swap the esp32 to another one i had lying around.

Thanks for your help :)

closed 2020-10-29 by bl-ack-bi-rd