Edzelf/ESP32-Radio #402: no Wifi access after saving config

bl-ack-bi-rd, 2020-10-29:
###Description

If i save my config in the browser, it wont recognize my wifi anymore and creates an AP.
The work-around is to type my password in the config every time, then it has no problem connecting.

I tested this with two different ESP32 Boards. 
In the documentation it states, that the password is being preserved if not changed in the config. Is this a bug?

Any suggestions?

Edzelf, 2020-10-29:
The password in the config is used.

bl-ack-bi-rd, 2020-10-29:
my guess is that the "********" is being saved as a password, is there any way to check?

Edzelf, 2020-10-30:
Password is saved if it differs from "*******".
So you have to enter the real password just once and save the config.
You may add an extra dbgprint call to check it.

bl-ack-bi-rd, 2020-10-31:
how would i implement the dbgprint call?

Edzelf, 2020-10-31:
There are hundred examples in the source code.

Edzelf, 2020-10-31:
If you only have one wifi network configured, you can edit line 2127.  Fill inn ssid and password.

schraubgern, 2020-12-12:
Please give a short view of this code of this line, is it possible? Thx.