Edzelf/ESP32-Radio #403: VS1003

dagisr, 2020-10-31:
Hi all,
Have been using this VS1053/VS1003 unit in my older Esp8266 setup and it was working good.

Now in my new Esp32 setup everything seems to be working ok BUT no sound.
In my serial printout from the Esp32 it do tells me it is a VS1003 (I did not know that before)

SO what to do. Did anything change so my old VS1053 can not be used anymore?

brg
Dag in Karlskoga

dagisr, 2020-10-31:
well, well, well ..
I kind of solved it.
It's now working like in old esp8266 setup.
I removed the chip-test result in the program. I have now:

  if ( r1 !=  4 )                                       // Version 4 is a genuine VS1053
  {
    dbgprint ( "This is not a VS1053, "                 // Report the wrong chip
               "but a VS%d instead!",
               vstype[r1] ) ;
    okay = true ;
  }
  return ( okay ) ;                                     // Return the result

Anywhay - thank you very much for this fantastic ESP32 RADIO system!!!

Dag in Karlskoga, Sweden

henkiejan1, 2020-11-14:
Hmm, i have the same situation i think. Also used this vs1053 always without problemes on the esp32

Paul-OE8PCK, 2021-02-27:
Worked for me. Thank you for the tip!!!!