Edzelf/ESP32-Radio #405: https-streams

janvonnebenan, 2020-11-08:
Hello, is it possible to add support for https-streams?

For example: Deutschlandfunk
https://st01.sslstream.dlf.de/dlf/01/128/mp3/stream.mp3

Edzelf, 2020-11-08:
I have not tried it.  What is the point is sending public mp3 streams over https?

OtschiHorst, 2020-11-08:
https streams are supported.  I've been using them for a long time.  For example "Radio Paradise". The "https" is just not given.  As an example "https://stream.radioparadise.com/Aac-320".  Then just enter "stream.radioparadise.com/aac-320" in defaultsprf.h.
Gruß Otschi.
Ps:Wer hört denn heute noch DLF ;p

Edzelf, 2020-11-09:
Otschi: your example is http (port 80):
```
D: Connect to new host stream.radioparadise.com/aac-320
D: Connect to stream.radioparadise.com on port 80, extension /aac-320
D: Connected to server
D: Song stopped correctly after 0 msec
D: Switch to HEADER
D: Headerline: Server: openresty/1.17.8.2
D: Headerline: Content-Type: audio/aac
D: audio/aac seen.
D: Headerline: Connection: close
D: Headerline: icy-br: 320
D: Headerline: ice-audio-info: channels=2;samplerate=44100;bitrate=320
D: Headerline: icy-description: DJ-mixed blend of modern and classic rock, electronica, world music, and more. Always 100% commercial-free
D: Headerline: icy-genre: Eclectic
D: Headerline: icy-name: Radio Paradise (320k aac)

```

Edzelf, 2020-11-09:
Jan: This seems to work (port 80):
`dradio-edge-30af-fra-ts-cdn.cast.addradio.de/dradio/dlf/live/mp3/128/stream.mp3`

Kristian-12345, 2020-11-13:
I run into the same issue. For example all programs of Deutschlandfunk are only stramed in https (why ever).
https://www.deutschlandfunk.de/unsere-live-streams.2396.de.html?xtor=AD-251-%5B%5D-%5B%5D-%5B%5D-%5Bdlf-mobil%5D-%5B%5D-%5B%5D

To enter the address without the https does not work. Does anybody found a solution for this?

Beppi4U, 2020-11-13:
**Jup! I think i can help!! The most Web Radios are hidden or linked! To find the right Stream i found out this way! let me know if it workes for you too!
Greetings 
**Andreas**** 

**First copy your address of web radio into MS Edge browser: https://www.deutschlandfunk.de/unsere-live-streams.2396.de.html?xtor=AD-251-%5B%5D-%5B%5D-%5B%5D-%5Bdlf-mobil%5D-%5B%5D-%5B%5D
After it opened click right mouse and select "Untersuchen"**

[image]

[image]

**On the right window click on Network/Netzwerk
Then push theplay button on the Deutschlandfunk Website and the webplayer starts streaming the mp3**

[image]

**After that you can easily grab the streamurl by right click!**

[image]

**Check out Ed's WebRadio config site and generate a new preset like this!**

[image]

**Kill all unnecessary additions like in this pic shown!
Don't forget to hit save and you're good to go.........!**

[image]

Kristian-12345, 2020-11-13:
Thanks for your nice instructions,  
unfortunate I failed to do this with Firefox under Linux, but with your information I found another way;
 This side lists many stations:
https://www.radio-browser.info/#!/
It can only export playlists. If I open that playlist  in the vlc player, the context menu gives me this url:
http://st01.dlf.de/dlf/01/64/mp3/stream.mp3
Which is working.
No clue why they try so hard to hide it.

Beppi4U, 2020-11-13:
Usually on most play sites there is a commercial shown. So i think its a matter of money. But there is always  a way around every obstacle! 😉