Edzelf/ESP32-Radio #408: Problem with Config

Kristian-12345, 2020-11-12:
[Hi Edzelf,
I designed and made a PCB now.
I use a 1.8 128x160 SPI 1.8 TFT Module and the VS1003/1053 MC3 decoder.

For the connection I used those PINs you proposed in your Sketch.

>  GPIO32           -                   pin 1 XDCS            -       -
// GPIO5            -                   pin 2 XCS             -       -
// GPIO4            -                   pin 4 DREQ            -       -
// GPIO2            pin 3 D/C or A0     -                     -       -
// GPIO22           -                   -                     CS      -
// GPIO16   RXD2    -                   -                     -       TX of NEXTION (if in use)
// GPIO17   TXD2    -                   -                     -       RX of NEXTION (if in use)
// GPIO18   SCK     pin 5 CLK or SCK    pin 5 SCK             CLK     -
// GPIO19   MISO    -                   pin 7 MISO            MISO    -
// GPIO23   MOSI    pin 4 DIN or SDA    pin 6 MOSI            MOSI    -
// GPIO15           pin 2 CS            -                     -       -
// GPI03    RXD0    -                   -                     -       Reserved serial input
// GPIO1    TXD0    -                   -                     -       Reserved serial output
// GPIO34   -       -                   -                     -       Optional pull-up resistor
// GPIO35   -       -                   -                     -       Infrared receiver VS1838B
// GPIO25   -       -                   -                     -       Rotary encoder CLK
// GPIO26   -       -                   -                     -       Rotary encoder DT
// GPIO27   -       -                   -                     -       Rotary encoder SW
// -------  ------  ---------------     -------------------  ------   ----------------
// GND      -       pin 8 GND           pin 8 GND                     Power supply GND
// VCC 5 V  -       pin 7 BL            -                             Power supply
// VCC 5 V  -       pin 6 VCC           pin 9 5V                      Power supply
// EN       -       pin 1 RST           pin 3 XRST                    -

**Those Pins I put in the init file:**

>   preferences.putString ( "pin_vs_cs",   "5                                      # GPIO Pin number for VS1053 CS" ) ;
  preferences.putString ( "pin_vs_dcs",  "32                                     # GPIO Pin number for VS1053 DCS" ) ;
  preferences.putString ( "pin_vs_dreq", "4                                      # GPIO Pin number for VS1053 DREQ" ) ;
  //
  preferences.putString ( "pin_enc_clk", "25                                     # GPIO Pin number for rotary encoder CLK" ) ;
  preferences.putString ( "pin_enc_dt",  "26                                     # GPIO Pin number for rotary encoder DT" ) ;
  preferences.putString ( "pin_enc_sw",  "27                                     # GPIO Pin number for rotary encoder SW" ) ;
  //
  preferences.putString ( "pin_tft_cs",  "15                                     # GPIO Pin number for TFT CS" ) ;
  preferences.putString ( "pin_tft_dc",  "2                                      # GPIO Pin number for TFT DC" ) 
**I also put my Wifi SSD ans pwd in there
But if i upload and start the sketch, thediplay ist just white, there is no sound and on the console I have this output**

> 

> ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 19 Oct 2020 14:12:00 GMT.  Free memory 287432
D: Display type is LCD2004
D: Partition nvs found, 20480 bytes
D: Read 43 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 21
D: pin_ch376_cs set to -1
D: pin_ch376_int set to -1
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Enable pin 35 for IR
D: Start display
D: Init LCD2004, I2C pins -1,-1
D: Init LCD2004 failed!
D: SD Card Mount Failed!
D: Create list with acceptable WiFi networks
D: Added easybell DSL-CC0A05 to list of networks
D: Added SSID2 to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 6
D:  1 - easybell DSL-CC0A05       Signal: -73 dBm, Encryption WPA2_PSK, Acceptable
D:  2 - FRITZ!Box Gastzugang      Signal: -73 dBm, Encryption WPA2_PSK, 
D:  3 - FRITZ!Box 7430 TN         Signal: -88 dBm, Encryption WPA2_PSK, 
D:  4 - easybell DSL-CC0A05       Signal: -90 dBm, Encryption WPA2_PSK, Acceptable
D:  5 - EasyNachbarn5.1           Signal: -90 dBm, Encryption WPA2_PSK, 
D:  6 - FRITZ!Box Gastzugang      Signal: -92 dBm, Encryption WPA2_PSK, 
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: gpio_00 with parameter uppreset = 1
D: Command: gpio_12 with parameter upvolume = 2
D: Command: gpio_13 with parameter downvolume = 2
D: Command: ir_40bf with parameter upvolume = 2
D: Command: ir_c03f with parameter downvolume = 2
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttprefix with parameter none
D: Command: mqttuser with parameter none
D: Command: pin_enc_clk with parameter 25
D: Command: pin_enc_dt with parameter 26
D: Command: pin_enc_sw with parameter 27
D: Command: pin_ir with parameter 35
D: Command: pin_sd_cs with parameter 21
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 32
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 72
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Start CH376

**Seems I missed some config, but I can not figure out what I did wrong. Could you give me a hint?

I Uploaded a Picture of the pcb here:
https://share-your-photo.com/3aa61d44af/album

Thanks
Kristian

Edzelf, 2020-11-12:
Hello Kristian,
I think you configured the LCD2004 display.  See lines 173..180:
```
// Define (just one) type of display.  See documentation.
//#define BLUETFT                        // Works also for RED TFT 128x160
//#define OLED                         // 64x128 I2C OLED
//#define DUMMYTFT                     // Dummy display
//#define LCD1602I2C                   // LCD 1602 display with I2C backpack
#define LCD2004I2C                   // LCD 2004 display with I2C backpack
//#define ILI9341                      // ILI9341 240*320
//#define NEXTION                      // Nextion display. Uses UART 2 (pin 16 and 17)
//

```Uncomment the line with BLUETFT and comment the line with LCD2004I2c.

Kristian-12345, 2020-11-13:
Now it works,  Thank you so much :-)

closed 2020-11-13 by Kristian-12345