Edzelf/ESP32-Radio #409: Volume control is strange

ThomasSchattat, 2020-11-15:
Hi there,
the volume control in my radio is somewhat strange. Between 0 and 50 it is quiet, I begin hearing something from 50 until 100 full sound.
How can I change the behaviour so that 0 is the level that I have at 50 now? The way it is now the volume control is useless belw 50. In other words: I want to leave 0-49 away.
Thanks and regards Thomas

clear-sky, 2020-11-15:
i also have this problem
try this 
https://github.com/schreibfaul1/ESP32-vs1053_ext

B4stl3r, 2020-11-15:
have you checked the code from schreibfaul1 and how the Volume is written to the register??
https://github.com/schreibfaul1/ESP32-vs1053_ext/blob/master/src/vs1053_ext.h
=> const char volumetable

    if(vol > 21) vol=21;
    vol=volumetable[vol];
    if(vol != curvol){
        curvol=vol;                                      // Save for later use
        value=map(vol, 0, 100, 0xF8, 0x00);              // 0..100% to one channel
        value=(value << 8) | value;
        write_register(SCI_VOL, value);                  // Volume left and right
    }

especially the map function.

Edzelf, 2020-11-15:
Change line 989, the value 0xF8 to about 0x7C.

ThomasSchattat, 2020-11-15:
I changed the value as suggested by Ed, but this only helped slighty. When the value is too small I don't reach silence, too big and the old effect is back. The best approach was for me to use the lookup table, however what schreibfaul suggested would not work either. Any value above 20 would be full volume as the function expects values 0-21, Ed's software however uses 0-100. So I changed that a bit to get the best of both: Before I go to the lookup table I divide the value by 5 and reduced one element from the list. So my values 0-100 now take values 0-20 from the LUT and it works nicely with 0 silent and 100 full volume. Resolved for me.

closed 2020-11-15 by ThomasSchattat

Lukor-1975, 2020-11-23:
> I changed the value as suggested by Ed, but this only helped slighty. When the value is too small I don't reach silence, too big and the old effect is back. The best approach was for me to use the [...]

Hi Thomas,

could you please explain what you did for a Noob :-)

Thank You