Edzelf/ESP32-Radio #410: Bluetooth audio

LZ1PPL, 2020-11-16:
I know that this is not a issue but more like request. I was wondering is it possible to implement also bluetooth stream frome phones. So we can use the progect as Bluetooth speaker as well.

Kristian-12345, 2020-12-08:
would like that as well

macikm, 2021-01-03:
same as my "wish" :-)

https://github.com/Edzelf/ESP32-Radio/issues/287

so I vote up ...

Minfred, 2021-02-13:
Unfortunately, the bt-mode of my speaker works just when i disconnect the aux-cable... so i would also like the idea of a bluetooth-streamingmode from the esp32-radio itself. (+1)

Heiner1967, 2021-02-24:
I would like this as well, and closed my parallel ask to keep it all in one

Heiner1967, 2021-02-24:
Hi,
it seems someone did already used the ESP32 as Bluetooth Receiver of an audio file. However I am not clever enough to get the right piecs dotted together and would appreciate if an expert could think it through.
https://www.youtube.com/watch?v=4a10Tl5eR30

xP373Rx, 2021-05-05:
im using this  and switch it on via relay

https://www.amazon.com/Bluetooth-Receiver-Wireless-Electronics-Speaker/dp/B085TQQ9KC