Edzelf/ESP32-Radio #412: Some Radio Stations don't work as Preset

Lukor-1975, 2020-11-24:
Hi,

I have added several Radiostations as Presets and first tried if the URL is working by copy it in the "PLAY" field in the Web Interface.
When it worked I added it as a preset.

Unfortunately the Presets don't work, if I select one the radio tries to connect to the URL but fails:

D: New preset/file requested (16/0) from http://s5-webradio.antenne.de/black-beatz
D: Connect to new host http://s5-webradio.antenne.de/black-beatz
D: Connect to http on port 0, extension //s5-webradio.antenne.de/black-beatz
D: Request http://s5-webradio.antenne.de/black-beatz failed!

When I add the Same Line in the Play field it works:

D: Get command is: station=http://s5-webradio.antenne.de/black-beatz
D: Send reply for station=http://s5-webradio.antenne.de/black-beatz
D: Command: station with parameter s5-webradio.antenne.de/black-beatz
D: Connect to new host s5-webradio.antenne.de/black-beatz
D: Connect to s5-webradio.antenne.de on port 80, extension /black-beatz
D: Connected to server
D: Switch to HEADER
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Cache-Control: no-cache
D: Headerline: Connection: close
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: Pragma: no-cache
D: Headerline: Server: streaMonkey streaming Server Native
D: Headerline: Sessionid: 5fbd14aee000e328c4fa8bb4
D: Headerline: Set-Cookie: streamingsession=5fbd14aee000e328c4fa8bb4; Path=/; Domain=antenne.de; Max-Age=9999999999
D: Headerline: icy-audio-info: ice-channels=2;ice-samplerate=44100;ice-bitrate=128
D: Headerline: icy-br: 128
D: Headerline: icy-description: ANTENNE BAYERN Black Beatz
D: Headerline: icy-genre: Pop
D: Headerline: icy-metaint: 16000
D: Headerline: icy-name: ANTENNE BAYERN Black Beatz
D: Headerline: icy-pub: 1
D: Headerline: icy-url: http://www.antenne.de
D: Switch to DATA, bitrate is 128, metaint is 16000
D: Duration mp3loop 73
D: Metadata block 48 bytes
D: Streamtitle found, 41 bytes
D: StreamTitle='Alicia Keys - Girl On Fire';

What am I doing wrong??

Thanks

Edzelf, 2020-11-24:
Remove the "http://" and try again.

Lukor-1975, 2020-11-24:
Wow, quick response and quick solution.
You made my Day, Thanks Ed

closed 2020-11-24 by Lukor-1975