Edzelf/ESP32-Radio #413: play music via ir remote

clear-sky, 2020-12-01:
hi dear ed!
i am trying to play music via ir remote but couldnt find any command for it
please help

Beppi4U, 2020-12-01:
Hi there!
First apply IR sensor to your system. Check tutorial for pinout. Not every IR sensor works.

[image]

Open Serial Monitor of choice and fire some Buttons on IR remote contoller!
As you can see the IR codes are written to serial! 

[image]

Usually you copy and paste it in the config webpage! Check manual for command selection! 
ir_48B7 = downpreset=1 # BACK            --> ir_48B7  -- was signalcode from IR copied from serialCOM
                                                                 --> downpreset=1  -- Keyword for the programm
ir_6897 = uppreset=1 # FORWARD
ir_8877 = upvolume = 2 # UP VOLUME
ir_D827 = downvolume = 2 # DOWN VOLUME

[image]

This worked for me. 
Hope i could help you!
If.....let me know! 
Greetings   :-)
Andreas

clear-sky, 2020-12-02:
hi thanks for reply
I knew that before
i have some mp3 music in usb flash that connected to esp32 via ch376
i can play music via web interface but could not find any command  to playing those files via ir remote

Beppi4U, 2020-12-02:
Hmmmh...Should be the same proceed. I gonna check that! Unfortunately i didn't have  USB connection to the ESP32 but maybe i can figure it out.

Edzelf, 2020-12-02:
Try:
`ir_xxxx = mp3track = 0`

clear-sky, 2020-12-04:
hi dear ed and thanks for reply
this command works but everytime i press this bottom it plays different music and drops in a loop playback
is there a command to play a specific file for example play music 1 in folder 5 
another question 
is there any commands for standard players like play ,next ,prev ,pause, stop , etc
thanks again for your notice

Edzelf, 2020-12-04:
From the document (mostly):
mptrack = 0   means play random track.
mptrack = 5,1 means play track one in folder 5
uppreset = 1   means play next track.
downpreset = 1 means play previous track.
stop means stop
resume means resume

clear-sky, 2020-12-04:
thanks dear ed!
where can i find this document?
would like to find full ir remote commands.

Edzelf, 2020-12-04:
Document is in the doc directory of this repositry.

clear-sky, 2020-12-04:
thanks dear ed!
i had read that doc but mp3 section is Ambiguous!
web ui make it hard to me to understand your code
for example when i press play mp3  in web ui could not find which command is excuted!
at the end big thanks for your notice and your reply

closed 2020-12-04 by clear-sky

tonypascal, 2023-05-25:
Hello @Beppi4U 
How did you get an "equalizer" / "analyser" on the screen? is there another version?
Best