Edzelf/ESP32-Radio #415: ESP32Radio

andrus-65, 2020-12-08:
Hello! I encountered such a problem when making your WEB-radio project on ESP 32. For some reason, the volume is constantly decreasing. Regardless of whether I press the button or twist the encoder. What could be the reason? I used the following modules: ESP32 DEVKITV1, LCD I2C 2004, vs1053 module.
I tried connecting other pins, but the problem remained.

Edzelf, 2020-12-08:
Show your logging.

andrus-65, 2020-12-08:
> Show your logging.

17:56:58.750 -> D: 89 tracks on SD
17:56:58.784 -> D: Create list with acceptable WiFi networks
17:56:58.784 -> D: Added Keenetic-3339 to list of networks
17:56:58.784 -> D: Added SSID2 to list of networks
17:56:58.784 -> D: End adding networks
17:56:58.953 -> D: Scan Networks
17:57:01.032 -> ets Jun  8 2016 00:22:57
17:57:01.032 -> 
17:57:01.032 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
17:57:01.032 -> configsip: 0, SPIWP:0xee
17:57:01.032 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
17:57:01.065 -> mode:DIO, clock div:1
17:57:01.065 -> load:0x3fff0018,len:4
17:57:01.065 -> load:0x3fff001c,len:1044
17:57:01.065 -> load:0x40078000,len:8896
17:57:01.065 -> load:0x40080400,len:5816
17:57:01.065 -> entry 0x400806ac
17:57:01.404 -> 
17:57:01.404 -> D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 16 May 2019 12:10:00 GMT.  Free memory 282948
17:57:01.437 -> D: Display type is LCD1602
17:57:01.437 -> D: Partition nvs found, 20480 bytes
17:57:01.437 -> D: Read 52 keys from NVS
17:57:01.437 -> D: pin_ir set to 35
17:57:01.437 -> D: pin_enc_clk set to 25
17:57:01.437 -> D: pin_enc_dt set to 26
17:57:01.437 -> D: pin_enc_sw set to 27
17:57:01.437 -> D: pin_tft_cs set to -1
17:57:01.437 -> D: pin_tft_dc set to -1
17:57:01.437 -> D: pin_tft_scl set to 33
17:57:01.437 -> D: pin_tft_sda set to 32
17:57:01.437 -> D: pin_tft_bl set to -1
17:57:01.437 -> D: pin_tft_blx set to -1
17:57:01.437 -> D: pin_sd_cs set to 21
17:57:01.472 -> D: pin_vs_cs set to 5
17:57:01.472 -> D: pin_vs_dcs set to 16
17:57:01.472 -> D: pin_vs_dreq set to 4
17:57:01.472 -> D: pin_shutdown set to -1
17:57:01.472 -> D: pin_shutdownx set to -1
17:57:01.472 -> D: pin_spi_sck set to 18
17:57:01.472 -> D: pin_spi_miso set to 19
17:57:01.472 -> D: pin_spi_mosi set to 23
17:57:01.472 -> D: GPIO0 is HIGH
17:57:01.472 -> D: GPIO2 is LOW, probably no PULL-UP
17:57:01.504 -> D: GPIO4 is HIGH
17:57:01.504 -> D: GPIO5 is HIGH
17:57:01.504 -> D: GPIO12 is HIGH
17:57:01.538 -> D: GPIO13 is HIGH
17:57:01.538 -> D: GPIO14 is HIGH
17:57:01.538 -> D: GPIO15 is HIGH
17:57:01.538 -> D: GPIO16 is HIGH
17:57:01.571 -> D: GPIO17 is HIGH
17:57:01.571 -> D: GPIO18 is HIGH
17:57:01.571 -> D: GPIO19 is HIGH
17:57:01.604 -> D: GPIO21 is HIGH
17:57:01.604 -> D: GPIO22 is HIGH
17:57:01.604 -> D: GPIO23 is HIGH
17:57:01.638 -> D: GPIO25 is HIGH
17:57:01.638 -> D: GPIO26 is HIGH
17:57:01.638 -> D: GPIO27 is HIGH
17:57:01.638 -> D: GPIO32 is HIGH
17:57:01.672 -> D: GPIO33 is HIGH
17:57:01.672 -> D: GPIO34 is LOW, probably no PULL-UP
17:57:01.672 -> D: GPIO35 is HIGH
17:57:01.705 -> D: GPIO39 is LOW, probably no PULL-UP
17:57:01.705 -> D: gpio_00 will execute uppreset = 1
17:57:01.705 -> D: gpio_12 will execute upvolume = 2
17:57:01.705 -> D: gpio_13 will execute downvolume = 2
17:57:01.705 -> D: Enable pin 35 for IR
[... 128 lines cut ...]
17:57:43.311 -> D: Rotation count -1
17:57:44.962 -> D: Rotation count -1
17:57:45.471 -> D: Rotation count -1
17:57:46.450 -> D: Rotation count -1
17:57:46.954 -> D: Rotation count -1
17:57:47.424 -> D: Rotation count -1
17:57:49.439 -> D: Rotation count -1
17:57:50.720 -> D: Rotation count -1
17:57:51.090 -> D: Rotation count -1
17:57:52.947 -> D: Rotation count -1
17:57:54.060 -> D: Rotation count -1
17:57:54.839 -> D: Rotation count -1
17:57:55.781 -> D: Rotation count -1
17:57:56.188 -> D: Rotation count -1
17:57:56.661 -> D: Rotation count -1
17:57:56.864 -> D: Rotation count -1
17:57:58.516 -> D: Rotation count -1
17:57:59.995 -> D: Rotation count -1
17:58:00.501 -> D: Rotation count -1
17:58:02.665 -> D: Rotation count -1
17:58:03.038 -> D: Rotation count -1
17:58:04.627 -> D: Rotation count -1
17:58:05.438 -> D: Rotation count -1
17:58:06.725 -> D: Rotation count -1
17:58:08.009 -> D: Rotation count -1

Edzelf, 2020-12-08:
Probably you have no rotary decoder.  Remove lines in your config that configures the rotary encoder: pin_enc_clk, pin_enc_dt and pin_enc_sw.

andrus-65, 2020-12-08:
> Probably you have no rotary decoder. Remove lines in your config that configures the rotary encoder: pin_enc_clk, pin_enc_dt and pin_enc_sw.

The encoder is standing. And it works. When I turn its rotor to +, it turns up the volume. I assumed that the encoder was faulty, so I put another one on. The result is the same.

andrus-65, 2020-12-09:
Dear Ed! Possible is turn on MP3 player with the help Ir remote controle? if so, how?

Edzelf, 2020-12-09:
See issue 413.

andrus-65, 2020-12-09:
> See issue 413.

Thanks!

fenyvesi, 2021-02-14:
Program VERSION     "Mon, 19 Oct 2020 14:12:00 GMT"

I also get continuous (3/sec) encoder switch interrupt with this setup. The encoder pins are pulled up with resistors. 
pin_enc_clk = 39                                     # GPIO Pin number for rotary encoder "CLK"
pin_enc_dt = 26                                      # GPIO Pin number for rotary encoder "DT"
pin_enc_sw = 25                                      # GPIO Pin number for rotary encoder "SW"
I can't relocate PINs, because the bulit-in OLED version has no more appropriate pin.
This version also identfies the VS chip as not VS1053, but perhaps 1001 (I can't remeber exactly). I think it is 1053.

I went back to an older version (VERSION "Fri, 18 Aug 2017 14:15:00 GMT") and it works wihout any problem.

Thanks.