Edzelf/ESP32-Radio #420: ESP32Radio takes an assigned IP Address

schraubgern, 2020-12-12:
When I start the ESP32 Radio it starts first with AP mode. After a reboot it takes an IP Address which is assigned to an other device in my case a Shelly Switch. Then it plays radio stations until I reboot it. My Network is **D: Connected to OsterHase**
**D: IP = 192.168.178.44** . You can see this in my Debug: 1. AP mode with the changed AP IP Address, 2. after Reboot starting with the IP Address from my Shelly Device.

****************************************************************************************************************
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:9564
ho 0 tail 12 room 4
load:0x40080400,len:6320
entry 0x400806a8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 19 Oct 2020 14:12:00 GMT.  Free memory 294360
D: Display type is NEXTION
D: Partition nvs found, 20480 bytes
D: Read 43 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to 25
D: pin_enc_dt set to 26
D: pin_enc_sw set to 27
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to 21
D: pin_ch376_cs set to -1
D: pin_ch376_int set to -1
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Enable pin 35 for IR
D: Start display
[... 197 lines cut ...]
D: Sync TOD, new value is 14:35:56
D: STOP requested
D: New preset/file requested (1/0) from airspectrum.cdnstream1.com:8114/1648_128
D: Connect to new host airspectrum.cdnstream1.com:8114/1648_128
D: Connect to airspectrum.cdnstream1.com on port 8114, extension /1648_128
D: Song stopped correctly after 0 msec
D: Nextion t0.txt="ESP32-Radio"
D: Nextion t1.txt=""
D: Connected to server
D: Duration mp3loop 1
D: Switch to HEADER
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-br:128
D: Headerline: ice-audio-info: ice-samplerate=44100;ice-bitrate=128;ice-channels=2
D: Headerline: icy-br:128
D: Headerline: icy-description:Today's Easy Hits from the 70s, 80s, 90s through Today!
D: Headerline: icy-genre:Easy Listening
D: Headerline: icy-name:Easy Hits Florida
D: Headerline: icy-private:0
D: Headerline: icy-pub:1
D: Headerline: icy-url:http://www.easyhitsflorida.com
D: Headerline: Server: Icecast 2.4.0-kh4
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Access-Control-Aorida"

schraubgern, 2020-12-12:
[image]

macikm, 2021-01-03:
Shelly Switch and ESP gets IPs from DHCP?
Is IP reserved for Shelly Switch or for ESP?

schraubgern, 2021-02-01:
Hello @macikm , my configuration is as follow: ESP gets an IP via DHCP, its alway the same because by daily use. All my Shelly's have assigned fix IP Adresses. In my Fritz Box you see the same addresses and my Shelly is working since one Year. Since four month later the ESP Radio was ready for use in my network and gets the same IP? That is very strange.

xP373Rx, 2021-02-05:
Could you have won in the lottery and the mac address of both is the same??

BePeMee, 2021-02-09:
So your ESP get an IP from DHCP and the shelly's have a fixed IP. Then I think you should tell your DHCP server not to assign addresses in the range where you have assigned fixed IP's to your shelly's.

closed 2021-11-08 by schraubgern