Edzelf/ESP32-Radio #421: compiler error micogear

donnersm, 2020-12-13:
When Compiling, this is my error:
WARNING: library ESP8266_Microgear claims to run on esp8266 architecture(s) and may be incompatible with your current board which runs on esp32 architecture(s).
C:\Users\chord\Documents\Arduino\libraries\ESP8266_Microgear\MicroGear.cpp: In member function 'void MicroGear::syncTime(Client*, long unsigned int*)':
C:\Users\chord\Documents\Arduino\libraries\ESP8266_Microgear\MicroGear.cpp:141:9: error: 'WiFiClientSecure' was not declared in this scope
         WiFiClientSecure *clientsecure = (WiFiClientSecure *)(client);
         ^
C:\Users\chord\Documents\Arduino\libraries\ESP8266_Microgear\MicroGear.cpp:141:27: error: 'clientsecure' was not declared in this scope
         WiFiClientSecure *clientsecure = (WiFiClientSecure *)(client);
                           ^
C:\Users\chord\Documents\Arduino\libraries\ESP8266_Microgear\MicroGear.cpp:141:61: error: expected primary-expression before ')' token
         WiFiClientSecure *clientsecure = (WiFiClientSecure *)(client);

Edzelf, 2020-12-14:
The sketch is for an ESP32 processor.

donnersm, 2020-12-14:
I know, I selected the ESP 32 Devkit and also tried several other ESP32 processors but somehow the sketch wont compile because it indirectly links to this ESP8266 Microgear lib.
I will try a new slean  install of arduino on a different computer, see if that works

donnersm, 2020-12-15:
Turns out ( and this will help others who strugle with this) I had Boardpackage ESP32 and ESP8266 both installed and the compiler randomly chooses what it wants...thats why it was refering to the ESP8266 lib.. AFter a clean install with only board package ESP8266 and the missing lib (Ch376msc-1.4.4 and pubsubclient 2.6) all compiled fine! 
One problem left after compiling: the vs1053 didn't work this was because in the config file one of the lines refers to a different pin than what is mentioned in the ino file comment top section.....So a clean install, lib install and correct wiring/config file...made it all work perfectly!!

closed 2020-12-15 by donnersm