Edzelf/ESP32-Radio #423: White Noise

B4stl3r, 2020-12-25:
Hi,

I used the schematics from https://github.com/Edzelf/ESP32-Radio/issues/66 and also used the transistors.

I still have some "white noise" left, which might come from the ESP itself. https://electronics.stackexchange.com/questions/368867/esp32-microcontroller-generates-noise-on-microphone
You can hear it very clearly until ~30% of max volume..

I tried serveral things, from using 2 different power supplys, also both of the  VS1053 Lineout solution (which still has then the white noise issue). https://www.sparkfun.com/datasheets/Widgets/MP3Trigger/vs10XXan_output.pdf

Do you have the same isse or have any filtering solution inplace ?

Thanks in advance!

Edzelf, 2020-12-25:
It is unlikely that the noise comes from the ESP32.  The problem must be in the VS0153 or in the analog circuit.  I never noticed a significant noise.  Did you try a headphone on the VS105e output?

utw3v0, 2020-12-25:
You could try an audio transfomer like the Monacor LTR110, that should help.

Greets

⁣TypeApp for Android herunterladen ​

B4stl3r, 2020-12-25:
yep, had the headset before.. 
unfortunately cannot remember if the issue already existed there..

removed the 3.5 connection and using pins instead right now.

B4stl3r, 2020-12-25:
@utw3v0 i've already transformers in place.

here's what it currently looks like:
[image]

utw3v0, 2020-12-25:
maybe a grounding problem .... theres a digital and a analog ground can you try to seperate them ? And maybe the esp is too close to the vs1053 so it might generate some em interterence ....

⁣TypeApp for Android herunterladen ​

B4stl3r, 2020-12-25:
hi @utw3v0 , already read alot about it, but was not very clear, how i shoud seperate it.

also from the pcb layout of eXadra, i do not see, where the grounds are seperated.

also variated the distance between VS, ESP and SD reader with unfortunately no difference..

would love to give it a try, but don't really have a cloud, what i should do.. Maybe you could give me some further advise?

thanks!!

utw3v0, 2020-12-25:
I'll give you my schematic, it works like a charm, theres an sd - card connection a well a ch376 wired but that's optional ...

[image]

Greets Werner

utw3v0, 2020-12-25:
i've just seen on your picture that the vs1053 the esp and your am are hooked up to the same power source, try to separate them, at least your amp (8610) and the rest ...

B4stl3r, 2020-12-25:
Hi Werner,

thx for providing your schematic, unfortunately it's still the same background noise :(

[image]

I also tried it with a second powersource for the amp (and also amp + vs), but issue keeps the same.. 

maybe my VS1053 is broken ?

utw3v0, 2020-12-25:
I dont think that your vs is broken, i kicked out all of my breadboards, had always trouble with the cabling so i chose to solder everything, maybe you mic is activated and its amp is generating the noise ? Maybe your pam is making the noise it self ? Try a headphone without anything else except the esp ofc. Remove all unnecessary cables / modules .... my board pulls with ch chip and sd card up to 460 mA so your powersupply should deliver that on the 5 volts rail ... i use an powersupply for laboratory so its generating clean 5 volts with very low ripple try an pc speaker system to see if there is the noice too

utw3v0, 2020-12-25:
Maybe you can make a video of the problem ?

utw3v0, 2020-12-25:
Ive just seen your 5v and 12v gnd is connected via bridge, you should remove that bridge so the both grounds are not connected to each other

B4stl3r, 2020-12-25:
Hi Werner,

thats what also my plan is, already preparing my perforated board, but with that white noise, i dont want to solder it..

just tried to record a video, just have an old iphone 5, sry ..
https://youtu.be/cPeA3fRwi4I

The bridge does not matter if its going into the gnd of the 5V DC DC converter or directly..

Pam alone is absolutely silent.. just when i hook up everything together, this noise appears..

Maybe its still with the differentiated outputs of the VS1053?

found this..
[image]

Edzelf, 2020-12-25:
Meassure the pgnd pin, that is the common of the 2 analog outputs, with a voltmeter.  It should be half the power voltage.  If it is is 0 Volt, you are in trouble.

B4stl3r, 2020-12-25:
measure between what exactly ?

https://www.mouser.com/datasheet/2/115/PAM8610-247304.pdf

Edzelf, 2020-12-25:
Between ground of your power supply and the pgnd pin of the vs1053.

Edzelf, 2020-12-25:
Meassure with pam connected and removed.

B4stl3r, 2020-12-25:
assume between GND and VS1053 GBuf (audiojack out) ? 
it's always around 1,4xx V, regardless if PAM connected or not..

Edzelf, 2020-12-25:
That's good.  So it looks like it is not a wiring problem.  Next thing to do is to replace the vs1053.

B4stl3r, 2020-12-25:
dont have another one around :(

Rainer-G, 2020-12-26:
Did you try with a larger capacitor ( 220µ or more ) on each power rail? I could not see any decoupling capacitor on your breadbord.
If I put digital and analog circuits on such a breadboard, I found it a good practice to add capacitors for decouplig.
Good luck
Rainer

B4stl3r, 2020-12-26:
Morning Rainer,

assume like this?

[image]

unfortunately no change.

utw3v0, 2020-12-26:
Unfortunately in your video theres no noise su i guess it's the same problem like in this video https://www.youtube.com/watch?v=4bm22sN5hWE Im going in to a preamp and then an poweramp 100Watts and its working fine i also had it hooked up directly to the poweramp no problems at all ....

B4stl3r, 2020-12-26:
Hi Werner,

you can hear the noise starting with 0:05.. its not a "loud" noise or cracking noise, its just rustling..

without the transformators its the same thing as in your posted vid but i'm using already audio transformators and still have some rustling left..

utw3v0, 2020-12-26:
Maybe its your vs then but i can't imagine the chip is very solid and stable i soldered one and made my self mp3 decoder as you can see in my schematics without any disturbing noise ... its a weird thing .... can you post your schematics ? What if the esp and the vs are all alone ? Still that noise ? Maybe shorter wires i could pickup an interference from another source around your setup ....

Rainer-G, 2020-12-26:
On your last picture, I can see that the ESP32 Board is not connected via USB to the PC, is that the case in all other testing scenarios? 
I'm asking because I had problems with setup ( Adafruit HUZZA32 and Adafruit VS1053, no tranformers, connected to a Tripath amplifier ). When conneted to the PC, I had some noise when volume was zero, it disappeared when I unplugged the USB cable!
I have one VS1053 board left from my first tries before I switched to the Adafruit board. 
I you want, I can give it to you free of charge if you let me know your address. 
If there are no private messages possible an ED agrees, you can send him your E-Mail address and he can send it to me or pass my address to you. 
Cheers
Rainer

B4stl3r, 2020-12-26:
Hi Werner,

i also cannot assume that the vs is broken right now.. currently i'm switching my schematics all the time to find the issue..
i could really think that the esp32 is creating so much noise that it will interfere with the VS and therefore i'm getting noise in my audio. dont have a oszi around to measure that..

Hi Rainer,
yes, i never connect the PC to the ESP.. code is already running, therefore no need to connect it to the pc. 
thanks for your offer, will come back to you if there's no other choice!

another option i think about is to lift the VREF pin from the PAM8610 and solder the GBUF to it as mentioned in a post above..

really wired.. trying around for several weeks now, really getting annoyed why i always have bad luck and issues that no one else is facing..

B4stl3r, 2020-12-26:
maybe another idea..

i regulate the volume on the VS1053, maybe this can be the root cause?

as you also might have a poti onboard, does this poti regulate the amplifier input? or do you connect it to the ESP and regulate the volume also on the VS1053?

utw3v0, 2020-12-26:
I can regulate from both, vs and preamp doesn't make a difference .....

B4stl3r, 2020-12-26:
when you set the VS to <10% max volume and the amp to 100%, do you have then noise ?

utw3v0, 2020-12-26:
Because of the ltr110 coils i can hear a sound when the volume goes over 50% everything else is too weak but theres absolutely nothing disturbing

Rainer-G, 2020-12-28:
Given up or found the problem?
If still a problem, I'd suggest to start from scratch: remove all wires, put the ESP32 on a breadboard and only connect the VS1053 board using the 5 Volt from the ESP. Use a powerbank or USB charger to supply 5 Volt and connect an earphone. Take absolute care that GBUF never ever is connected to GND! This will ruin the VS1053!
Use the WEB interface to control the radio, check if a station is playing and then set the volume to zero. You should hear nothing, otw something is wrong.  If silence is golden, connect one board after the other, repeat the noise check and continue, if silent. Finally, connect the PAM using the transformers.
I could not keep my fingers quite so I did the first step with an ESP32 and the VS1053. As I heard nothing, I connected the VS1053 to my Tripath amplifier, volume zero on VS1053 and full on amplifier, silence!
Not really trusting my ears, I cleaned up my workspace and connected the VS1053 to a scope and, surprise, I _saw_ noise I did not hear before. Not a surprise, if was around **270 kHz**, to higher than a human can hear!
[image]
So, both the ESP and the VS work as expected, might be due to the proper cabeling, see utw3v0 remarks above!

Have a try and please let us know the results
Rainer

B4stl3r, 2020-12-28:
Hi Rainer,

still struggling ;) waiting until i get a oszi from a friend of mine.

will you update as soon as I have news ;)

Meanwhile, i added a mail to this acc, you can send me a mail ;)

Thanks for you support till now !

xP373Rx, 2020-12-29:
You do not have to have a 300€ oszi for those kind of things any cheap ass oszi will be enough. i suggest you buy one it saves a lot of trouble especially when you troubleshooting  noise..

B4stl3r, 2021-01-06:
Hi Guys,

so I have at least **some** news from todays work..

as you can see, I soldered everything on a perforated board.. everything is working (as before)...

[image]

Additionally, I checked my VS with headphones..
There's no noise at all when using headphones..

Noise is just comming, when I use the Transformers or the recommended circuit from the VS guys (http://www.vsdsp-forum.com/phpbb/viewtopic.php?f=9&t=69)

Maybe the transformers are just shitty ? 10pc's for about 9$..?

xP373Rx, 2021-01-07:
That sounds like a noisy buckconverter  or  some dataline  spi/uart  is injecting noise to the power supply line. 
A real solution is to clean the the supply voltage instead of trying to clean the output.
The noise in the supply voltage will be amplified in the amp.

Using the same module. from there  directly to a hifi amp. no noise.
But i do not have a screen connected .

clear-sky, 2021-01-13:
hi all
>>Maybe the transformers are just shitty ?
i dont have this transformer
but decided to know it
as far as i know this tranformormer is 1:1 with 600ohm:600ohm impedance
i found this in an old shop
sansui st-71
http://www.op316.com/tubes/datalib/sansui-st.pdf
[image]
[image]
resistanse between pins is about 50ohm
i ask from true working radio owners with transformer and without white noise :
is this parameters true?

B4stl3r, 2021-02-05:
Hi all,

would like to update you on this topic.. tried several things and also @Rainer-G is supporting me great.

I stepped a litte bit back to a minimal example (also with a different VS, just to be sure):
[image]

What can I say.. I can hear the white / purple noise **also / already** in the earphones when volume is below 40..

Tried it with several Powersources (USB from PC, 230V=>USB PS, Powerbank), the Noise is always present..

In case if anybody wants to try it out, here's my config:
D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 19 Oct 2020 14:12:00 GMT.  Free memory 294048
D: Display type is DUMMYTFT
D: Partition nvs found, 20480 bytes
D: Read 15 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_ch376_cs set to -1
D: pin_ch376_int set to -1

**D: pin_vs_cs set to 15
D: pin_vs_dcs set to 26
D: pin_vs_dreq set to 2**

D: pin_shutdown set to -1
D: pin_shutdownx set to -1

**D: pin_spi_sck set to 14
D: pin_spi_miso set to 12
D: pin_spi_mosi set to 27**

Will keep you updated if there are any news to tell.

xP373Rx, 2021-02-05:
Tr4y to power the esp through the 5v pin and also the vs1053 .. do not get it from usb

xP373Rx, 2021-02-05:
i use this shield

https://www.amazon.de/gp/product/B08GYFVR5X/ref=ppx_yo_dt_b_asin_image_o03_s00?ie=UTF8&psc=1

and this board

https://www.amazon.de/gp/product/B08BTWK3VV/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1

works like a charm.. has also sd  config is:

clk_dst = 1
clk_offset = 1
clk_server = pool.ntp.org
#
ir_40BF = upvolume = 2
ir_C03F = downvolume = 2
#
mqttbroker = none
mqttpasswd = *******
mqttport = 1883
mqttprefix = none
mqttuser = none
#
pin_ir = -1                                     # GPIO Pin number for IR receiver VS1838B
pin_sd_cs = 13                                     # GPIO Pin number for SD card CS
pin_tft_cs = 16                                     # GPIO Pin number for TFT CS
pin_tft_dc = 17                                      # GPIO Pin number for TFT DC
pin_vs_cs = 27                                      # GPIO Pin number for VS1053 CS
pin_vs_dcs = 14                                     # GPIO Pin number for VS1053 DCS
pin_vs_dreq = 26                                      # GPIO Pin number for VS1053 DREQ
#
preset = 1
preset_00 = 192.95.18.39:5784
preset_01 = 192.95.18.39:5784   #  ChroniX | AGGRESSION

#
toneha = 2
tonehf = 2
tonela = 3
tonelf = 10
#
volume = 84
#

Edzelf, 2021-02-05:
Why set the volume to 40?  I usually set it to 76 and use a the volume control of the external amplifier to get a normal output level.
The VS1053 volume setting should be as high as possible, but below the level of distortion.

xP373Rx, 2021-02-05:
And yeah .. I cant hear any thing below 60 anyway ^^

B4stl3r, 2021-02-05:
Hi @Edzelf , @xP373Rx ,

as you can see here: https://github.com/Edzelf/ESP32-Radio/issues/423#issuecomment-755647412
I try to replace the electronics from a old radio with new one..

But of cause i want to use the old switches from the old radio.

Therefore i want to use the switch poti for power on and volume control.

So I thought i use a amp with no volume control (pam8610) and regulate the volume via the VS..

First of all i also used a regular amplifier but when the volume is ~40% of the AMP, 100% of the VS and I power on the whole circuit, current rushes through the amp and creates loude noises in the speakers.. also I would need additional mechanics to "bride" somehow the old switchpoti with the volume poti of the amp..

xP373Rx, 2021-02-05:
The issue is you are running the pam at full blast any noise or swiching spikes will get amplified to 100 %  A proper amp usualy has nice filtered DC voltage. If you turn up your hifi amp to 100 for example You are gonna still  hear some hiss and they have some nice filters in there.

The proper ways is to have an amp with volume control. it also could be that your input GND and output GND which are in the middle like at 2.5v do not match each other  lets say pam input expects to get  a signal with 0 at 2.5v and your vs has a signal of 2v at 0 your amp would be puting out energy all the time creating noises when there is no sound and this at 100%^^

Rainer-G, 2021-02-06:
Hi all,
the Corona Lockdown gave me some time to help B4stl3r and I want to share my findings.
Most of them were already mentioned before in various threads, using the PAM8610 made some issues more paramount.
The main issue the VS1053 brings along is the analog output referred to VREF and not to ground. This was solved in an earlier thread by using audio transformers to separate the different ground levels. This is good enough if there are no extreme high expectations regarding frequency and phase response. 
Two separate power supplies also help but do not allow the ESP to mute the amplifier using one of the GPIOs. 
The noise issue coming from the ESP32 through VCC / GND can be minimized by the layout of the power rails and filtering capacitors so it even works with a single power source and a PAM8403.

The PAM8610 has a much higher gain ( 33 vs. 24 dB ) so the the noise increases accordingly. Putting a volume control in front of the amp does not really help as the internal pre-amp still runs at highest gain and you can hear some pink noise even at lowest volume!
Nevertheless, the PAM8610 is a very interesting chip: it has balanced ( differential ) inputs, a DC volume control and pop-up reduction. Unfortunately, the breakout board B4stl3r and myself are using does not bring out the volume control pin and the balanced input RIN+ and LIN+. But, new boards are available, I found one with a stereo potentiometer for volume control and one with a mono potentiometer for DC Volume control. 
I'm currently testing the one with the DC Volume control and, at first look, running it with volume zero no noise is audible, even if you put the speaker directly next to your ear. Definitely the best choice if you want a small amp with higher power than the PAM8403! 
If it had the balanced inputs available, it would even make the transformers redundant as you can connect GBUF to RIN+ resp. LIN+!

I will keep you updated
Rainer

B4stl3r, 2021-02-06:
Hi together,

@xP373Rx  yep, correct, as you mentioned, the amp is always at full power.. the thing is, the VS Chip outputs a lot of noise after initialising, which will then be of cause amplifed extremly..

And as @Rainer-G mentioned, we also test different things on the amplifier side..

But in my point of view the question is, WHY does the VS output so much noise after initializing and when volume is muted or set low..

As you know, the VS chip has a lot of possibilities, maybe we just missed out something on how to work "properly" with the VS itself?
http://www.vlsi.fi/fileadmin/datasheets/vs1053.pdf
http://www.vlsi.fi/fileadmin/software/VS10XX/VS1053_VS1063_Equalizer.pdf

Rainer-G, 2021-02-06:
Hallo Martin, ich frage mich ob es der VS ist oder doch der ESP32, der die Störungen erzeugt.Wenn ich mir die Stromaufnahme des ESP32 beim Starten anschaue ( Danke an Andreas Spiess ) und mein Testen mit dem Abblockkondensator tippe ich eher auf den ESP32! Da das ESP Board, das ich habe, mit dem AMS1117 Spannungsregler die 3,3 Volt erzeugt ( der VS hat auch diesen Regler für 3,3 und 1,8 Volt ), habe ich die Versorgung auf 6 Volt hochgesetzt, der AMS ist nicht gerade ein Low Dropout Regler: Und er sollte mindestens 4,75 Volt am Eingang haben, 4,5 könnten auch reichen wenn die Stromaufnahme bei 100 mA liegt, das Datenblatt schweigt sich dazu leider aus:  Viele GrüßeRainer Gesendet von Mail für Windows 10 Von: BastlerGesendet: Samstag, 6. Februar 2021 13:14An: Edzelf/ESP32-RadioCc: Rainer-G; MentionBetreff: Re: [Edzelf/ESP32-Radio] White Noise (#423) Hi together,@xP373Rx yep, correct, as you mentioned, the amp is always at full power.. the thing is, the VS Chip outputs a lot of noise after initialising, which will then be of cause amplifed extremly..And as @Rainer-G mentioned, we also test different things on the amplifier side..But in my point of view the question is, WHY does the VS output so much noise after initializing and when volume is muted or set low..As you know, the VS chip has a lot of possibilities, maybe we just missed out something on how to work "properly" with the VS itself?http://www.vlsi.fi/fileadmin/datasheets/vs1053.pdfhttp://www.vlsi.fi/fileadmin/software/VS10XX/VS1053_VS1063_Equalizer.pdf

xP373Rx, 2021-02-06:
Well on init i also have some heavy noise coming out of the vs  before it is iniatlized as output. WHat i did  since i alos have a bluetoothmodule connected via relays on the output side.  I switch the relays on after init.

xP373Rx, 2021-02-06:
What yxou also could do is pin the mute pin with a 10k resitor to off state  and turn it on after init with pin that is not doing any funny stuff on esp boot

xP373Rx, 2021-05-05:
I found a solution to the spi humming in the audio if using amplifiers..

Do not use the audio GND from the jack .. use the real GND pin on the board and put  100nf or 220 nf  caps in series with the L and R channels to the input of the  amp.

B4stl3r, 2021-11-27:
Hi @xP373Rx 
I don't use GND (it's GBuf) from the Headphone Jack, I just use L + R with capacitors in series to the amp input.

when playing a silent mp3 file (https://bigsoundbank.com/detail-0917-one-minute-of-silence.html) I still have lot of noise..

When music is playing, you basically cant hear and pink noise.

Maybe you can give it a try.

xP373Rx, 2021-11-27:
Gbuf seems to have some resistance  which is making it picking up some of the spi signals..   better to connect to GND and  put 2  ceramic caps on L and R.

xP373Rx, 2021-11-27:
the noise is on the Gbuf not in the channel
 try GND to input and  LR with ceramics

B4stl3r, 2021-11-27:
Hi, here's a quick drawing of my connections:
[image]

xP373Rx, 2021-11-27:
Can you run the esp32  from the 3.3 v VS rail??
Because i belive that is where the  stuff commes from 2 different 3.2 step down  converter

B4stl3r, 2021-11-27:
The VS1053 has a 3.3V Rail ? Thought it requires 5V to run?

Or do you think about grabbing the convertet 3.3V after the "onboard" regulator from the VS Board?

Can it handle the additional Load from the ESP ?

xP373Rx, 2021-11-27:
Esp it self runs on 3.3v  if u connect 3.3v to the 3.3v pin of the board it will run

B4stl3r, 2021-11-27:
sure the esp will run on 3.3, but what you had in mind with the 3.3v rail of the VS ?

Regulator is a ams1117 t33 f16lc
http://www.advanced-monolithic.com/pdf/ds1117.pdf
should be safe to handle the esp in addition, as output is 1A.. correct?

how did you power esp and vs ?

closed 2022-09-28 by B4stl3r

clear-sky, 2023-02-12:
high all
i am 99% sure using Isolated DC/DC Converters like B0505S will kill this noise please see 
https://www.youtube.com/watch?v=n_b_0BWP0mI
i am looking to find this converter in market

Edzelf, 2023-02-12:
Yes, but they are usually only 1 Watt.  If you can find a 3 Watt type it will certainly work.

[4 more replies]