Edzelf/ESP32-Radio #424: Display LCD2004I2C

wuzi42, 2020-12-26:
Hi Edzelf, thank you for ESP32-Radio!
Only a bagatelle - but perheps you know a trick  ...
I build my radio using this oldfashioned LCD. And this unfortunately keeps empty until a station is linked and the music plays. 
Is it possible to have a "Splash-Screen" - saying: yes, the radio is living, wait a moment... ??
Tomas

Edzelf, 2020-12-27:
Try to add the lines:
    dline[0].str = "Radio starts" ;
    dsp_update_line ( 0 ) ;
in LCD1602.h or LCD2004.h  just after "tft = new (LCD....."

wuzi42, 2020-12-28:
thank you very much! It´s very nice that you care about such requests.

Within the function dsp_begin I had no luck, "dline" isn´t available so early.
But from esp32
 if ( dsp_begin() )    {                               // Init display
      dsp_setRotation() ;                                // Use landscape format
      dsp_erase() ;                                      // Clear screen
      dsp_setTextSize ( 1 ) ;                            // Small character font
      dsp_setTextColor ( WHITE ) ;                       // Info in white
      dsp_setCursor ( 0, 0 ) ;                           // Top of screen
      dsp_print ( "Starting..." "\n" "Version:" ) ;
      strncpy ( tmpstr, VERSION, 16 ) ;                  // Limit version length
      dsp_println ( tmpstr ) ;
      dsp_println ( "By Ed Smallenburg" ) ;
       dline[0].str = "Radio starts" ;
      dsp_update_line ( 0 ) ;
      dsp_update() ;                     

it works fine. 
Wish you a happy New Year
Tomas

closed 2020-12-28 by wuzi42