Edzelf/ESP32-Radio #427: [feature request] SPDIF digital output

Daniel-1276, 2021-01-07:
Hi, 
I really like the feature set of ESP32-Radio project. One thing I am really missing is a digital output, instead of DAC board / analog. Based on ESP80266audio library I was able to get a working http-webradio stream outputting SPDIF to my full size stereo amplifier. I am very pleased with the sound quality, but not with the user interface from ESP80266audio project. 
As their SPDIF function is somehow converting I2S data into SPDIF, could you think of a possible integration into your project ?
source: 
https://github.com/earlephilhower/ESP8266Audio/blob/master/src/AudioOutputSPDIF.cpp
https://github.com/earlephilhower/ESP8266Audio/blob/master/src/AudioOutputSPDIF.h

Edzelf, 2021-01-07:
The internet shoutcast stream is mostly MP3, not I2S.  So you need a MP3 converter (software or hardware).

Daniel-1276, 2021-01-07:
Absolutely, you most likely will stream mp3. As far as I understood, you are decoding mp3 on the ESP32 and pass via I2S audio data to VS1053. My point is, if you are willing to add different output protocols, you may could pass I2S data to the mentioned SPDIF library from esp8266audio project. Spdif is so much better on full featured audio amplifiers :) no buzzing, no ground issues, and simple and low cost. Used esp8266audio library together with a simple red Led, that's all you need for digital hifi sound :)

Daniel-1276, 2021-01-07:
[image]

Edzelf, 2021-01-08:
Again, there is no mp3 decoding in the ESP32.  The decoding is done by the VS1053.

clear-sky, 2021-01-08:
please see here 
https://github.com/schreibfaul1/ESP32-audioI2S

Daniel-1276, 2021-01-08:
External hardware is not needed. ESP8266Audio project is using libmad library https://github.com/earlephilhower/ESP8266Audio/tree/master/src/libmad for high quality 24bit MP3 decoding. 
The project is able to stream http-mp3 "radio stations" (AudioInput), process the input (AudioGenerator) and output (AudioOutput) the audio data. Multiple combinations are possible. In my test project I designed a simple TOSLINK adapter that takes a 3mm red diode + 150Ohm resistor. That's all you need to stream high quality Audio. I then added a few more library to output the ID3 tags to a TFT display, to see what is currently playing (nice scrolling text :)). I will upload a video, for demonstartion. My point was to may add SPDIF output capabilities to your user-friendly webradio interface. Adding MP3 decoding would even lower BOM for your project as well.

Rainer-G, 2021-01-08:
nice project, does it suppoort Ogg Vorbis / AAC / WMA / MIDI in addition to MP3?

Daniel-1276, 2021-01-08:
> nice project, does it suppoort Ogg Vorbis / AAC / WMA / MIDI in addition to MP3?

Yes, multiple, just check https://github.com/earlephilhower/ESP8266Audio/tree/master/src and look for AudioGenerator***. It seem the source format must be selected according to the mp3-radio-station you are selecting. To make a good webradio like the one from EDZELF out of ESP8266Audio project, there is still some work to do, but the base on both sides is promising ... it just needs some "best of both worlds" mixing :)

Daniel-1276, 2021-01-08:
Demo https://photos.app.goo.gl/CRukV68d9tjMy3hY9

plueschpruem, 2021-01-15:
While this library works - some time ago I tried it with an I2S codec and it worked right away - it lacks stability.
Using a dedicated decoder surely is better than soft-decoding on such a low-CPU and low-RAM device.
Also: http://www.vsdsp-forum.com/phpbb/viewtopic.php?t=837
You can just enable I2S output on the VS1053 and grab it with a DAC (mind the 16bit limit). Should be rather simple to do.

Derek-K, 2021-01-20:
If one prefer to offload MP3 decode to VS1053...

From the [VS1053 data sheet](https://www.sparkfun.com/datasheets/Components/SMD/vs1053.pdf), section 10.13 (page 76). Look like you just need to enable it via I2S_CONFIG 0xC040 bit 0-3 (including I2S bit rate0 which is set to DISABLE after reset by default.

But the challenge maybe depends on your breakout board has easy access to VS1053 GPIO4/5/6/7 pins.

IMHO, from my experience, I found that it is more stable with a dedicated MP3 decoder chip to decode the stream.

MichaelBuechel, 2022-01-01:
> Demo https://photos.app.goo.gl/CRukV68d9tjMy3hY9

Hi Daniel, can you pls provide a link where to buy the cable with the IR LED in the TOSLINK package ?

Daniel-1276, 2022-01-01:
> > Demo https://photos.app.goo.gl/CRukV68d9tjMy3hY9
> 
> Hi Daniel, can you pls provide a link where to buy the cable with the IR LED in the TOSLINK package ?

Sorry, random stuff laying in my electronic box. Just buy a standard 3mm red LED and I have created a custom 3D TOSLINK plug and that's it

joba-1, 2022-01-25:
> > Demo https://photos.app.goo.gl/CRukV68d9tjMy3hY9
> 
> Hi Daniel, can you pls provide a link where to buy the cable with the IR LED in the TOSLINK package ?

€ 0,32 | 1PC 015A (Transmitting/Receiving End) Dlt1150/Audio Optical Fiber Receiver Terminal Socket Optical Fiber Connector
https://a.aliexpress.com/_vPxTD4

Also had a hard time finding those. Now that I bought them (Rx and Tx) I get flooded with more suggestions :)

Promised delivery time is very long (3 months) but was just 1 month for me.

Cant tell if they work well. I just found your post while searching for info on how to do spdif out on esp32. Need to actually do it yet.