Edzelf/ESP32-Radio #431: Unable to read documentation

QuitButton, 2021-01-28:
What format is the .odt file for documentation? Why is it not in PDF or some standard format?

Edzelf, 2021-01-28:
Hello Kishor,It is the standard for open source Word replacements, like OpenOffice.  You can open the file with Microsoft Word (I think).Vriendelijke groeten,Ed.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Edzelf/ESP32-Radio/issues/431",
"url": "https://github.com/Edzelf/ESP32-Radio/issues/431",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

BePeMee, 2021-02-09:
MS Word is indeed able to open the file, unless you have a very ancient version.

learntoswim, 2021-02-10:
Mac user here. [I formatted it](https://docs.google.com/document/d/1AxJVR-o1UsTDf4eO2ZiYK0yW0ekb4i8i2doUX1KruGM/edit?usp=sharing) (to a level of legibility I was happy with) with Google Docs. No images, though.

Edzelf, 2021-02-11:
For those who have no access to Openoffice, LibreOffice, ...: I added a pdf version.

learntoswim, 2021-02-11:
Thank you, @Edzelf !