Edzelf/ESP32-Radio #432: listen local music on usb flash memory without wifi

clear-sky, 2021-01-29:
hi dear ed and thanks for your great work
i searched but could not find any music player with ch376 + vs1053
only you have done this great work but it does not work without wifi
any solution?

Edzelf, 2021-01-29:
I see no reason why it should not work without WiFi.  Show the logging.

clear-sky, 2021-01-29:
hi dear ed 
thanks for your notice 
please see my nvs keys
clk_dst = 1
clk_offset = 1
clk_server = pool.ntp.org
#
gpio_00 = uppreset = 1
gpio_12 = upvolume = 2
gpio_13 = downvolume = 2
#
ir_40BF = upvolume = 2
ir_C03F = downvolume = 2
#
mqttbroker = none
mqttpasswd = *******
mqttport = 1883
mqttprefix = none
mqttuser = none
#
pin_ch376_cs = 16
pin_ch376_int = 17
pin_enc_clk = -1                                     # GPIO Pin number for rotary encoder CLK
pin_enc_dt = -1                                     # GPIO Pin number for rotary encoder DT
pin_enc_sw = -1                                     # GPIO Pin number for rotary encoder SW
pin_ir = 35                                     # GPIO Pin number for IR receiver VS1838B
pin_sd_cs = -1                                     # GPIO Pin number for SD card CS
pin_shutdownx = 32                                     # GPIO Pin number for AMP shutdown
pin_tft_bl = -1                                     # GPIO Pin number for TFT backlight
pin_tft_cs = 4                                     # GPIO Pin number for TFT CS
pin_tft_dc = -1                                      # GPIO Pin number for TFT DC
pin_vs_cs = 5                                      # GPIO Pin number for VS1053 CS
pin_vs_dcs = 27                                     # GPIO Pin number for VS1053 DCS
pin_vs_dreq = 34                                      # GPIO Pin number for VS1053 DREQ
#
preset = 6
preset_00 = 109.206.96.34:8100                       #  0 - NAXI LOVE RADIO, Belgrade, Serbia
preset_01 = airspectrum.cdnstream1.com:8114/1648_128 #  1 - Easy Hits Florida 128k
preset_02 = us2.internet-radio.com:8050              #  2 - CLASSIC ROCK MIA WWW.SHERADIO.COM
preset_03 = airspectrum.cdnstream1.com:8000/1261_192 #  3 - Magic Oldies Florida
preset_04 = airspectrum.cdnstream1.com:8008/1604_128 #  4 - Magic 60s Florida 60s Classic Rock
preset_05 = us1.internet-radio.com:8105              #  5 - Classic Rock Florida - SHE Radio
preset_06 = icecast.omroep.nl:80/radio1-bb-mp3       #  6 - Radio 1, NL
preset_07 = 205.164.62.15:10032                      #  7 - 1.FM - GAIA, 64k
preset_08 = skonto.ls.lv:8002/mp3                    #  8 - Skonto 128k
preset_09 = 94.23.66.155:8106                        #  9 - *ILR CHILL and GROOVE
preset_10 = ihr/IHR_IEDM                             # 10 - iHeartRadio IHR_IEDM
preset_11 = ihr/IHR_TRAN                             # 11 - iHeartRadio IHR_TRAN
#
toneha = 0
tonehf = 0
tonela = 0
tonelf = 0
#
volume = 72
#
wifi_00 = darmstadt/*******
wifi_01 = SSID2/*******

and this is my log when i shutdown my wifi

ets Jun  8 2016 00:22:5SH⸮HH⸮⸮⸮0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLP*⸮%==Q⸮
configsip: 0, SPIWP:0xee
clk_drv:0x00,qE⸮⸮⸮0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_d.⸮⸮⸮⸮0
mode:DIO, clock div:2
load:0x3fff0018,len'SH⸮⸮+⸮⸮0x3fff001c,len:1216
ho 0 tail 12 room 4
⸮+⸮⸮0x40078000,len:9720
ho 0 tail 12 room 4
loa⸮⸮⸮⸮0080400,len:6364
[... 97 lines cut ...]
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 72
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Start CH376
D: Flash drive not ready
D: Connect to WiFi
D: WiFi Failed!  Trying to setup AP with name ESP32Radio and password ESP32Radio.
D: IP = 192.168.4.1
D: Start server for commands
D: Rotary encoder is disabled (-1/-1/-1)
D: STOP requested
D: Song stopped correctly after 0 msec

Edzelf, 2021-01-29:
The lines:
``D: Start CH376``
``D: Flash drive not ready``
Are interesting.  The ast line should read; 
``"Flash drive ready"``

So there is something wrong with the CH376.

clear-sky, 2021-01-29:
hi dear ed 
thank you for the time spend for me
flash drive works well
this is my log with wifi
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6364
entry 0x400806b8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 19 Oct 2020 14:12:00 GMT.  Free memory 293580
D: Display type is DUMMYTFT
D: Partition nvs found, 20480 bytes
D: Read 47 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to 4
D: pin_tft_dc set to -1
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_ch376_cs set to 16
D: pin_ch376_int set to 17
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 27
D: pin_vs_dreq set to 34
D: pin_shutdown set to -1
D: pin_shutdownx set to 32
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is LOW, probably no PULL-UP
D: GPIO26 is LOW, probably no PULL-UP
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is HIGH
D: GPIO35 is HIGH
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: Enable pin 35 for IR
[... 127 lines cut ...]
D: ID3 COMM = eng
D: ID3 WXXX = 
D: ID3 TENC = NetStream AudioLab
D: End of ID3 info
D: Command client available
D: handlehttp started
D: Get command is: status
D: Duration mp3loop 365
D: Send reply for status
D: Command: status with parameter 0
D: Command client available
D: handlehttp started
D: Get command is: status
D: Send reply for status
D: Command: status with parameter 0
D: Command client available
D: handlehttp started
D: Get command is: status
D: Send reply for status
D: Command: status with parameter 0
D: Command client available
D: handlehttp started
D: Get command is: status
D: Send reply for status
D: Command: status with parameter 0

Edzelf, 2021-01-29:
Strange...  I will try to reproduce it.

Edzelf, 2021-02-01:
Yes, I can reproduce the problem.  But I do not understand what could be wrong.

clear-sky, 2021-02-01:
hi dear ed
thank you very much for your notice
in my opinion it is better to add an option for user something like
1- start in radio mode
2- start in music player mode

Edzelf, 2021-02-01:
Well, it is meant to be an Internet radio....

clear-sky, 2021-05-04:
in setup section wifi is configured and starts
can you give me a costom setup to start mp3 at startup?(radio is not important)
thanks for notice.

LiJu09, 2021-05-04:
hello @clear-sky,
maybe someting like this https://www.aliexpress.com/item/32815183528.html would be more useful and really cheap, if you only need play from usb or sd card

clear-sky, 2021-05-04:
thanks for the answer
no- i dont want to use them
i want to have control and see playing tracks and read id3 tags
serial mp3 players like dfplayer mini have a very limited control of playing files but can not read id-3 tags
using vs1053 and ch376 is my solotion

LiJu09, 2021-05-05:
So I removed wifi config in setup, it might not work at all.
let me know @clear-sky

[ESP32-Radio-no-wifi.zip](https://github.com/Edzelf/ESP32-Radio/files/6425535/ESP32-Radio-no-wifi.zip)

clear-sky, 2021-05-05:
thanks for the time spend for me
it works !
but it needs to send a command via ir remote
please see 
https://github.com/Edzelf/ESP32-Radio/issues/413
looking for a "magic command"  in set up to play a file for example file 1 in folder 1 (without external command like ir remote)

LiJu09, 2021-05-05:
If you mean to automaticaly start playing from usb at startup then try this

[ESP32-Radio-no-wifi.zip](https://github.com/Edzelf/ESP32-Radio/files/6430027/ESP32-Radio-no-wifi.zip)

clear-sky, 2021-05-05:
thanks again for the time spend for me
i had tested that command later but not works "analyzeCmd("mp3track = 0")"
system drops in a rebooting loop
please see system log:
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Thu, 29 Apr 2021 09:20:00 GMT.  Free memory 340704
D: Display type is DUMMYTFT
D: Partition nvs found, 20480 bytes
D: Read 60 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to 4
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_ch376_cs set to 16
D: pin_ch376_int set to 17
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 27
D: pin_vs_dreq set to 34
D: pin_shutdown set to -1
D: pin_shutdownx set to 32
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is LOW, probably no PULL-UP
D: GPIO26 is LOW, probably no PULL-UP
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is HIGH
D: GPIO35 is HIGH
D: GPIO39 is LOW, probably no PULL-UP
D: Enable pin 35 for IR
D: Start display
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: ir_08 with parameter uppreset = 1
D: Command: ir_38c7 with parameter mp3track = 0
D: Command: ir_48 with parameter upvolume = 2
D: Command: ir_4d with parameter downvolume = 2
D: Command: ir_5e with parameter downpreset = 1
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttprefix with parameter none
D: Command: mqttuser with parameter none
D: Command: pin_ch376_cs with parameter 16
D: Command: pin_ch376_int with parameter 17
D: Command: pin_ir with parameter 35
D: Command: pin_shutdownx with parameter 32
D: Command: pin_tft_cs with parameter 4
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 27
D: Command: pin_vs_dreq with parameter 34
D: Command: preset with parameter 7
D: Command: preset_00 with parameter 0n-60s.radionetz.de/0n-60s.mp3
D: Command: preset_01 with parameter 0n-70s.radionetz.de/0n-70s.mp3
D: Command: preset_02 with parameter 0n-80s.radionetz.de/0n-80s.mp3
D: Command: preset_03 with parameter 0n-90s.radionetz.de/0n-90s.mp3
D: Command: preset_04 with parameter 0n-country.radionetz.de/0n-country.mp3
D: Command: preset_05 with parameter 0n-dance.radionetz.de/0n-dance.mp3
D: Command: preset_06 with parameter 0n-disco.radionetz.de/0n-disco.mp3
D: Command: preset_07 with parameter 0n-deutschpop.radionetz.de/0n-deutschpop.mp3
D: Command: preset_08 with parameter 0n-gold.radionetz.de/0n-gold.mp3
D: Command: preset_09 with parameter 0n-kpop.radionetz.de/0n-kpop.mp3
D: Command: preset_10 with parameter 0n-kult.radionetz.de/0n-kult.mp3
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 80
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Start CH376
D: Flash drive ready
D: Locate mp3 files on USB drive, may take a while...
D: 51 tracks on USB
D: Rotary encoder is disabled (-1/-1/-1)
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d9863  PS      : 0x00060530  A0      : 0x800da4b4  A1      : 0x3ffb1f50  
A2      : 0x3f4019ad  A3      : 0x3f4019b6  A4      : 0x00000800  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x00000000  A8      : 0x00000000  A9      : 0x3ffb1f30  
A10     : 0x3f4019b6  A11     : 0x0000003d  A12     : 0x00000000  A13     : 0x3ffaf0a8  
A14     : 0x00000000  A15     : 0x3ffc4348  SAR     : 0x00000020  EXCCAUSE: 0x00000003  
EXCVADDR: 0x3f4019b6  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x400d9863:0x3ffb1f50 0x400da4b1:0x3ffb1f70 0x400e0b3a:0x3ffb1fb0 0x40086fe6:0x3ffb1fd0

Rebooting...

LiJu09, 2021-05-05:
Use previous one and add `mp3track = 0` to the end of config then.

clear-sky, 2021-05-05:
thanks dear Liju09
found the magic comand:
for example if i want to play file 5 in folder 1 use this command:
analyzeCmd("mp3track" ,"1,5,0,0") ;
thanks again for the time spend for me

closed 2021-05-05 by clear-sky

tonypascal, 2023-05-13:
hello,
I have the same issue to solve. I would like to use 3 SPI devices :
+ ch376
+ vs1053
+ SSD1309
I made a custom SSD1309 "library" 
here is my config and log
I manage to have the USB stick and VS1053 work together with the dummy display.
the display works well also, but not the 3 together

D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to 4
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_ch376_cs set to 16
D: pin_ch376_int set to 17
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 27
D: pin_vs_dreq set to 34
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23

can you suggest something ?
best

tonypascal, 2023-05-13:
more details 
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: fs with parameter USB
D: Command: gpio_00 with parameter resume
D: Command: gpio_12 with parameter upvolume = 5
D: Command: gpio_13 with parameter downvolume = 5
D: Command: gpio_14 with parameter stop
D: Command: gpio_15 with parameter resume
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttprefix with parameter none
D: Command: mqttuser with parameter none
D: Command: pin_ch376_cs with parameter 16
D: Command: pin_ch376_int with parameter 17
D: Command: pin_tft_cs with parameter 4
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 27
D: Command: pin_vs_dreq with parameter 34
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 70
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 SPI error. SB:012C R1:0000 R2:0000
D: VS1053 SPI error. SB:0258 R1:0000 R2:0000
D: VS1053 SPI error. SB:0384 R1:0000 R2:0000
D: VS1053 SPI error. SB:04B0 R1:0000 R2:0000
D: VS1053 SPI error. SB:05DC R1:0000 R2:0000
D: VS1053 SPI error. SB:0708 R1:0000 R2:0000
D: VS1053 SPI error. SB:0834 R1:0000 R2:0000
D: VS1053 SPI error. SB:0960 R1:0000 R2:0000
D: VS1053 SPI error. SB:0A8C R1:0000 R2:0000
D: VS1053 SPI error. SB:0BB8 R1:0000 R2:0000
D: VS1053 SPI error. SB:0CE4 R1:0000 R2:0000
D: VS1053 SPI error. SB:0E10 R1:0000 R2:0000
D: VS1053 SPI error. SB:0F3C R1:0000 R2:0000
D: VS1053 SPI error. SB:1068 R1:0000 R2:0000
D: VS1053 SPI error. SB:1194 R1:0000 R2:0000
D: VS1053 SPI error. SB:12C0 R1:0000 R2:0000
D: VS1053 SPI error. SB:13EC R1:0000 R2:0000
D: VS1053 SPI error. SB:1518 R1:0000 R2:0000
D: VS1053 SPI error. SB:1644 R1:0000 R2:0000
D: VS1053 SPI error. SB:1770 R1:0000 R2:0000
D: This is not a VS1053, but a VS1001 instead!
D: Start CH376
D: Flash drive not ready
D: Connect to WiFi
D: Try WiFi Panda
D: Connected to Panda

Edzelf, 2023-05-13:
Publish your complete project, so we can try to reproduce the problem.Verzonden vanaf mijn Galaxy

"target": "https://github.com/Edzelf/ESP32-Radio/issues/432#issuecomment-1546590022",
"url": "https://github.com/Edzelf/ESP32-Radio/issues/432#issuecomment-1546590022",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

tonypascal, 2023-05-13:
all right. How do you wan't me to share it ? zip ? github ?

tonypascal, 2023-05-13:
[ESP32-Radio.zip](https://github.com/Edzelf/ESP32-Radio/files/11469228/ESP32-Radio.zip)

Edzelf, 2023-05-13:
Can you show a picture of your SSD1309? Or is it a SSD1306?  Are you sure is is an SPI type?

tonypascal, 2023-05-13:
Hello
Yes I am almost sure 
The screen works actually, but it seems to be the only one able to use the SPI when I activate the library.
MP3 work (randomly though) when I disable the library. Only the screen works when I activate it…

tonypascal, 2023-05-13:
The model is the DIY more

Edzelf, 2023-05-13:
Picture please....Verzonden vanaf mijn Galaxy

tonypascal, 2023-05-13:
[image]
[image]

tonypascal, 2023-05-13:
Sure

tonypascal, 2023-05-13:
[image]

tonypascal, 2023-05-13:
What do you think ?

Edzelf, 2023-05-13:
Not this weekend.Verzonden vanaf mijn Galaxy

Edzelf, 2023-05-15:
The problem is that you included the library for the SSD1306.  There is no Adafruit library for the SSD1309.
So the best thing is to switch to a display taht is supported.

tonypascal, 2023-05-15:
Thnaks for your feedback! really appreciated!
yes indeed I used the SSD1306 librar, tha actually works with the SSD1309 displays. As you can see, the display also works with the code, except that the rest does not work when I plug it... I'll try with another screen though... but I love this blue LED... 
I am sure other people have the same configuration

Edzelf, 2023-05-16:
If you think the SSD1306 driver should work, change your SSD1309. h like in the attached file.  It will use the hardware SPI driver, instead of the bitbanging version that collides with the SPI bus.
[SSD1309.h.txt](https://github.com/Edzelf/ESP32-Radio/files/11484900/SSD1309.h.txt)
This is the way to use the hardware SPI:
```
tft = new Adafruit_SSD1306 (SCREEN_WIDTH, SCREEN_HEIGHT, &SPI,
                                ini_block.tft_dc_pin, -1, 
                                ini_block.tft_cs_pin ) ;
```

tonypascal, 2023-05-16:
it's working, thank you so much !!!!

tonypascal, 2023-05-25:
hello @clear-sky 

did you manage to make the USB work?
I think I have pretty much the same need as you: simply read MP3 (an PICTURE) files from the USB, and stream them on to VS1053 module. Is there anybody who made a simpler version of the the code without all the fonctions?
I just need:

- playing MP3 files from USB (CH376 + vs1053)
- reading / displaying ID3 on display (ST7789)
- reading / displaying PICTURE from USB on display
- managing USB attached / detached

somebody thinks it is feasible ?
best

clear-sky, 2023-05-25:
Hi dear tonypascal
ch376 driver has a big bug!
i asked from diver owner to correct it but he said that ch376 is chinese and  he can not get enough information from chinese to correct driver!
see here https://github.com/djuseeq/Ch376msc/issues/64
i tried to work with native usb of esp-s3 but its driver is not ready yet!
sorry!
all you need is possible if you forget ch376 and use sd card   
see here 
https://github.com/schreibfaul1/ESP32-vs1053_ext
https://github.com/Bodmer/TFT_eSPI
https://github.com/Bodmer/JPEGDecoder

clear-sky, 2023-05-25:
The Chinese sell the hardware, but they are awful in the field of hardware support, that's exactly the reason why Chinese goods are cheap

tonypascal, 2023-05-26:
sure, but @Edzelf managed to make it work ;-) or almost
who is the diver owner you're talking about?

clear-sky, 2023-05-26:
Hi
when you use ch376 with this driver
https://github.com/djuseeq/Ch376msc
you are limited to use big letters for your file names in usb flash
another limitation is your files must have maximum 8 letters for name and three letters for extention for example : ABCDEFGH.MP3
in my opinion this driver is not usable
ed is done this project with accepting this limitations
i mentioned my conversation with driver owner but you did not notice
djuseeq/Ch376msc#64

clear-sky, 2023-05-26:
even Ed is not happy with this driver please see line 570
https://github.com/Edzelf/ESP32-Radio/blob/master/src/CH376.h

tonypascal, 2023-05-26:
Yes ok
I think it’s maybe more a limitation of the file system. And actually it’s not a problem because the reader can read the ID3 tags. 
So the song name can be stored. Renaming files accordingly can be done too.
Best

goguelnikov, 2023-05-27:
My bad, it’s a driver pb as the chip supports it…
But 2 order of priority problem, as long sa I can read files…