Edzelf/ESP32-Radio #434: Online radio database

xP373Rx, 2021-02-11:
Here  is a little radio database.  By providing your local ip as ipaddr will make you able to start stream directly from the page.
http://gsgrid.net/st1.php

example
http://gsgrid.net/st1.php?ipaddr=192.168.0.167

where 192.168.0.167 is the local esp32 radio ip address.

xP373Rx, 2021-02-11:
`<?php header('Access-Control-Allow-Origin: *'); ?>
<!DOCTYPE HTML>
<html>
<head>

<style>
table {
    border-width: 1px;
    border-style:solid;
    border-color: #000000;
	background-color: #222222;
	margin: 0px;
	padding: 3px;
	/*background-image: url(metal.jpg);*/
	background-attachment: fixed;
	background-repeat: repeat;
	color: #222222;
    border-radius: 15px;
    opacity: 0.9;
}
td { 
border: thin solid #000000;
padding: 4px;
margin: 2px;
color: #cccccc; 
background-color:#111111;
opacity: 0.9;

font-size:13px;
/*background-image: url(td.jpg);*/
background-attachment: fixed;
background-repeat: repeat;
border-radius: 10px;
}

a{
	color: #eeeeee;
    font-style: normal;
	text-decoration: none;
	font-weight: bold;
    font-size:12px;
    
}

.invistd { 
color: #eeeeee;
}
    body {
	background-color: #000000;
    
    background-size: auto;
	color: #cccccc;
	font-size:13px;
    font-family:Snell Roundhand, cursive;

}
input{
    background-color: #000000;
    color: #cccccc; 
    border-color: #222222;
    border-width: 1px;
    border-style:solid;
  
    padding: 3px;
}
select{
    background-color: #000000;
    color: #cccccc; 
    border-color: #222222;
    border-width: 1px;
[... 341 lines cut ...]
     
       // $lasturl=$st->name;

   //}
   
}

echo "</table>"
?>

<script>
boxesEL = document.getElementsByName("playbutton");
      for(var x=0; x < boxesEL.length; x++)   // comparison should be "<" not "<="
      {
        //boxesEL[x].value = "some value";
        boxesEL[x].style.backgroundColor = "#333333";
        boxesEL[x].style.color = "#AAAAAA";
        boxesEL[x].style.fontWeight = 900;
        boxesEL[x].style.width = "450px";
      }
</script>

</center>
</body>
</html>`

Tibertrix, 2021-02-12:
nice !
I'm wondering now ... how difficult is it to make an android app that asks for the ip address and then opens the page?

Edzelf, 2021-02-12:
Yes, great!  I will add a page like that to the webinterface.

xP373Rx, 2021-02-12:
Just open the page on your phone :D

xP373Rx, 2021-02-12:
I intergrated this page via  iframe in to the esps webinterface and forwarded the esp´s ip as arg.

Edzelf, 2021-02-15:
I have added a "search" page to the web interface to explore stations that are available in the "Community Radio Browser", see https://www.radio-browser.info

xP373Rx, 2021-02-16:
Noice works like a charm.

Rainer-G, 2021-02-16:
Excellent idea and fast implementation, 4 days from opening the issue until it found the way into Ed's code!
Many thanks to xP373Rx and Ed!
Compiled and worked from the start, this radio becomes more and more a Swiss Army Knife!
Rainer

xP373Rx, 2021-02-16:
That thing is my sleeping pill right now .. i got some rain tracks and other stuff and implemented a sleep function.
Also working  on adding stations directly from the list  through a store button or popup menu with slot numbers.

utw3v0, 2021-02-21:
Really NICE work guys !!! Do you think it's possible putting this on the Nextion Screen so you don't need a web interface or app ?

steevemadison, 2021-11-06:
message to xP373Rx , are you the maker of "http://gsgrid.net/st1.php" ???
if so there's a mistake in the genre , metal and punk share the same url , but if i cut and paste the url and manually 
enter punk it's working .
ex : gsgrid.net/st1.php?ipaddr=&genre=punk
 but i'm loosing the esp32 index html page , is it pssible to fix that ???
thank you

xP373Rx, 2021-11-06:
I fixed it on the web server .. but if you are using the search code on the esp you need to change it before compile.

xP373Rx, 2021-11-06:
BTW new browsers seem to block access from external pages to local network. I had to disable something in chrome to get it to play on click again.. i moved my whole code to the esp.

steevemadison, 2021-11-09:
thank you for the fix , i appreciate