Edzelf/ESP32-Radio #437: how to add station to presets

MHz000, 2021-02-13:
I'm struggeling adding an other station to the preset list.
So long I've added
preset_12 = st01.sslstream.dlf.de/dlf/01/128/mp3/stream.mp3 # 12 - D_Radio MP3 128k
to the defaultprefs.h recompiled and flashed
Also tried the WEB interface However no success.
Pls advice.
Thank-you

Edzelf, 2021-02-13:
Do not change the source code.  Add stations in the config page of the web interface.

MHz000, 2021-02-14:
Hi Ed,
sorry for me asking again - still the same issue: can't add and play more stations. My presets look like this
#
preset = 1
preset_00 = 109.206.96.34:8100                       #  0 - NAXI LOVE RADIO, Belgrade, Serbia
preset_01 = airspectrum.cdnstream1.com:8114/1648_128 #  1 - Easy Hits Florida 128k
preset_02 = us2.internet-radio.com:8050              #  2 - CLASSIC ROCK MIA WWW.SHERADIO.COM
preset_03 = airspectrum.cdnstream1.com:8000/1261_192 #  3 - Magic Oldies Florida
preset_04 = airspectrum.cdnstream1.com:8008/1604_128 #  4 - Magic 60s Florida 60s Classic Rock
preset_05 = us1.internet-radio.com:8105              #  5 - Classic Rock Florida - SHE Radio
preset_06 = icecast.omroep.nl:80/radio1-bb-mp3       #  6 - Radio 1, NL
preset_07 = 205.164.62.15:10032                      #  7 - 1.FM - GAIA, 64k
preset_08 = skonto.ls.lv:8002/mp3                    #  8 - Skonto 128k
preset_09 = 94.23.66.155:8106                        #  9 - *ILR CHILL and GROOVE
preset_10 = ihr/IHR_IEDM                             # 10 - iHeartRadio IHR_IEDM
preset_11 = ihr/IHR_TRAN                             # 11 - iHeartRadio IHR_TRAN
preset_12 = http://bbcwssc.ic.llnwd.net/stream/bbcwssc_mp1_ws-einws  # 12 - BBC world service
preset_13 = https://st01.sslstream.dlf.de/dlf/01/128/mp3/stream.mp3  # 13 - DLF
#
BBC URL is according to a message from BBC, DLF URL found on the net:
"Below, I have linked the World Service URL's for you to input:
http://bbcwssc.ic.llnwd.net/stream/bbcwssc_mp1_ws-eieuk
http://bbcwssc.ic.llnwd.net/stream/bbcwssc_mp1_ws-einws"
#
tested BBC URL by open it in the web-browser. It is working. Added in WEB Configuration page, saved and restarted ESP32_radio.Both show up in controlpage and can be selected but do not connect/play the station.
Thank-you

MHz000, 2021-02-14:
Ed, in may config the http::/ was not present other as I reported

Edzelf, 2021-02-15:
Yes, sometimes it is hard to find a certain station.
Your preset 13 is probably:

dradio-edge-1097-dus-dtag-cdn.cast.addradio.de/dradio/dlf/live/mp3/128/stream.mp3

MHz000, 2021-02-15:
Success Ed,
that's the station I was after. How did you figure it out? I'm too
interested in BBC world services. Any idea?
Thank-you for your patience.
All the best Joschen

Edzelf, 2021-02-16:
The BBC station works now.  Aparently BBC takes some more tie to react on the initial connection. I jaust added an extra delay of 1 second.
The German URL is a redirection to HTTPS://dradio-....
HTTPS is not supported, but in the new version the radio tries to connect using HTTP and that works,

MHz000, 2021-02-16:
Awesome, thanks! That's really great how much you care. Do I/we have to
download the modified code from github?
I'll wait with testing until I receive PCB's and will report.
Regards