Edzelf/ESP32-Radio #439: Proposal for SSD1306 display's backlight timeout

fenyvesi, 2021-02-16:
The excellent, feature rich program manages turning off a display, but only in case of TFT screens.
I propose to extend it for OLED, too.
1. A public routine in SSSD1306.h:
`void SSD1306::displaySwitch(bool enable)
{
	i2c_cmd_handle_t cmd ;
	cmd = i2c_cmd_link_create() ;
	i2c_master_start ( cmd ) ;
    i2c_master_write_byte ( cmd, (OLED_I2C_ADDRESS << 1) | I2C_MASTER_WRITE, true ) ;
    i2c_master_write_byte ( cmd, OLED_CONTROL_BYTE_CMD_SINGLE, true ) ;
	if (enable ) {
		i2c_master_write_byte ( cmd, OLED_CMD_DISPLAY_ON, true ) ;
	} else {
		i2c_master_write_byte ( cmd, OLED_CMD_DISPLAY_OFF, true ) ;
	}
	i2c_master_stop ( cmd ) ;
	i2c_master_cmd_begin ( I2C_NUM_0, cmd, 10 / portTICK_PERIOD_MS ) ;
	i2c_cmd_link_delete ( cmd ) ;
}`
2. Defining a standalone ckecking-time routine (getting from timer100() interrupt routine) called from  handle_spec(),  avoiding guru meditation error.
`void check_display_live() {							// check switching display off
	if ( ++bltimer == BL_TIME )                   // Time to blank the TFT screen?
    {
      bltimer = 0 ;                               // Yes, reset counter
      blset ( false ) ;                            // Disable TFT (backlight)
    }
}`
Calling:
void handle_spec() {
...
`if ( time_req )                                             // Time to refresh timetxt?
  {
    time_req = false ;                                        // Yes, clear request
    if ( NetworkFound  )                                      // Time available?
    {
		displaytime ( timetxt ) ;                               // Write to TFT screen
		displayvolume() ;                                       // Show volume on display
		displaybattery() ;                                       // Show battery charge on display
		check_display_live();					// switch off display?
    }
  }
...
}`

fenyvesi, 2021-02-16:
I don't know why the inserted code looks so strange. Sorry.

Edzelf, 2021-02-16:
Yes, I refuse to read it.

fenyvesi, 2021-02-16:
Once more. The code feature button somehow distorted the lines.

The excellent, feature rich program manages turning off a display, but only in case of TFT screens.
I propose to extend it for OLED, too.

   1. A public routine in SSSD1306.h:
```
void SSD1306::displaySwitch(bool enable)
{
	i2c_cmd_handle_t cmd ;
	cmd = i2c_cmd_link_create() ;
	i2c_master_start ( cmd ) ;
        i2c_master_write_byte ( cmd, (OLED_I2C_ADDRESS << 1) | I2C_MASTER_WRITE, true ) ;
        i2c_master_write_byte ( cmd, OLED_CONTROL_BYTE_CMD_SINGLE, true ) ;
	if (enable ) {
		i2c_master_write_byte ( cmd, OLED_CMD_DISPLAY_ON, true ) ;
	} else {
		i2c_master_write_byte ( cmd, OLED_CMD_DISPLAY_OFF, true ) ;
	}
	i2c_master_stop ( cmd ) ;
	i2c_master_cmd_begin ( I2C_NUM_0, cmd, 10 / portTICK_PERIOD_MS ) ;
	i2c_cmd_link_delete ( cmd ) ;
}
```
2. Defining a standalone time-ckecking routine (getting from timer100() interrupt routine) called from handle_spec(), avoiding guru meditation error.

```
void check_display_live() {			   // check switching display off
	if ( ++bltimer == BL_TIME )                // Time to blank the TFT screen?
    {
      bltimer = 0 ;                                // Yes, reset counter
      blset ( false ) ;                            // Disable TFT (backlight)
    }
}
```
3. Calling:
```
void handle_spec() {
...
  if ( time_req )                                             // Time to refresh timetxt?
  {
    time_req = false ;                                        // Yes, clear request
    if ( NetworkFound  )                                      // Time available?
    {
		displaytime ( timetxt ) ;                     // Write to TFT screen
		displayvolume() ;                             // Show volume on display
		displaybattery() ;                            // Show battery charge on display
		check_display_live();			      // switch off display?
    }
  }
.....
}
```

fenyvesi, 2021-02-17:
I missed the modified blset() routine itself.
```
void  blset ( bool enable )						// mod by fgy (guru meditation!!!
{
#ifdef OLED
	tft->displaySwitch(enable);
#else
........
#endif
}
```