Edzelf/ESP32-Radio #441: Can some kind person change the Nextion screen to NX4832T035_011 and publish the TFT file for me. Thank you.

Paul-OE8PCK, 2021-02-21:
I would like to use the above mentioned Nextion. NX4832T035_011
If it is not too much work I would require he TFT for this Nection screen. NX4832T035_011
The TFT code published works only for the NX3224T024_011

utw3v0, 2021-02-22:
You just need to Download the Nextion Editor and edit the HMI File to your size you need it's really simple to do

Paul-OE8PCK, 2021-02-22:
Hi,

Sorry I think this is way above my head. I am even having trouble setting up the Arduino sketch with all the headers.

Can I use the existing file and just change the screen size, where can I do that in the edit screen?

 

I have been given the Nextion by someone who also said it was easy, it always is when you know how.

 

Any way thanks for the information. I thought I could just load the existing file to a new screen size but it looks like I have to reprogram the whole thing which is way out of my knowledge window.

Pity, I would love to make the radio, the software part is the hardest for me the other part is really easy. J

Best wishes,

 

Paul

OE8PCK

utw3v0, 2021-02-22:
> Can I use the existing file and just change the screen size, where can I do that in the edit screen?

It's just that easy give me a minute or two i can do it really quick

Paul-OE8PCK, 2021-02-22:
Thank you Sooooooooooooooo much!!!

utw3v0, 2021-02-22:
https://drive.google.com/file/d/1PvF3Lcifan_XRJfCPNFLqzg8PkZIuXR9/view?usp=sharing

Paul-OE8PCK, 2021-02-22:
Hi,

 

Thank you for your kind help,

It is one my big problems out of the way.

I have not yet been able to compile at thee moment, but I will keep looking.

ESP8266.h is missing and I cannot find it in the Arduino library.

 

Thank you once again, at 72 I shold relax and not get into this!!!! J   ._9

 

Very Best wishes,

 

 

Paul

 

https://drive.google.com/file/d/1PvF3Lcifan_XRJfCPNFLqzg8PkZIuXR9/view?usp=sharing

utw3v0, 2021-02-22:
that's wired ... are you compiling at a esp8266 ? did you add the url for your esp to your arduino ? just google arduino and esp8266 /32 you'll get it

Paul-OE8PCK, 2021-02-22:
I am compiling to a ESP32 .

And get this compiling error

 

 

  -> candidates: []

C:\Users\pclay\Documents\Arduino\libraries\Antares_ESP8266_MQTT\src\AntaresESP8266MQTT.h:5:25: fatal error: ESP8266WiFi.h: No such file or directory

compilation terminated.

Multiple libraries were found for "SD.h"

Used: C:\Users\pclay\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD

Not used: C:\Users\pclay\Desktop\Arduino Program IDE\arduino-nightly\libraries\SD

Not used: C:\Users\pclay\Documents\Arduino\libraries\SD

utw3v0, 2021-02-22:
it's complaining abt `C:\Users\pclay\Documents\Arduino\libraries\Antares_ESP8266_MQTT\src\AntaresESP8266MQTT.h:5:25: fatal error: ESP8266WiFi.h: No such file or directory` we don't use that library. Try to install the necessary libraries ... i know it's kinda frustrating i guess i can make a package with all needed libraries ... give me a minute ...

utw3v0, 2021-02-22:
It's  a set from my Masters Project so it should work ... maybe Arduino tells you that there are some updates ... it's fine to update so far .... https://drive.google.com/file/d/1-LpkgHirCYHcny0Hv5KjPgh9xE0vWlJd/view?usp=sharing

Paul-OE8PCK, 2021-02-22:
Thank you, do I just put these files in my arduino library?

What happens if I already have them partly in there?

 

May I please say that I am surprised at your helpfulness, WHY, because when I normally ask questions I get no answers or something that is abusive.

You are really a breath of fresh air in the programming community.. THAN YOU.

 

Paul

utw3v0, 2021-02-22:
Yes, you can make a backup of your libraries if you wish just as a zip file for example but i would use something more reliable like seven zip or  winrar etc. then just copy in everything in your library folder and overwrite existing files ... arduino will tell you that there are updates after a restart just update and you should be godd to go and yes i know there's a harsh wind sometimes, if i can i like to help bc. i know exactly how it is getting no or bad help ... i don't know why it is that way and yes it's sad but it shouldn't make you give up ... giving up is not an option ;) You're Welcome

Greetz Werner

Paul-OE8PCK, 2021-02-22:
Dear Greetz Werner,

 

 

Which Partition Scheme do I need?

I am using an ESP Wroom 32 Dev Module.

 

Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-unused-but-set-parameter -Wno-missing-field-initializers -Wno-sign-compare -fno-rtti -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10814 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"doitESP32devkitV1\"" -DESP32 -DCORE_DEBUG_LEVEL=0 "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4\\cores\\esp32" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4\\variants\\doitESP32devkitV1" "-IC:\\Users\\pclay\\Documents\\Arduino\\libraries\\pubsubclient-master\\src" "-IC:\\Users\\pclay\\Documents\\Arduino\\libraries\\WiFi\\src" "-IC:\\Users\\pclay\\Documents\\Arduino\\libraries\\ESPmDNS\\src" "-IC:\\Users\\pclay\\Documents\\Arduino\\libraries\\SPI\\src" "-IC:\\Users\\pclay\\Documents\\Arduino\\libraries\\ArduinoOTA\\src" "-IC:\\Users\\pclay\\Documents\\Arduino\\libraries\\Update\\src" "-IC:\\Users\\pclay\\Documents\\Arduino\\libraries\\Adafruit_ST7735_and_ST7789_Library" "-IC:\\Users\\pclay\\Documents\\Arduino\\libraries\\Adafruit-GFX-Library-master" "-IC:\\Users\\pclay\\Documents\\Arduino\\libraries\\Ch376msc\\src" "-IC:\\Users\\pclay\\Documents\\Arduino\\libraries\\SD\\src" "-IC:\\Users\\pclay\\Documents\\Arduino\\libraries\\FS\\src" "C:\\Users\\pclay\\Documents\\Arduino\\libraries\\Adafruit-GFX-Library-master\\Adafruit_GrayOLED.cpp" -o nul

Alternatives for Adafruit_I2CDevice.h: []

ResolveLibrary(Adafruit_I2CDevice.h)

  -> candidates: []

Multiple libraries were found for "Update.h"

Used: C:\Users\pclay\Documents\Arduino\libraries\Update

In file included from C:\Users\pclay\Documents\Arduino\libraries\Adafruit-GFX-Library-master\Adafruit_GrayOLED.cpp:20:0:

Not used: C:\Users\pclay\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\Update

C:\Users\pclay\Documents\Arduino\libraries\Adafruit-GFX-Library-master\Adafruit_GrayOLED.h:30:32: fatal error: Adafruit_I2CDevice.h: No such file or directory

Multiple libraries were found for "Adafruit_ST7735.h"

compilation terminated.

Used: C:\Users\pclay\Documents\Arduino\libraries\Adafruit_ST7735_and_ST7789_Library

Not used: C:\Users\pclay\Documents\Arduino\libraries\Adafruit-ST7735-Library-master

Multiple libraries were found for "SD.h"

Used: C:\Users\pclay\Documents\Arduino\libraries\SD

Not used: C:\Users\pclay\Desktop\Arduino Program IDE\arduino-nightly\libraries\SD

Not used: C:\Users\pclay\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SD

Multiple libraries were found for "FS.h"

Used: C:\Users\pclay\Documents\Arduino\libraries\FS

Not used: C:\Users\pclay\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\FS

Multiple libraries were found for "SPI.h"

Used: C:\Users\pclay\Documents\Arduino\libraries\SPI

Not used: C:\Users\pclay\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\SPI

Multiple libraries were found for "WiFiMulti.h"

Used: C:\Users\pclay\Documents\Arduino\libraries\WiFi

Not used: C:\Users\pclay\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi

Multiple libraries were found for "ESPmDNS.h"

Used: C:\Users\pclay\Documents\Arduino\libraries\ESPmDNS

Not used: C:\Users\pclay\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ESPmDNS

Multiple libraries were found for "ArduinoOTA.h"

Used: C:\Users\pclay\Documents\Arduino\libraries\ArduinoOTA

Not used: C:\Users\pclay\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ArduinoOTA

Multiple libraries were found for "PubSubClient.h"

Used: C:\Users\pclay\Documents\Arduino\libraries\pubsubclient-master

Not used: C:\Users\pclay\Documents\Arduino\libraries\PubSubClient-2.7.0

Not used: C:\Users\pclay\Documents\Arduino\libraries\Antares_ESP8266_MQTT

Using library pubsubclient-master at version 2.8 in folder: C:\Users\pclay\Documents\Arduino\libraries\pubsubclient-master 

Using library WiFi at version 1.0 in folder: C:\Users\pclay\Documents\Arduino\libraries\WiFi 

Using library ESPmDNS at version 1.0 in folder: C:\Users\pclay\Documents\Arduino\libraries\ESPmDNS 

Using library SPI at version 1.0 in folder: C:\Users\pclay\Documents\Arduino\libraries\SPI 

Using library ArduinoOTA at version 1.0 in folder: C:\Users\pclay\Documents\Arduino\libraries\ArduinoOTA 

Using library Update at version 1.0 in folder: C:\Users\pclay\Documents\Arduino\libraries\Update 

Using library Adafruit_ST7735_and_ST7789_Library at version 1.6.0 in folder: C:\Users\pclay\Documents\Arduino\libraries\Adafruit_ST7735_and_ST7789_Library 

Using library Adafruit-GFX-Library-master at version 1.7.5 in folder: C:\Users\pclay\Documents\Arduino\libraries\Adafruit-GFX-Library-master 

Using library Ch376msc at version 1.4.3 in folder: C:\Users\pclay\Documents\Arduino\libraries\Ch376msc 

Using library SD at version 1.0.5 in folder: C:\Users\pclay\Documents\Arduino\libraries\SD 

Using library FS at version 1.0 in folder: C:\Users\pclay\Documents\Arduino\libraries\FS 

exit status 1

Error compiling for board DOIT ESP32 DEVKIT V1.

utw3v0, 2021-02-22:
can you show me your sketch ? just the Esp32_radio.ino ? seems you chose the wrong display ? cant really tell whats wrong sry

utw3v0, 2021-02-22:
seems this library is still missing -> https://github.com/adafruit/Adafruit_BusIO.git

Paul-OE8PCK, 2021-02-22:
Using library SD at version 1.0.5 in folder: C:\Users\pclay\Documents\Arduino\libraries\SD 

Using library FS at version 1.0 in folder: C:\Users\pclay\Documents\Arduino\libraries\FS 

exit status 1

'CH376_ATTR_HIDDEN' was not declared in this scope

 

This appears now.

utw3v0, 2021-02-22:
ah looks better now 👍 well are you using a ch376 module ? that's for usb and my group and me have asked ed to help us implement it in the radio so if not just uncomment the line from `#define CH376` to this `//#define CH376` so it's commented out and the compiler will ignore it, otherwise you can put this in the top of ch376.h `#define ATTR_READ_ONLY 0x01 //read-only file
#define ATTR_HIDDEN 0x02 //hidden file
#define ATTR_SYSTEM 0x04 //system file
#define ATTR_VOLUME_ID 0x08 //Volume label
#define ATTR_DIRECTORY 0x10 //subdirectory (folder)
#define ATTR_ARCHIVE 0x20 //archive (normal) file` that should help

Paul-OE8PCK, 2021-02-22:
I decided not to use the ch376 module now but perhaps later…

I get this error message now.

I // the define in the Main.h as you wrote.

 

\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H -DGCC_NOT_5_2_0=0 -DWITH_POSIX "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/config" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_trace" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/app_update" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/asio" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bootloader_support" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/bt" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/coap" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/console" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/driver" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp-tls" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp32" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_event" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_client" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_http_server" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_https_ota" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/esp_ringbuf" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/ethernet" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/expat" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/fatfs" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freemodbus" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/freertos" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/heap" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/idf_test" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/jsmn" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/json" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/libsodium" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/log" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/lwip" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mbedtls" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mdns" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/micro-ecc" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/mqtt" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/newlib" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nghttp" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/nvs_flash" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/openssl" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/protobuf-c" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/protocomm" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/pthread" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/sdmmc" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/smartconfig_ack" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/soc" "-IC:\\Users\\pclay\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.4/tools/sdk/include/spi_flash" "-IC:\\Users\\pclay\\A [... cut ...]
Using library SPI at version 1.0 in folder: C:\Users\pclay\Documents\Arduino\libraries\SPI 

Using library ArduinoOTA at version 1.0 in folder: C:\Users\pclay\Documents\Arduino\libraries\ArduinoOTA 

Using library Update at version 1.0 in folder: C:\Users\pclay\Documents\Arduino\libraries\Update 

Using library SD at version 1.0.5 in folder: C:\Users\pclay\Documents\Arduino\libraries\SD 

Using library FS at version 1.0 in folder: C:\Users\pclay\Documents\Arduino\libraries\FS 

exit status 1

Error compiling for board DOIT ESP32 DEVKIT V1.

utw3v0, 2021-02-22:
i meant in main.cpp line 175 from 
### 
// Define type of local filesystem(s).  See documentation.
#define CH376                          // For CXH376 support (reading files from USB stick)

to this

### 
// Define type of local filesystem(s).  See documentation.
//#define CH376                          // For CXH376 support (reading files from USB stick)

again it would be helpful to see your main.cpp is see ed changed to another ide ....

Paul-OE8PCK, 2021-02-22:
I did change the correct line.

I have enclosed a copy of the main.cpp.

The only change I made before was the Nextion screen enabled and the other screen that was enable I  putt // infront of the enable line.

I hope that I am not annoying you with my problems.

I have to now take my medicine and rest afterwards, would it be OK to contact you tomorrow.

Thank you very much.

Best Wishes,

 

Paul

utw3v0, 2021-02-22:
sure, i am out the house tomorrow but you could mail me your main.cpp to utw3v0@gmail.com so i can take a look at it or i can give you on of mine to see if the error still will occur we will get this fixed ... have a nice day

Paul-OE8PCK, 2021-02-22:
I already mailed it. Did you get it?Sent from my Samsung Galaxy smartphone.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/Edzelf/ESP32-Radio/issues/441#issuecomment-783495708",
"url": "https://github.com/Edzelf/ESP32-Radio/issues/441#issuecomment-783495708",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

marik2500, 2021-03-22:
I tried to play a little with Nextion. Maybe the created file will help. If YES Like !!! 
 https://github.com/marik2500/ESP32_WEBRADIO_NEXTION_FILE/blob/main/ESP32-Radio-Nextion4832.rar