Edzelf/ESP32-Radio #442: no audio signal

MHz000, 2021-02-22:
compiled latest version and download it all seems to work (LCD-display and Web-interface are ok)
BUT no audio signal. Please have a look at my config:

--- Miniterm on COM6  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
D: Scan Networks
D: Scan completed
D: Number of available networks: 12
D:  1 - ESP_8F3C9F                Signal: -33 dBm, Encryption OPEN,
D:  2 - ESP_DF7CED                Signal: -50 dBm, Encryption OPEN, 
D:  3 - worldgate                 Signal: -50 dBm, Encryption WPA2_PSK, Acceptable
D:  4 - JCB                       Signal: -51 dBm, Encryption WPA2_PSK, 
D:  5 - WLAN-XJJMED               Signal: -87 dBm, Encryption WPA2_PSK,
D:  6 - Parents                   Signal: -87 dBm, Encryption WPA2_PSK,
D:  7 - Elektrosmog34b            Signal: -88 dBm, Encryption WPA_WPA2_PSK, 
D:  8 - Gamm                      Signal: -88 dBm, Encryption WPA2_PSK,
D:  9 - Telekom_FON               Signal: -89 dBm, Encryption OPEN, 
D: 10 - FRITZBox-Bodensee         Signal: -90 dBm, Encryption WPA_WPA2_PSK,
D: 11 - Gamm                      Signal: -90 dBm, Encryption WPA2_PSK,
D: 12 - Parents                   Signal: -90 dBm, Encryption WPA2_PSK, 
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: gpio_00 with parameter uppreset = 1
D: Command: gpio_12 with parameter upvolume = 2
D: Command: gpio_13 with parameter downvolume = 2
D: Command: ir_40bf with parameter upvolume = 2
D: Command: ir_c03f with parameter downvolume = 2
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttprefix with parameter none
D: Command: mqttuser with parameter none
D: Command: pin_enc_clk with parameter 25
D: Command: pin_enc_dt with parameter 26
D: Command: pin_enc_sw with parameter 27
D: Command: pin_ir with parameter 35
D: Command: pin_sd_cs with parameter 21
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 16
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 3
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 94
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Connect to WiFi
D: Try WiFi worldgate
D: Connected to worldgate
D: IP = 192.168.178.140
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MDNS responder started
D: Rotary encoder is enabled
D: Sync TOD
D: Sync TOD, new value is 10:37:29
D: STOP requested
D: New preset/file requested (3/0) from airspectrum.cdnstream1.com:8000/1261_192
D: New station request
D: Connect to new host airspectrum.cdnstream1.com:8000/1261_192
D: Connect to airspectrum.cdnstream1.com on port 8000, extension /1261_192
D: Song stopped incorrectly!
D: REG   Contents
D: ---   -----
D:   0 -   808
D:   1 -    40
D:   2 -     0
D:   3 -  6000
D:   4 -     0
D:   5 -  AC44
D:   6 -     0
D:   7 -  1E06
D: D:   8 -     0
  8 -     0
D:   9 -     0
D: SPI semaphore not taken within 270 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D:   A -     0
D:   B -  F8F8
D: SPI semaphore not taken within 280 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D:   C -     0
D:   D -     0
D: SPI semaphore not taken within 290 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D:   E -     0
D:   F -     0
D: Switch to HEADER
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-br:192
D: Headerline: ice-audio-info: ice-samplerate=44100;ice-bitrate=192;ice-channels=2
D: Headerline: icy-br:192
D: Headerline: icy-description:Magic Oldies Florida HD 192 Kbps with the Best Oldies on the Planet!
D: Headerline: icy-genre:Oldies 60s, 70s, 80s
D: Headerline: icy-name:Magic Oldies Florida
D: Headerline: icy-private:0
D: Headerline: icy-pub:1
D: Headerline: icy-url:http://www.magicoldiesflorida.com
D: Headerline: Server: Icecast 2.4.0-kh4
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
D: Headerline: Connection: Close
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 192, metaint is 16000
D: Duration mp3loop 93
D: Metadata block 48 bytes
D: Streamtitle found, 40 bytes
D: StreamTitle='Dionne Warwick - Deja Vu ';
Exception in thread rx:
Traceback (most recent call last):
  File "C:\Users\Win10\.platformio\python3\lib\threading.py", line 926, in _bootstrap_inner
    self.run()
  File "C:\Users\Win10\.platformio\python3\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "c:\users\win10\.platformio\penv\lib\site-packages\serial\tools\miniterm.py", line 499, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "c:\users\win10\.platformio\penv\lib\site-packages\serial\serialwin32.py", line 259, in in_waiting
    raise SerialException("ClearCommError failed ({!r})".format(ctypes.WinError()))
serial.serialutil.SerialException: ClearCommError failed (PermissionError(13, 'Zugriff verweigert', None, 5))

utw3v0, 2021-02-22:
Check your SPI Cabling `D: SPI semaphore not taken within 270 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong` is usually a problem with connection hardware wise

MHz000, 2021-02-22:
SPI cabeling seems to bee ok.
I have changed my HW by replacing BLUETFT by a LCD2004I2C.  Audio signal disappears after that change. Cabeling for the BLUETFT has be removed. Now there is no other SPI-device other than th VS1053 codec. 

The lcd display is working fine as well as the WEB interface. 
Maybe I have to scratch ESP32 NV ram holding the former config?
To play save I'm looking for a simple VS1053 test program. Any hint?
Thank-you

Paul-OE8PCK, 2021-02-27:
I am having the same problem with the VS1003B board. The board says VS1003/1053 MP3 Codec but has a VS1003B chip on it.
The start up listing shows that it is a VS1003B and not a VS1053. Is there a work around?

Paul-OE8PCK, 2021-02-27:
THIS WORKED FOR ME!!!!!!

dagisr commented on 31 Oct 2020
well, well, well ..
I kind of solved it.
It's now working like in old esp8266 setup.
I removed the chip-test result in the program. I have now:

if ( r1 != 4 ) // Version 4 is a genuine VS1053
{
dbgprint ( "This is not a VS1053, " // Report the wrong chip
"but a VS%d instead!",
vstype[r1] ) ;
okay = true ;
}
return ( okay ) ; // Return the result

Anywhay - thank you very much for this fantastic ESP32 RADIO system!!!

Dag in Karlskoga, Sweden

pjritch, 2021-05-24:
I too have the same problem it seems
My "Genuine" 1053 is stamped 1053/1003 but reports as a VS1001 Grrr!

Reports a bunch of SPI errors . Then:-
D: This is not a VS1053, but a VS1001 instead!
D: endFillByte is 0

The above hack doesn't work for me :(