Edzelf/ESP32-Radio #445: Previous station name retained when current name is absent

dheydeck, 2021-02-24:
First, my compliments on this excellent project.
When listening to the following radio stations:
preset_00 = 79.120.77.11:8000/deltablues             #  0 - Radio Caprice: Delta Blues
preset_01 = 79.111.119.111:8000/contclass            #  1 - Radio Caprice: Contemporary Classics
The name of the station appears to be absent. If one of these is the first station to be listened to,  just a hyphen followed by the track name is displayed:
- Stockhausen - Gruss
If a different station has been played previously, its name is prepended to the track name:
Blue Skies Radio - Stockhausen - Gruss
[image]
I have just recompiled with the 19-feb-2021 commit and the issue persists.

utw3v0, 2021-02-24:
i could imagine it has to do with those colons but i'm not sure so maybe you just try it ?

dheydeck, 2021-02-24:
Thanks, utw3v0- I really appreciate the thought. In fact the colons were originally hyphens,  and I changed them to colons because I thought the hyphens might be the problem. The issue is the same in both cases! Here's an example of it working with colons:
[image]

dheydeck, 2021-03-14:
I have submitted Pull Request #450, my first attempt so I may not have followed the correct process. The code nulls the icyname String prior to dbgprint ( "Switch to HEADER" ) ; and also caters for the case where icyname has not been populated and suppresses the hyphen that is usually present between icyname and the the stream title icystreamtitle.
Any suggestions/corrections relating to my use of GitHub are appreciated.

closed 2021-12-13 by dheydeck