Edzelf/ESP32-Radio #447: IR control not responding

MHz000, 2021-02-26:
Now I've that great program up and running - thank-you Ed!
Have connected a VS1838 to GPIO35 and as well pin_ir = 35 in the Config (WEB) page declared
However I don't see any rIR elated output on the serial console (tried different remote controls)
other than this:
D: Command client available
D: handlehttp started
D: Get command is: getnetworks
D: Send reply for getnetworks
D: Command: getnetworks with parameter 0
D: Command client available
D: handlehttp started
D: Get command is: getprefs
D: Send reply for getprefs
...
What is missing in my setup/configuration?

Edzelf, 2021-02-26:
There is probably no signal  on pin 35.  Do you have a way to measure this?

MHz000, 2021-02-26:
Thanks for the fast response Ed. 
For the signal at pin 35 see attached file. **It looks valide.**
My configuration settings @ (Web-page) looks like
pin_enc_clk = 25                                     # GPIO Pin number for rotary encoder CLK
pin_enc_dt = 26                                     # GPIO Pin number for rotary encoder DT
pin_enc_sw = 27                                     # GPIO Pin number for rotary encoder SW
pin_ir = 35                                     # GPIO Pin number for IR receiver VS1838B
pin_sd_cs = 21                                     # GPIO Pin number for SD card CS
pin_tft_cs = 15                                     # GPIO Pin number for TFT CS
pin_tft_dc = 2                                      # GPIO Pin number for TFT DC
pin_vs_cs = 5                                      # GPIO Pin number for VS1053 CS
pin_vs_dcs = 32                                   # GPIO Pin number DCS CHANGED from 16 to 32
pin_vs_dreq = 4 

**Do I have to put the pin settings in defaultprefs.h too?** 
At the moment there is no declaration for pin 35
#define defaultprefs_version 1808016
const char defprefs_txt[] PROGMEM = R"=====(
# Example configuration
# Programmable input pins:
gpio_00 = uppreset = 1
gpio_12 = upvolume = 2
gpio_13 = downvolume = 2
gpio_14 = stop
gpio_17 = resume
gpio_34 = station = icecast.omroep.nl:80/radio1-bb-mp3
#
[image]

Edzelf, 2021-02-26:
Your signal looks good. There is no need to change the defaultprefs.  You could try to use a different pin.

MHz000, 2021-02-26:
just to be sure:
to bring up the I2C connected LCD I had to change pins from -1 to 22/21 in main.cpp at line 2581-82
  struct iosetting klist[] = {                            // List of I/O related keys
    { "pin_ir",        &ini_block.ir_pin,                     -1 }, //                ****************************??
    { "pin_enc_clk",   &ini_block.enc_clk_pin,       -1 },
    { "pin_enc_dt",    &ini_block.enc_dt_pin,        -1 },
    { "pin_enc_sw",    &ini_block.enc_sw_pin,       -1 },
    { "pin_tft_cs",    &ini_block.tft_cs_pin,             -1 }, // Display SPI version
    { "pin_tft_dc",    &ini_block.tft_dc_pin,           -1 }, // Display SPI version
    { "pin_tft_scl",   &ini_block.tft_scl_pin,           22 }, // Display I2C version                       ********
    { "pin_tft_sda",   &ini_block.tft_sda_pin,        21 }, // Display I2C version                       ********
...
do I have to change pin_ir from -1 to 35 too?

do I listen to the right port? Reads terminal output like

D: handlehttp started
D: Get command is: status
D: Send reply for status
D: Command: status with parameter 0
D: Command client available
D: handlehttp started
D: Get command is: status
D: Send reply for status
D: Command: status with parameter 0
Thank-you

Edzelf, 2021-02-26:
Do not change the source files unless you know what you are doing, except for the definition of the display type.
The rest has to be configured in the config page of the web interface.

MHz000, 2021-02-26:
changed HW to  pin 34 and  Config page set pin_ir = 34  save  & restart
terminal output still show no IR response 
power cycle  ... nothing to note
Do I look at the right place? 
Do I have to switch into debug mode?

MHz000, 2021-02-26:
maybe an external pullup R is needed?

Edzelf, 2021-02-26:
I just tested it.  With 5 remotes, input pin 35.  Pull-up is 10k.  They all work.
Remotes are: KPn (TV settop box), Samsung TV, Samsung DVD, and 2 Chinese (for Car DVD and color lamp control).

MHz000, 2021-02-26:
Ed, I've added 10k pull up and switched the IR-remote control
**I see IR-codes NOW** at least with one of several testd IR-remots
Thank-you for your patience
The IR-remotes I've tested are for LED-stripes like the one / similar to the one mentioned  in the instructions. The one for Sony TV-set dosn't work with the radio (but with TV-set)
Maybe the decoding rules are harsh. I'll make some tests.
Thank-you so much

closed 2021-03-02 by MHz000

MHz000, 2021-06-16:
finaly a made it
Thank-you Ed
[image]
[image]

Edzelf, 2021-06-16:
Looks good!  And listening to Petula Clark also a good thing...