Edzelf/ESP32-Radio #448: shutdown not working

MHz000, 2021-03-01:
Hi Ed
declared in Config  pin_shutdown = 34                                  
Save & Restart, source new compiled and flashed 
neithher set Vol = 0, Mute nor Stop resets pin 34 to GND
wiring is ok
What is missing, what wents wrong?
Thank-you

NonaSuomy, 2021-03-01:
GPIO 34-39 have no internal configurable pullup/down, you have to add a 10kohm resistor from the GPIO to 3.3V.

MHz000, 2021-03-01:
> Maybe pin 34 in input only, I am not sure. Please check.

Ok, I'll check.
In the source code? In one of the header files? Pls advice.
Thank you

MHz000, 2021-03-01:
Nona,
as output port there is no need for a pull up R
Thank-you

NonaSuomy, 2021-03-01:
34-39 are input only.

MHz000, 2021-03-01:
in main.cpp line 946:
pinMode(shutdown_pin, OUTPUT);

in Config page WEB interface
pin_shutdown = 34

something missing, wrong interpreted?

MHz000, 2021-03-01:
Nona thank-you for the hint, I'm not aware of that HW limitation. *)
Try to find an other - useable - pin. Maybe Edzelf will give some advice how to use the shutdown function of ESP32-Radio.

*) my ESP32 pin map GPIO34/35  is yellow colored. Are output GPIOs blue colored?

NonaSuomy, 2021-03-01:
https://drive.google.com/file/d/1gbKM7DA7PI7s1-ne_VomcjOrb0bE2TPZ/view

MHz000, 2021-03-02:
using pin 7 GPIO17 => U2_TXD as shutdown works fine now!
Set: pin_shutdown = 17   in Web Config page
Thank-you Nona for pointing me to special HW caracteristics

closed 2021-03-02 by MHz000