Edzelf/ESP32-Radio #449: Sendernamen auf tft refresh

smufte, 2021-03-10:
Hi

I think this project is great.

I only have the problem that the third line (RadioStation) does not empty with shorter station names.

how and where can i change this?

[image]

[image]

I had already increased the font size to 2 on the ILI9341. it may even have something to do with it.
I would still like to leave the font size at 2. 

Many thanks in advance.

Edzelf, 2021-03-11:
The space that is cleared before displaying the station name is defined in "scrseg_struct tftdata[TFTSECS]".  The contents n my source file is now:
```
scrseg_struct     tftdata[TFTSECS] =                        // Screen divided in 3 segments + 1 overlay
{                                                           // One text line is 8 pixels
  { false, WHITE,   0,  8, "" },                            // 1 top line
  { false, CYAN,   20, 64, "" },                            // 8 lines in the middle
  { false, YELLOW, 90, 32, "" },                            // 4 lines at the bottom
  { false, GREEN,  90, 32, "" }                             // 4 lines at the bottom for rotary encoder
} ;

```
The third row (tftdata[2]) is the definition for the station name.  It starts at y=90 and the height is 32.  If you change the font size, you have to change this table accordingly.  Note that the space for rotary encoder text shares the space with the station name.

smufte, 2021-03-11:
Thanks for the tip.
I played a little with the values in this exact section.
And so I solved the problem.