Edzelf/ESP32-Radio #452: PubSubClient.h missing

Zinci, 2021-03-16:
Does anybody know where to get PubSubClient.h?

I am very new to PlatformIO. Just managed to test blink on my ESP32 having to search all the boards, what a pain!

Oh, how I _miss_ Arduino IDE :sunglasses:

Edzelf, 2021-03-16:
Well, with PlatformIO, the libraries needed for a project are all in platforio.ini.  For this project it is like:
```
lib_deps = 
	Wire
	SD(esp32) @^1.0.5
	knolleary/PubSubClient@^2.8
	adafruit/Adafruit ST7735 and ST7789 Library@^1.6.0
	adafruit/Adafruit BusIO@^1.7.1
	adafruit/Adafruit GFX Library@^1.10.4
	djuseeq/Ch376msc @ ^1.4.4

```
The libraries will be intstalled automagically when building the project.

pierangelof, 2021-03-18:
Hi @Zinci 
I had a similar problem. Despite the PubSubClient library should be automatically loaded as @Edzelf said, I had to do it manually.
I suggest you to read the following post about the installation of libraries in Platform IO.
https://community.platformio.org/t/how-do-i-add-libraries-after-importing-an-arduino-sketch/10274/2
I hope it helps.

Zinci, 2021-03-18:
Thanks for the tips.
I was pulling my hair out. 
I will try this solution and search the web some more.

closed 2021-03-28 by Zinci