Edzelf/ESP32-Radio #453: help for adding TFT_eSPI

clear-sky, 2021-03-21:
hi all
i am very happy with this great work
very very thanks especially for adding search tool for this great project
i am trying to add ili9488 spi 480x320
the only working driver that i found is here
https://github.com/Bodmer/TFT_eSPI
can anybody give me some help for using this display for this project
my problem is here
tft = new Adafruit_ILI9341 ( ini_block.tft_cs_pin,ini_block.tft_dc_pin ) ; 
if you see driver code will know all pins of display is configured in driver folder (User_Setup_Select.h) 
i could not find any way to config display pin  in sketch

silk-indus, 2021-05-02:
Dear clear-sky.
Look on my fork, I did it.

clear-sky, 2021-05-05:
thanks for reply

closed 2021-05-05 by clear-sky