Edzelf/ESP32-Radio #456: Reboot loop when NEXTION display is chosen.  Other display options seem to work.

truckershitch, 2021-04-05:
Hi there,

This project has been working for me with both the BLUETFT and DUMMYTFT display options.  I am able to also play files off the sd card.

I got a Nextion NX4024K032 display a couple of days ago in the mail and a few CR1220 batteries for it.  The display runs the test program fine, and I managed to scale the included HMI file to my resolution.  It starts up when I supply the power while disconnected from the project.

Unfortunately, my setup does not like the Nextion configuration option and reboots repeatedly as soon as the WiFi starts.  This happens whether the display is connected or not.  Switching back to "DUMMYTFT" fixes everything and the program runs as it should.

I have the project set up in Visual Studio Code with PlatformIO.  I am familiar with Visual Studio Code but I'm unfamiliar with PlatformIO.  I think it's pretty great so far but I don't know the particulars yet.

The first paste is the `platformio.ini` configuration.  It could be incorrect.

I have a ESP32-WROVER module connected to the project.

This log was recorded after erasing the flash, using the Clean option and then choosing Upload and Monitor.

Thanks!

```
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp-wrover-kit]
platform = espressif32
board = esp-wrover-kit
framework = arduino
upload_port = /dev/ttyUSB0
monitor_port = /dev/ttyUSB0
monitor_speed = 115200
monitor_dtr = 1
monitor_rts = 1
monitor_flags = 
	--filter=esp32_exception_decoder
	--echo
build_flags = 
	-DCORE_DEBUG_LEVEL=0
	-Os
lib_deps = 
	Wire
	SD(esp32) @^1.0.5
	knolleary/PubSubClient@^2.8
	adafruit/Adafruit ST7735 and ST7789 Library@^1.6.0
	adafruit/Adafruit BusIO@^1.7.1
	adafruit/Adafruit GFX Library@^1.10.4
	djuseeq/Ch376msc @ ^1.4.4

```
Here is the serial port log:
```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x37 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Fri, 19 Feb 2021 11:50:00 GMT.  Free memory 307968
D: Display type is NEXTION
D: Partition nvs found, 20480 bytes
D: Read 0 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_ch376_cs set to -1
D: pin_ch376_int set to -1
D: pin_vs_cs set to -1
D: pin_vs_dcs set to -1
D: pin_vs_dreq set to -1
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: Create list with acceptable WiFi networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 13
D:  1 - XXX Signal: -38 dBm, Encryption WPA2_PSK, 
D:  2 - XXX Signal: -63 dBm, Encryption WPA2_PSK, 
D:  3 - XXX Signal: -76 dBm, Encryption WPA2_PSK, 
D:  4 - XXX Signal: -85 dBm, Encryption WPA2_PSK, 
D:  5 - XXX Signal: -85 dBm, Encryption WPA2_PSK, 
D:  6 - XXX Signal: -89 dBm, Encryption WPA2_PSK, 
D:  7 - XXX Signal: -90 dBm, Encryption WPA_WPA2_PSK, 
D:  8 - XXX Signal: -91 dBm, Encryption WPA2_PSK, 
D:  9 - XXX Signal: -91 dBm, Encryption WPA2_PSK, 
D: 10 - XXX Signal: -91 dBm, Encryption WPA2_PSK, 
D: 11 - XXX Signal: -92 dBm, Encryption WPA2_PSK, 
D: 12 - XXX Signal: -92 dBm, Encryption WPA2_PSK, 
D: 13 - XXX Signal: -93 dBm, Encryption WPA2_PSK, 
D: End of list
D: Slow SPI, Testing VS1053 read/write registers...
D: VS1053 not properly installed!
D: Connect to WiFi
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400dfd43  PS      : 0x00060830  A0      : 0x800d2006  A1      : 0x3ffb1e60  
A2      : 0x00000000  A3      : 0x3f400304  A4      : 0x0000003f  A5      : 0x3ffcb81c  
A6      : 0x3ffb1e9c  A7      : 0x3ffb1e9c  A8      : 0x800dfd11  A9      : 0x3ffb1e20  
A10     : 0x0000001b  A11     : 0x00000040  A12     : 0x3f400304  A13     : 0x3ffb1ee0  
A14     : 0x3ffb1e60  A15     : 0x00000008  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

ELF file SHA256: 0000000000000000

Backtrace: 0x400dfd43:0x3ffb1e60 0x400d2003:0x3ffb1f00 0x400d30d5:0x3ffb1f20 0x400d99e9:0x3ffb1f40 0x400e2b82:0x3ffb1fb0 0x4008a4ca:0x3ffb1fd0
  #0  0x400dfd43:0x3ffb1e60 in Print::printf(char const*, ...) at /home/hitch/.platformio/packages/framework-arduinoespressif32/cores/esp32/Print.cpp:261
  #1  0x400d2003:0x3ffb1f00 in dsp_println(char const*) at src/main.cpp:2001
  #2  0x400d30d5:0x3ffb1f20 in tftlog(char const*) at src/main.cpp:2001
  #3  0x400d99e9:0x3ffb1f40 in setup() at src/main.cpp:2001
  #4  0x400e2b82:0x3ffb1fb0 in loopTask(void*) at /home/hitch/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:18
  #5  0x4008a4ca:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

Edzelf, 2021-04-06:
It looks like it crashes because there is no SSID/password in NVS.  Try to run Esp32_radio_init.ino once in the tools directory with your WiFi SSID and password in lines 40-41.  After that, upload the Esp32-radio sketch.

truckershitch, 2021-04-07:
Thanks, Ed.  That worked well.