Edzelf/ESP32-Radio #457: Problem connecting to .m3u-URLs with latest version

EFWob, 2021-04-24:
I wanted to migrate to the latest version only to encounter a serious issue (it is serious,
as favorite radio station of wife is affected). I am no longer able to play playlist-URLs 
(ending in ".m3u"). I have a few such stations in the presets, all of them I tried did not work.

I did not find any related issue, so may be it is just me!? 

For reference, my setup:
barebone system, just ESP32 (Olimex PoE) and VS1053. Defined is DUMMYTFT and undefined
are CH376 and SDCARD. I also use a different NAME to identify NVS partition, but except from 
this the software is as on Github.
Everything works fine, but starting with commits after Feb. 11 (change to PlatformIO version)
I can no longer connect to m3u. Test setup is as follows: radio is playing preset_01 correctly
(st01.dlf.de/dlf/01/128/mp3/stream.mp3). Then command "preset=0" is issued 
(in preferences "preset_00 = metafiles.gl-systemhaus.de/hr/hr1_2.m3u  #   HR1")

Serial Output is as follows:
```
23:54:43.508 -> D: Command: preset with parameter 0
23:54:43.508 -> D: Preset is now 0
23:54:43.508 -> D: STOP requested
23:54:43.541 -> D: Stopping client
23:54:44.038 -> D: New preset/file requested (0/0) from metafiles.gl-systemhaus.de/hr/hr1_2.m3u
23:54:44.038 -> D: New station request
23:54:44.038 -> D: Connect to new host metafiles.gl-systemhaus.de/hr/hr1_2.m3u
23:54:44.038 -> D: Playlist request, entry 1
23:54:44.038 -> D: Connect to metafiles.gl-systemhaus.de on port 80, extension /hr/hr1_2.m3u
23:54:44.138 -> D: Connected to server
23:54:44.171 -> D: Song stopped correctly after 20 msec
23:55:07.574 -> D: STOP requested
23:55:07.574 -> D: Stopping client
23:55:08.104 -> D: New preset/file requested (1/0) from st01.dlf.de/dlf/01/128/mp3/stream.mp3
```
So no audio, and after a while, at 23:55:08.104 (last line) the radio decided to switch to preset 1.

I first suspected the addition of a 1000ms delay in connecttohost() might cause a problem.
So I removed that delay but that did not help. I even copied connecttohost() from a working version
(Feb 11.), but that does not change it at all. Just the timestamps might differ.

Things do not improve if URL is entered on Web-Interface (Serial output slightly differes).

Anybody else had the same problem and did possibly locate the root cause? I noticed that the commit
from Feb. 15 caused even more trouble (even most of .mp3-URLs did not work correctly). Things have 
improved since, but for me .m3u-URLs are not working.

For comparison, the Serial output with the same test case with latest working version 
(Feb. 11 commit 98a67db329fdc3030e19a9dc127c7219d8eae86b)

```
00:02:21.844 -> D: Command: preset with parameter 0
00:02:21.844 -> D: Preset is now 0
00:02:21.844 -> D: STOP requested
00:02:21.844 -> D: Stopping client
00:02:22.341 -> D: New preset/file requested (0/0) from metafiles.gl-systemhaus.de/hr/hr1_2.m3u
00:02:22.341 -> D: Connect to new host metafiles.gl-systemhaus.de/hr/hr1_2.m3u
00:02:22.375 -> D: Playlist request, entry 1
00:02:22.375 -> D: Connect to metafiles.gl-systemhaus.de on port 80, extension /hr/hr1_2.m3u
00:02:22.474 -> D: Connected to server
00:02:22.514 -> D: Read from playlist
00:02:22.514 -> D: Playlistheader: HTTP/1.1 200 OK
00:02:22.514 -> D: Playlistheader: Accept-Ranges: bytes
00:02:22.514 -> D: Playlistheader: Content-Type: audio/x-mpegurl
00:02:22.514 -> D: Playlistheader: ETag: "after 20 msec (shorted because of strange char included in output)
00:02:22.562 -> D: Playlistheader: Last-Modified: Sat, 13 May 2017 07:40:14 GMT
00:02:22.562 -> D: Playlistheader: Last-Modified: Sat, 13 May 2017 07:40:14 GMT
00:02:22.562 -> D: Playlistheader: Server: AkamaiNetStorage
00:02:22.562 -> D: Playlistheader: Content-Length: 66
00:02:22.562 -> D: Content-Length is 66
00:02:22.562 -> D: Playlistheader: Expires: Sat, 24 Apr 2021 22:02:22 GMT
00:02:22.562 -> D: Playlistheader: Cache-Control: max-age=0, no-cache, no-store
00:02:22.562 -> D: Playlistheader: Pragma: no-cache
00:02:22.562 -> D: Playlistheader: Date: Sat, 24 Apr 2021 22:02:22 GMT
00:02:22.562 -> D: Playlistheader: Connection: close
00:02:22.562 -> D: Playlistheader: Access-Control-Max-Age: 86400
00:02:22.598 -> D: Playlistheader: Access-Control-Allow-Credentials: false
00:02:22.598 -> D: Playlistheader: Access-Control-Allow-Headers: *
00:02:22.598 -> D: Playlistheader: Access-Control-Allow-Methods: GET,POST
00:02:22.598 -> D: Playlistheader: Access-Control-Allow-Origin: *
00:02:22.598 -> D: Playlistheader: Content-Type: audio/x-mpegurl
00:02:22.598 -> D: Playlistheader: 
00:02:22.598 -> D: Switch to PLAYLISTDATA, search for entry 1
00:02:22.598 -> D: Playlistdata: http://hr-hr1-live.cast.addradio.de/hr/hr1/live/mp3/128/stream.mp3
00:02:22.633 -> D: Entry 1 in playlist found: http://hr-hr1-live.cast.addradio.de/hr/hr1/live/mp3/128/stream.mp3
00:02:22.633 -> D: Connect to new host hr-hr1-live.cast.addradio.de/hr/hr1/live/mp3/128/stream.mp3
00:02:22.633 -> D: Connect to hr-hr1-live.cast.addradio.de on port 80, extension /hr/hr1/live/mp3/128/stream.mp3
00:02:22.706 -> D: Connected to server
00:02:22.706 -> D: Duration mp3loop 193
00:02:22.739 -> D: Switch to HEADER
00:02:22.739 -> D: Headerline: Server: nginx
00:02:22.739 -> D: Headerline: Content-Length: 0
00:02:22.739 -> D: Headerline: Connection: close
00:02:22.739 -> D: Headerline: Cache-Control: no-cache, no-store
00:02:22.782 -> D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
00:02:22.782 -> D: Headerline: Access-Control-Allow-Origin: *
00:02:22.782 -> D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
00:02:22.782 -> D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
00:02:22.782 -> D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
00:02:22.782 -> D: Headerline: Location: http://hr-edge-3049-dus-ts-cdn.cast.addradio.de/hr/hr1/live/mp3/128/stream.mp3?_art=dj0yJmlwPTIxMy4xNjIuMTM3Ljc1JmlkPWljc2N4bC1
00:02:22.782 -> D: Switch to DATA, bitrate is 0, metaint is 0
00:02:22.818 -> D: Connect to new host hr-edge-3049-dus-ts-cdn.cast.addradio.de/hr/hr1/live/mp3/128/stream.mp3?_art=dj0yJmlwPTIxMy4xNjIuMTM3Ljc1JmlkPWljc2N4bC1wMm9qeGps
00:02:22.818 -> D: Connect to hr-edge-3049-dus-ts-cdn.cast.addradio.de on port 80, extension /hr/hr1/live/mp3/128/stream.mp3?_art=dj0yJmlwPTIxMy4xNjIuMTM3Ljc1JmlkPWljc2
00:02:22.905 -> D: Connected to server
00:02:22.971 -> D: Switch to HEADER
00:02:22.971 -> D: Headerline: Content-Type: audio/mpeg
00:02:22.971 -> D: audio/mpeg seen.
00:02:22.971 -> D: Headerline: icy-name:hr1
00:02:22.971 -> D: Headerline: icy-pub:0
00:02:22.971 -> D: Headerline: icy-url:http://www.hr.de
00:02:22.971 -> D: Headerline: Server: dg-hr-http_dus-ts_edge_e0c8fb7a4f267059e4cc214ce367edc1
00:02:23.010 -> D: Headerline: Cache-Control: no-cache, no-store
00:02:23.010 -> D: Headerline: Access-Control-Allow-Origin: *
00:02:23.010 -> D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type
00:02:23.010 -> D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type seen.
00:02:23.010 -> D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, HEAD
00:02:23.010 -> D: Headerline: Connection: Close
00:02:23.010 -> D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
00:02:23.041 -> D: Headerline: icy-metaint:16000
00:02:23.041 -> D: Switch to DATA, bitrate is 0, metaint is 16000
00:02:23.104 -> D: Metadata block 48 bytes
00:02:23.104 -> D: Streamtitle found, 33 bytes
00:02:23.104 -> D: StreamTitle=' - hr1 Nachrichten';
```

Edzelf, 2021-04-25:
Fixed.

EFWob, 2021-04-25:
Fix confirmed. Excellent, thank you!

closed 2021-04-25 by EFWob