Edzelf/ESP32-Radio #458: Wrong offset on SSD1309

LiJu09, 2021-04-28:
I have SSD1309 OLED in I2C mode.

[image]

Also tried demo from U8G2 library and its working properly, so it looks like problem is in this library.

Thanks for help.

Log:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Sun, 25 Apr 2021 12:30:00 GMT.  Free memory 288032
D: Display type is OLED1309
D: Partition nvs found, 20480 bytes
D: Read 43 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to 5
D: pin_tft_sda set to 21
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_ch376_cs set to -1
D: pin_ch376_int set to -1
D: pin_vs_cs set to 22
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_12 will execute upvolume = 2
D: gpio_13 will execute downvolume = 2
D: gpio_14 will execute downpreset = 1
D: gpio_33 will execute uppreset = 1
D: Enable pin 35 for IR
D: Start display
D: Init OLED, I2C pins 21,5
D: ISSH116 is 0
D: Create list with acceptable WiFi networks
D: Added *** to list of networks
D: Added SSID2 to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 8
D:  1 ***
D:  2 ***
D:  3 ***
D:  4 ***
D:  5 ***
D:  6 ***
D:  7 ***
D:  8 ***
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: gpio_12 with parameter upvolume = 2
D: Command: gpio_13 with parameter downvolume = 2
D: Command: gpio_14 with parameter downpreset = 1
D: Command: gpio_33 with parameter uppreset = 1
D: Command: ir_40bf with parameter upvolume = 2
D: Command: ir_c03f with parameter downvolume = 2
D: Command: mqttbroker with parameter ***
D: Command: mqttpasswd with parameter ***
D: Command: mqttport with parameter 1883
D: Command: mqttprefix with parameter espradio
D: Command: mqttuser with parameter espradio
D: Command: pin_ir with parameter 35
D: Command: pin_sd_cs with parameter -1
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_tft_scl with parameter 5
D: Command: pin_tft_sda with parameter 21
D: Command: pin_vs_cs with parameter 22
D: Command: pin_vs_dcs with parameter 32
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 2
D: Command: preset_00 with parameter stream.funradio.sk:8000/fun192.mp3
D: Command: preset_01 with parameter stream.funradio.sk:8000/dance128.mp3
D: Command: preset_02 with parameter streaming.radionomy.com/JamendoLounge
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 1
D: Command: tonela with parameter 15
D: Command: tonelf with parameter 15
D: Command: volume with parameter 60
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Connect to WiFi
...
```

Edzelf, 2021-04-28:
I do not have a 1309, so I can't test it.

LiJu09, 2021-04-28:
But why there is choice,
https://github.com/Edzelf/ESP32-Radio/blob/b56c3256dece8603ec8df7df94fe57b62c1247af/src/main.cpp#L180-L181

when it use same library anyway?
https://github.com/Edzelf/ESP32-Radio/blob/b56c3256dece8603ec8df7df94fe57b62c1247af/src/main.cpp#L1132-L1137

LiJu09, 2021-04-29:
When I defined OLED1106 is it little bit better, but still wrong.

[image]

LiJu09, 2021-04-29:
When still defined OLED1106 and comented out these lines

https://github.com/Edzelf/ESP32-Radio/blob/b56c3256dece8603ec8df7df94fe57b62c1247af/src/oled.h#L402-L405

even better but still wrong.

[image]

LiJu09, 2021-04-29:
Still same code but rebooted ESP few times and its fixed.

@Edzelf Can I make PR with my fix?

[image]

LiJu09, 2021-04-29:
Fixed with commit https://github.com/Edzelf/ESP32-Radio/commit/503ef872ea8bb3fedc2e532e14548a0438c4b569

closed 2021-04-29 by LiJu09