Edzelf/ESP32-Radio #463: Production Building?

hughesdo, 2021-05-13:
Is anyone building these is a production manner?  I've been monitoring this site for over a year. I would simply like to know if I can buy a pre-built one of these that is semi professionally built.  I've made a couple of half hearted attempts to build this myself with varying degrees of success.  Mainly failing at the TTF screens.  At any rate I give up!  I would just like to see a fully populated one to buy pre-built with software that will compile and included so that I can still "play" with the code.

NonaSuomy, 2021-09-01:
Lilygo.cn used to produce this board http://www.lilygo.cn/prod_view.aspx?TypeId=50032&Id=1157&FId=t3:50032:3 https://www.aliexpress.com/item/32848882218.html you can still find them around (not sure why they still don't produce them if there was an issue with the board) Edzelf said he is trying to add that I2S chipset for ver 2 so that would basically be the full package. My only issue with it was that it didn't use a prepackaged esp32 with the rf shielding can on top and the latest ESP32 ECO V3 chipset.

Ne3Mx, 2021-09-01:
> Lilygo.cn used to produce this board http://www.lilygo.cn/prod_view.aspx?TypeId=50032&Id=1157&FId=t3:50032:3 https://www.aliexpress.com/item/32848882218.html you can still find them around [...]

I think we need to spec some kind of feature list that the new unit(pcb) needs and do an opensource pcb. For me it'll need to have a class D amp 5W - 30W, be able to run from 12V, an monochrome OLED display, a rotary dial and some buttons. Is the ESP32 ECO V3 the dual core ESP32?

NonaSuomy, 2021-09-01:
There was a bug in the silicon that caused glitches with PSRAM and a few other things all they did was fix the silicon for these issues document about it below but it's still the same ESP32.

https://www.espressif.com/sites/default/files/documentation/ESP32_ECO_V3_User_Guide__EN.pdf