Edzelf/ESP32-Radio #464: Compile prob in Arduino IDE

pjritch, 2021-05-23:
Thank you for a most excellent project.
I found this watching the guy with the Swedish accent and it is so well documented I decided to jump in at the deep end and have a go.
I am new to coding and don't yet fully understand how some of this stuff works,
and I have a problem that I hope you can help me.

Compiling fails for me at line 342: undefined reference to `PubSubClient::~PubSubClient()
Seems like something to do with MQTT?
Have I missed something?

Maybe I am like Andreas Spiess would say "Most stupid user"! lol

Edzelf, 2021-05-23:
Seems like you have the wrong MTTQ library.
If you are using PlatformIO, the right library is always included.  If you are still using the Arduino IDE, take care that you include "knolleary/PubSubClient@^2.8".

pjritch, 2021-05-23:
Thank you so much for the fast response
You were, of course, correct
I had the wrong library
problem appears to be fixed now but no more time to play today so setting it all up will have to wait for another day

closed 2021-05-24 by pjritch