Edzelf/ESP32-Radio #465: simple esp radio continu boot

benstoffels, 2021-05-24:
Hello the esp32 doit continued boot simple web radio, before the simple web radio work fine!

Greets 

ets Jun  8 2016 00:22:57
19:13:14.564 -> 
19:13:14.564 -> rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
19:13:14.564 -> configsip: 0, SPIWP:0xee
19:13:14.564 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
19:13:14.564 -> mode:DIO, clock div:1
19:13:14.564 -> load:0x3fff0018,len:4
19:13:14.564 -> load:0x3fff001c,len:1044
19:13:14.597 -> load:0x40078000,len:10124
19:13:14.597 -> load:0x40080400,len:5856
19:13:14.597 -> entry 0x400806a8
19:13:52.886 -> .....WiFi connected
19:13:54.906 -> IP address:192.168.178.31
19:13:54.906 -> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
19:13:54.906 -> Core 1 register dump:
19:13:54.906 -> PC      : 0x400014fd  PS      : 0x00060a30  A0      : 0x800d4ff1  A1      : 0x3ffb1ea0  
19:13:54.906 -> A2      : 0x00000050  A3      : 0x0000004c  A4      : 0x000000ff  A5      : 0x0000ff00  
19:13:54.906 -> A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3ffb1e70  
19:13:54.939 -> A10     : 0x00000000  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3ffb1f46  
19:13:54.939 -> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
19:13:54.939 -> EXCVADDR: 0x00000050  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff

Edzelf, 2021-05-24:
2 posibilities: Bad power supply or NVS is empty.

benstoffels, 2021-05-24:
What means NVS 
Power is ok

Greets Ben

benstoffels, 2021-05-24:
Other programs work fine

benstoffels, 2021-05-25:
How can i resolve empty NVS?

Greets Ben

Edzelf, 2021-05-25:
See doc, page 21, "Alternative way of configuration."

benstoffels, 2021-05-25:
Ik heb helemaal geen ervaring hiermee. 
Heeft u een code voor mij. 

Bij voorbaat dank

Verstuurd vanaf mijn iPhone

Edzelf, 2021-05-25:
See tools directory for the init sketch.

benstoffels, 2021-05-27:
No succes

closed 2021-05-27 by Edzelf