Edzelf/ESP32-Radio #468: Station name and song title not showing with a specific station.

ve2cfb, 2021-06-20:
Hello Ed,

Here in Canada, our national radio CBC has removed all info regarding song title and station ID when streaming all CBC stations..
Here is an example:  cbcmp3.ic.llnwd.net/stream/cbcmp3_M-7QMTL0_MTL 
Preset file:  preset_56 = cbcmp3.ic.llnwd.net/stream/cbcmp3_M-7QMTL0_MTL # 56 - Ici Musique 100.7
I would like to  to  use the NVS preset info instead.
I beleive there was an issue concerning this problem, but cannot find it.
How can i do it.

Thanks a lot...  Pierre

Edzelf, 2021-06-21:
Hello Pierre,
I fixed this in the new version.  If you have comment in the preset line, it will display the comment, otherwise the station URL.  In your case:
`56 - Ici Musique 100.7`
or
`cbcmp3.ic.llnwd.net/stream/cbcmp3_M-7QMTL0_MTL`

ve2cfb, 2021-06-21:
Thanks very much Ed, 
That is exactly what i want.
I will check the new Version.

Have a nice day...  Pierre

closed 2021-06-21 by ve2cfb