Edzelf/ESP32-Radio #472: Debug output looks correct, but no sound plays

robertpi, 2021-07-10:
Hello,

First, thanks for all the hard work on this project, it looks great.

I'm trying to get this project working on a very basic setup of just an ESP32 and VS1053. 

I think I've wired things up correctly, as I've checked it a couple of times. The debug output looks correct, yet no sound is played when I connect speakers to the ESP32. I can connect to the web server on the IP shown in the output.

Any ideas what's going wrong?

Many thanks,
Robert

```
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at http://bit.ly/pio-monitor-filters
--- forcing DTR inactive
--- forcing RTS inactive
--- Miniterm on COM4  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
D: Partition nvs found, 20480 bytes
D: Read 31 keys from NVS
D: pin_ir set to -1
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to -1
D: pin_tft_dc set to -1
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_ch376_cs set to -1
D: pin_ch376_int set to -1
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 16
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is LOW, probably no PULL-UP
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is HIGH
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is LOW, probably no PULL-UP
D: GPIO39 is LOW, probably no PULL-UP
D: Create list with acceptable WiFi networks
D: Added Livebox-A510 to list of networks
D: Added Quiet batperson to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 1, number of configured networks: 2
D:  1 - Livebox-A510              Signal: -44 dBm, Encryption WPA_WPA2_PSK, Acceptable
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttprefix with parameter none
D: Command: mqttuser with parameter none
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 16
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.omroep.nl:80/radio1-bb-mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 0
D: Command: tonehf with parameter 0
D: Command: tonela with parameter 0
D: Command: tonelf with parameter 0
D: Command: volume with parameter 72
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Connect to WiFi
D: Connected to Livebox-A510
D: IP = 192.168.1.14
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MDNS responder started
D: Rotary encoder is disabled (-1/-1/-1)
D: STOP requested
D: New preset/file requested (6/0) from icecast.omroep.nl:80/radio1-bb-mp3
D: New station request
D: Connect to new host icecast.omroep.nl:80/radio1-bb-mp3
D: Connect to icecast.omroep.nl on port 80, extension /radio1-bb-mp3
D: Connected to server
D: Song stopped incorrectly!
D: REG   Contents
D: ---   -----
D:   0 -   808
D:   1 -    40
D:   2 -     0
D:   3 -  6000
D:   4 -     0
D:   5 -  AC44
D:   6 -     0
D:   7 -  1E06
D:   8 -     0
D: SPI semaphore not taken within 270 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D:   9 -     0
D:   A -     0
D: SPI semaphore not taken within 280 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D:   B -  F8F8
D:   C -     0
D: SPI semaphore not taken within 290 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsong
D:   D -     0
D:   E -     0
D: SPI semaphore not taken within 300 ticks by CPU 0, id hspec
D: Semaphore is claimed by stopsongD:   F -     0

D: Switch to HEADER
D: Headerline: Content-Type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-br:192
D: Headerline: ice-audio-info: samplerate=48000;channels=2;bitrate=192
D: Headerline: icy-br:192
D: Headerline: icy-genre:Talk
D: Headerline: icy-metadata:1
D: Headerline: icy-name:NPO Radio1
D: Headerline: icy-pub:0
D: Headerline: icy-url:http://www.radio1.nl
D: Headerline: Server: Icecast 2.4.0-kh15
D: Headerline: Cache-Control: no-cache, no-store
D: Headerline: Expires: Mon, 26 Jul 1997 05:00:00 GMT
D: Headerline: Connection: Close
D: Headerline: Access-Control-Allow-Origin: *
D: Headerline: Access-Control-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type, Icy-MetaData
D: l-Allow-Headers: Origin, Accept, X-Requested-With, Content-Type, Icy-MetaData seen.
D: Headerline: Access-Control-Allow-Methods: GET, OPTIONS, SOURCE, PUT, HEAD, STATS
D: Headerline: icy-metaint:16000
D: Switch to DATA, bitrate is 192, metaint is 16000
D: Duration mp3loop 83
D: Metadata block 80 bytes
D: Streamtitle found, 71 bytes
D: StreamTitle='NPO Radio 1 - Docs presenteert de NTR podcastprijs - NOS';
```

Edzelf, 2021-07-10:
Is there an amplifier for the speakers?
Try earphones frist.

robertpi, 2021-07-10:
Yes, they're active speakers, so built have a built in amplifier. I also tested with a set head phones and same result, no sound.

utw3v0, 2021-07-10:
Check your pin definitions again semaphore not taken .... i think there's a pin number wrong 

⁣TypeApp for Android herunterladen ​

robertpi, 2021-07-11:
Yes, semaphore not taken is an issue, I didn't spot that error at first. If you check the logs, you see:

```
 Semaphore is claimed by stopsong
```

And working back a bit:

```
Song stopped incorrectly!
```

So, that seems consistent, but didn't really help me understand which pin was mis-configured :)

Anyway, I went and double checked the pins and they seem consistent with what's provided in `ESP32-radio.pdf`. This led me to see that the wire diagram says:

```
 ESP32dev Signal Wired to LCD Wired to VS1053 SDCARD Wired to the rest
 -------- ------ -------------- ------------------- ------ ---------------
 GPIO32 - pin 1 XDCS - -
```

But I took my default config from here: https://github.com/Edzelf/ESP32-Radio/blob/master/tools/Esp32_radio_init.ino#L72 

Which sets DCS connection to 16:

```
preferences.putString ( "pin_vs_dcs",  "16                                     # GPIO Pin number for VS1053 DCS" ) ;
```

Once I changed it to 32 it started working, so all good now. Many thanks for the hints.

closed 2021-07-11 by robertpi

ceinem, 2022-06-25:
I had this issue as well! I think the pdf is wrong, no?