Edzelf/ESP32-Radio #475: Configuring the network

samand587, 2021-08-28:
I'm a beginner and I'm asking for help setting up the network. After filling the sketch, an ESP32 network is created. The web interface does not open at the address 192.168.4.1. The Esp32_radio_init file has corrected itself. How can I add data for my networks to the project?

samand587, 2021-08-28:
Заливаю Esp32_radio_init - успешно. Далее заливаю Esp32_radio_v05 и получаю в мониторе порта вот это: "create list with acceptable wi-fi networks 
end adding networks" Что можно посмотреть?

samand587, 2021-08-28:
I fill in Esp 32_radio_init-successfully. Next, I fill in Esp 32_radio_v05 and get this in the port monitor: "create list with acceptable wi-fi networks
and adding networks" What can I see?

samand587, 2021-08-28:
Here's what I have on the port monitor:
failed trying to setup ap with name esp32radio and password esp32radio
ip192.168.4.1
start server for commands
rotary encoder is enabled
stop requested
After that, "Starting" is always lit on the display

Edzelf, 2021-08-28:
Connect to network "EspRadio" with the same password.  Then goto 192.268.4.1 and got to the config page.  Enter said and credentials as described in the document.

samand587, 2021-08-28:
What password do you mean?

samand587, 2021-08-28:
I figured it out! Thanks!

closed 2021-08-28 by samand587