Edzelf/ESP32-Radio #476: Poor quality of stations in AAC and AAC +

hevet, 2021-08-29:
When I turn on a station in AAC or AAC + format, for a split second you hear good quality, and then the quality drops and you hear that it is in mono. This is not a problem with MP3. On the computer, these stations can be heard in stereo and in good quality. Where is the problem?

Edzelf, 2021-08-29:
Always give an example of the station that gives the problem.  Maybe someone can try to reproduce it.

hevet, 2021-08-29:
Exemplary two stations, at each station in these formats I have poor quality.

http://0n-disco.radionetz.de/0n-disco.aac
http://live.elka.pl:8000/miedziowe

Edzelf, 2021-09-20:
Sorry, All 4 stations sound good on my set-up.  Are you sure you have the VS1053 chip on the decoder board?

hevet, 2021-09-20:
Yes, I have tiles with VS1053B. I checked on two different ones. I found another radio project and there the AAC and AAC + stations sound very good. Unfortunately, the stations are of poor quality in your project. I also checked the stations in OGG and these also sound good. The problem is clearly with AAC and AAC + stations.

Rainer-G, 2021-09-20:
All stations play without drop-outs, but with a rather low Bitrate_ 64k http://0n-disco.radionetz.de/0n-disco.aac, 64k http://stream2.nadaje.com:8094/ 48k http://uk2.internet-radio.com:8024/stream,48k http://saw-de-hz-fal-stream06-cluster01.radiohost.de/saw-fitness_48?ref=radiosure-1a,  if you replace the 48 with 128, this Station http://saw-de-hz-fal-stream06-cluster01.radiohost.de/saw-fitness_48?ref=radiosure-1a will Play at a higher Bitrate! Winamp shows the same rate so it looks as if it is not a problem of the ESP32 Webradio ? Same Station with different bitrates ***@***.*** : would it be possible to show format and stereo/mono in the Web Interface when sending the Test command? RegardsRainer Gesendet von Mail für Windows Von: JacekGesendet: Sonntag, 29. August 2021 17:41An: Edzelf/ESP32-RadioCc: SubscribedBetreff: Re: [Edzelf/ESP32-Radio] Poor quality of stations in AAC and AAC + (#476) Exemplary two stations, at each station in these formats I have poor quality.http://0n-disco.radionetz.de/0n-disco.aachttp://live.elka.pl:8000/miedziowe