Edzelf/ESP32-Radio #484: webinterface states "sorry" after update

dkotte, 2021-11-10:
after click to "update" within webinterface, the TFT freezes (keeps black after reboot) and webinterface states only "sorry" after update.
Music is playing and within serial-output i can't find problems ...
Would be great, i got back control.

Edzelf, 2021-11-10:
Upload the files in the data directory to SPIIFS.
run -t uploaden  or something like that.

dkotte, 2021-11-12:
Thanks for the answer.
But for that I need platformio ..?
I have no experience with this tool.

I clicked "update" in the web interface and got a version for which files are missing? Is there an option to upload/update these files more simple? I am able to input "update" serial. But there i get "no new version"

Edzelf, 2021-11-12:
My fault.  The result of the update is indeed the V2 version of the radio. 
To correct this, just use the V1 version with Arduino IDE and upload it to your ESP32.
But may also try PlatformIO and switch to V2.

dkotte, 2021-11-12:
where do i find  "the V2.0 document".

Edzelf, 2021-11-12:
[https://github.com/Edzelf/ESP32Radio-V2/tree/main/doc](https://github.com/Edzelf/ESP32Radio-V2/tree/main/doc)