Edzelf/ESP32-Radio #485: Enhancement Request: Button on Search Page

craiglindley, 2021-11-24:
It would be really cool if there was a button on the Search page which would add the station you have selected and were listening to, to your presets. I know this can be done using cut and paste but a button would be really convenient.

EFWob, 2021-11-27:
Coincidentally I have just coded this for my spinoff project (it was a wife request to be able to store a randomly selected station from the RDBS to recall it later).

I opted against adding to the presets though but added a second list of favorites (valid are fav_01 to fav_100) to the preference settings and there is a command interface to add ("favorite=+") or remove ("favorite=-") the current station from/to the favorite list at a free spot. A bit more defaults can be found here: [https://github.com/EFWob/RetroRadio](https://github.com/EFWob/RetroRadio#favorite-concept).

The project is PlatformIO only, however may be it helps you to add it to your version self.

I also added a dropdown list to select a favorite and an Add/Remove button to the main control page.
[image]

craiglindley, 2021-11-28:
Interesting that I am not the only one seeking this kind of feature. I'll
look at your code and see if I can adapt it to what I want.

-- 
Craig Lindley / Heather Hubbard

New Recordings are here <http://craigandheather.net/spellbound.html>

Personal Website is here <http://craigandheather.net>

Please call the Rockrimmon house. Our cell phones don't work at home.
Rockrimmon House: (719) 426-9873
Craig's Cell: (719) 502-7925
Heather's Cell: (719) 571-0944

If you’re one in a million, there are now more than seven thousand people
exactly like you.