Edzelf/ESP32-Radio #489: search stations

mirek76, 2021-12-13:
Hi
Is it possible to search for stations only in a given country?
I would like to search for stations in e.g. Germany
What to change in file search_index.h? I would like to use country instead of genre

Edzelf, 2021-12-14:
I think you only have to change search.html in the data directory.  The API used is described at https://nl1.api.radio-browser.info.

mirek76, 2021-12-14:
Thank you for your answer.
Is that the part of search_html.h?

` // Get info from a radiobrowser.  Working servers are:
 // https://de1.api.radio-browser.info, https://fr1.api.radio-browser.info, https://nl1.api.radio-browser.info
 function listStats ( genre )
 {
  var theUrl = "https://nl1.api.radio-browser.info/json/stations/bytag/" +
               genre +
               "?hidebroken=true" ;`

What is the format for a given country?
Thx

Edzelf, 2021-12-14:
Read the docs.  Instead of "bytag", you have to use "bycountry".  Examples in the docs, like:
https://nl1.api.radio-browser.info/xml/stations/bycountry/austria

mirek76, 2021-12-15:
Thank you Edzelf
Works great.
`<td><button class="button" onclick="listStats('united states')">USA</button></td>
   <td><button class="button" onclick="listStats('germany')">Germany</button></td>
   <td><button class="button" onclick="listStats('poland')">Poland</button></td></tr>`

`// Get info from a radiobrowser.  Working servers are:
 // https://de1.api.radio-browser.info, https://fr1.api.radio-browser.info, https://nl1.api.radio-browser.info
 function listStats ( country )
 {
  var theUrl = "https://nl1.api.radio-browser.info/json/stations/bycountry/" +
               country +
               "?hidebroken=true" ;
  var xhr = new XMLHttpRequest() ;
  xhr.onreadystatechange = function()`

It would be nice to add a double search in search_html.h
According to genre and country.

closed 2022-01-08 by mirek76