Edzelf/ESP32-Radio #490: Sound only few sec, cut and restarts randomly

ChristianTardif, 2021-12-17:
Hello

First of all very good project.

I wired everything correctly and I'm able to have some sound from the VS1053B, but the sound is randomly stoping and restarting again.  The longer sound I got is few sec no more than 5 sec. 

Do you have an idea? 

I have no screen (dummytft)

any help ? 

Thank you in advance

Edzelf, 2021-12-17:
Do you mean that the radio switches to the next preset?
And does that happen after the sound output stops?
In that case it looks like the data stream has stopped for some reason.  You may try to use a different WiFi network.

ChristianTardif, 2021-12-17:
It is not always switching

And I’m very close to the router, which is a quality one and also have no problem with internet radio on my phone (passing from the router)

Did it may come frime settings that need to be optimized? What to optimize?

Thank you to help
Christian

Téléchargez Outlook pour iOS<https://aka.ms/o0ukef>

Edzelf, 2021-12-17:
First try to connect to a different wifi network, to rule out ESP hardware or software errors.

ChristianTardif, 2021-12-20:
Finally, it was the ESP32 board (Heltec wifi kit) that was the problem.  I had an other board (firebeetle esp32) used for an other perpose that I tryed and it work like a charme.

closed 2021-12-20 by ChristianTardif