Edzelf/ESP32-Radio #493: Where is the ino file?

BX2ABT, 2022-01-17:
Can't find any ino file to compile with the Arduino IDE. The documentation states there should be an ino file, but this repository only has a wrapper file. Read something about PlatformIO, but no instructions. Please advice.

Edzelf, 2022-01-17:
The file does not longer exist.  But you may rename the main.cpp file to ESP32-Radio.ino.
Compiling with Arduino IDE is more complicated, as you need to install a number of libraries.
So I advice to switch to PlatformIO.
These libraries are required:
```
	SD(esp32) @^1.0.5
	knolleary/PubSubClient@^2.8
	adafruit/Adafruit BusIO@^1.9.3
	adafruit/Adafruit GFX Library@^1.10.4
	adafruit/Adafruit ILI9341 @ ^1.5.9
	adafruit/Adafruit ST7735 and ST7789 Library @ ^1.7.5
	ESP Async WebServer@^1.2.3
	AsyncTCP@^1.1.1

```

Edzelf, 2022-01-17:
Correction.  The .ino file is still available for the V1 version of the Radio.
See [here](https://github.com/Edzelf/ESP32-Radio/blob/master/ESP32-Radio.ino).

BX2ABT, 2022-01-17:
Thanks for the quick reply. Installing libraries is not such a problem, learning a new platform more and for me - being just a small tinkerer - I'd rather stick with the familiar Arduino IDE. But I get why a professional would choose PlatformIO.

BTW, the link to the ino file is the wrapper I was talking about:
`// ESP32-Radio.ino
//
// This is just a wrapper for the Arduino IDE version of ESP32-Radio.
// ESP32-Radio is designed to be compiled in the PlatformIO IDE.
// main.cpp will be compiled and linked automatically by the Arduino IDE.
// main.cpp contains the setup() and the loop() functions required for the Arduino environment.
//
// 21-09-2020, ES
//`

BX2ABT, 2022-01-17:
Thanks for the quick reply. Installing libraries is not such a problem, learning a new platform more and for me - being just a small tinkerer - I'd rather stick with the familiar Arduino IDE. But I get why a professional would choose PlatformIO.

Bedankt! En tot ziens.

eduszesz, 2022-01-20:
Hi @Edzelf 

Could I use the method you mentioned of renaming the main.cpp file in version 2 to compile it using the Arduino IDE?

I have version 1 running with chopping audio for some high bitrate stations and I would like to try version 2, but I have no experience with PlatformIO.

Thanks!

Edzelf, 2022-01-20:
Yes.

eduszesz, 2022-01-20:
Hi @Edzelf 

Thanks for the quick replay!