Edzelf/ESP32-Radio #494: Choppy audio on one station

buntings, 2022-01-19:
Forgive me if there is another forum for asking for help or there are any diagnostics I should have performed - I am new to the ESP32 and also "github".

I have everything set up and working perfectly  ( just the ESP32 and VS1053 at the moment - no display or amplifier). Hardest part has been tracking down the HTTP streams for all the HTTPS stations! 

However all stations bar one are working perfectly – my favourite station of course!

Arctic Outpost Radio with the preset:-

preset_30 = 216.126.196.154:4047/stream/             # Arctic Outpost Radio

The audio is very choppy. If I program the ESP32 via the Arduino IDE with one of the many “internet radio”  files ( ie the .ino ) the station works perfectly.  Given all that 
a)	other stations are working perfectly
b)	that the same circuit programmed with a basic arduino file works perfectly on the same stream
c)	ESP is about 2 metres from the router ( I have also tried connecting to another Wifi hotspot ) - no wifi speed issues

I suspect this is something odd with that station and the code.
Searching git up I found a previous issue where someone reported the same but also found that pressing the “test” button repeatedly fixed it -  “Choppy playback at high bitrates  #138”

I have tried the same here and it works – each and everytime I select that station it starts choppy and remains so until I press the test button repeatedly. There is no connection between the number of times I press or the frequency. Sometimes it fixes it within 5 or 6 clicks, sometimes over 20 – but it always fixes it.

I don’t understand fully what the test data is that appears in the webpage however, when the station is choppy the text looks like this:-

Free memory is 221356, chunks in queue 121, stream 0, bitrate 117 kbps

The bitrate starts with  a large number then settles down to around 110-130kbs. The chunks is always around 120. However the “stream” is always “0”

Once I start clicking “test” the chunks increases (usually roughly doubles)  and the streams value goes from zero to several 1000....after which the audio is perfect.

This is repeatable each and every time.

Any ideas?
Thanks

Gareth

Edzelf, 2022-01-19:
Please try the V2 version of the radio.  I tested the station, no problem.
As you can see, the queue is filled with almost 500 chunks.
It may have something to do with your network.
[image]

buntings, 2022-01-19:
Thanks for the reply. But as I noted a basic ESP32 Arduino sketch works perfectly each and every time which I take to mean that this is not a network issue - otherwise the problem should be repeatable using other software.  I dont know what the test function does but it makes it work .... but it cannot fix an external network problem :-)

I will try V2 shortly if only to prove its not a network issue.

I have an LCD1602 sat here waiting to go in (it has to be a 1602 to fit in the recycled DAB cases I am using) - I didnt go for V2 simply as I understood that LCD wasnt supported, is that still correct?

Edzelf, 2022-01-19:
It may not be a network isssue, but it may be a network difference.  So if you have the opportunity to test with a different network, it might help to find a solution.
The V2 version allows a 1602 display.  But for testing a bare ESP32 will work (specify dummyftf and DEC_HELIX in config.h).  Use the webinterface to examine the queue.

buntings, 2022-01-19:
Thanks, fair comment. I have installed V2. I specified dummytft and DEC_HELIX in config.h (everything else commented out) however this does not work. The ESP connects to the stream but then this:-

Connected to MP3 host at 216.126.196.154 on port 4047
D: send GET command
D: Switch to HEADER
D: Headerline: Accept-Ranges:none
D: Headerline: Access-Control-Allow-Origin:*
D: Headerline: Cache-Control:no-cache,no-store,must-revalidate,max-age=0
D: Headerline: Connection:close
D: Headerline: icy-name:Arctic Outpost AM1270
D: Headerline: icy-genre:Jazz Blues Bigband Swing Classic Jazz
D: Headerline: icy-br:128
D: Headerline: icy-sr:44100
D: Headerline: icy-url:https://aor.am
D: Headerline: icy-pub:1
D: Headerline: content-type:audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-metaint:8192
D: Headerline: X-Clacks-Overhead:GNU Terry Pratchett
D: Switch to DATA, bitrate is 128, metaint is 8192
D: Starting I2S playtask..
D: Output to internal DAC
D: Playtask start song
D: helixInit called for audio/mpeg
D: Playtask stop song
D: Command: status (without parameter)
D: Sync found at 0x0143
D: Bitrate     is 0
D: Samprate    is 0
D: Channels    is 0
D: Bitpersamp  is 0
D: Outputsamps is 0
E (18538) I2S: clkmdiv is too large

D: MP3Decode error -2
D: helixInit called for audio/mpeg
D: Sync found at 0x002B
D: Bitrate     is 128000
D: Samprate    is 44100
D: Channels    is 1
D: Bitpersamp  is 16
D: Outputsamps is 1152
D: Command: status (without parameter)
D: Command: status (without parameter)

The last line then repeats continously. Similar on other stations.

I replaced DEC_HELIX with DEC_VS1053 and I get audio but it is again "choppy". The output looks like this:-

: New preset/file requested (30/0) from 216.126.196.154:4047/stream/
D: Connect to new host 216.126.196.154:4047/stream/
D: Connect to 216.126.196.154 on port 4047, extension /stream/
D: Song stopped correctly after 0 msec
D: Connected to MP3 host at 216.126.196.154 on port 4047
D: send GET command
D: Switch to HEADER
D: Headerline: Accept-Ranges:none
D: Headerline: Access-Control-Allow-Origin:*
D: Headerline: Cache-Control:no-cache,no-store,must-revalidate,max-age=0
D: Headerline: Connection:close
D: Headerline: icy-name:Arctic Outpost AM1270
D: Headerline: icy-genre:Jazz Blues Bigband Swing Classic Jazz
D: Headerline: icy-br:128
D: Headerline: icy-sr:44100
D: Headerline: icy-url:https://aor.am
D: Headerline: icy-pub:1
D: Headerline: content-type:audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-metaint:8192
D: Headerline: X-Clacks-Overhead:GNU Terry Pratchett
D: Switch to DATA, bitrate is 128, metaint is 8192
D: Command: status (without parameter)

The "chunks" is up and down between 60/70 and 200  - until I keep clicking the "test" button then stabilises at 495 and never changes. If I hit "stop" and "unstop" it resets, starts jumping - until I keep clicking "test" rapidly and it stabilises again.

buntings, 2022-01-19:
Some more results.

I have connected the ESP to another wifi source ( a powerline adaptor that is connected to the main router via the house mains).

This connects and works perfectly from the start.

So....

ESP32Radio(both versions) with Arctic Outpost Radio(AOR) - jumpy on the router wifi unless I press "test" a few times
But a basic arduino Internet Radio sketch - works fine on wifi from the router.

ESP32radio works perfectly when connected to a TPLINK powerline adaptor that is fed from the router ( and is actually a much weaker signal ).

Something on the router connection that ESP32radio doesnt like with AOR stream where the Arduino sketch is happy.

All very strange, why just AOR and why does "test" fix it?

I have noticed something else - if I left V1 alone it never fixed itself. However V2 if left untouched for 4 or 5 minutes seems to eventually fix itself (without pressing "test" ).

The router is a Linksys WRT1900ACS ( only a year or two old, CPU runs at 1.6ghz with 500mb RAM - running DD-WRT - not exactly a basic router ).

Not sure if that helps or just confuses things!

xP373Rx, 2022-04-22:
Had issues like that  too.   It was the esp32 sleep function . After WiFi.mode ( WIFI_STA ) ; put  
delay (500) ;                                      
WiFi.setSleep(false);

in your setup function

garbo5005, 2023-01-20:
I am also having an issue with choppy audio ( with v2) , I hope the sleep function will fix it. I just have the system on the breadboard and was please that it is working, but was let down when the audio quality wasn't good. I have no reason to complain since there was a lot of hardwork put into  designing this radio system

aly-fly, 2023-01-21:
Choppy audio can also be caused by bad wiring. SPI bus carries high frequency signals that are time sensitive and longer / messy wires will cause all sorts of problems. Bect length of total SPI wires is < 5 cm.