Edzelf/ESP32-Radio #495: "No data input" when playing stream

h1aji, 2022-01-27:
When I am trying to play Radio Svoboda, I see this in the log 

```
D: New preset/file requested (10/0) from n0d.radiojar.com:80/hcrb063nn3quv
D: Connect to new host n0d.radiojar.com:80/hcrb063nn3quv
D: Connect to n0d.radiojar.com on port 80, extension /hcrb063nn3quv
D: Connected to server
D: Switch to HEADER
D: Access-Control-Allow-Origin: *
D: No data input
D: No data input
D: Trying other station/file...
D: STOP requested
```

Works ok on PC and Raspberry Pi.

h1aji, 2022-01-27:
[image]

Edzelf, 2022-01-27:
A connect to this station will be redirected to [n0f.radiojar.com/hcrb063nn3quv?rj-ttl=5&rj-tok=AAABfpvIFa8AQZtzd7HsThOmMA](n0f.radiojar.com/hcrb063nn3quv?rj-ttl=5&rj-tok=AAABfpvIFa8AQZtzd7HsThOmMA).
At the moment, ESP-Radion does not support redirection.

h1aji, 2022-01-29:
Thank you Ed.

I assume these line are handling redirect
https://github.com/Edzelf/ESP32Radio-V2/blob/cefc93782583e00fe0768d88dc7570e73ee8ea01/src/main.cpp#L3333

```
        if ( lcml.startsWith ( "location: http://" ) ) // Redirection?
        {
          host = metaline.substring ( 17 ) ;           // Yes, get new URL
          hostreq = true ;                             // And request this one
        }
        if ( lcml.startsWith ("location: https://") )  // Redirection?
        {
          host = metaline.substring ( 18 ) ;           // Yes, get new URL
          hostreq = true ;                             // And request this one
        }
```

But in my case I cant even see **location:http** Its because high latency, over 300ms.
Is there a way to increase timeout before 

`D: No data input
`

Edzelf, 2022-01-29:
Did you really try the V2 version?  If the latency is a problem, you would have problems with all stations.
You should see:
```
D: Connect to new host n0d.radiojar.com:80/hcrb063nn3quv
D: Connect to n0d.radiojar.com on port 80, extension /hcrb063nn3quv
D: Connected to MP3 host at 51.178.22.38 on port 80
D: send GET command
D: Switch to HEADER
D: Headerline: Access-Control-Allow-Origin:
D: Headerline: Location: http://n0d.radiojar.com/hcrb063nn3quv?rj-ttl=5&rj-tok=AAABfqYb-B8AMBkk246kUcjyYQ
D: Headerline: Content-Length: 0
D: Headerline: Connection: close
D: MP3 host disconnected!
D: Connect to new host n0d.radiojar.com/hcrb063nn3quv?rj-ttl=5&rj-tok=AAABfqYb-B8AMBkk246kUcjyYQ
D: Connect to n0d.radiojar.com on port 80, extension /hcrb063nn3quv?rj-ttl=5&rj-tok=AAABfqYb-B8AMBkk246kUcjyYQ
D: Song stopped correctly after 0 msec
D: Connected to MP3 host at 51.178.22.38 on port 80
D: send GET command
D: Switch to HEADER
D: Headerline: content-type: audio/mpeg
D: audio/mpeg seen.
D: Headerline: icy-name: -
D: Headerline: icy-metaint: 16000
D: Switch to DATA, bitrate is 0, metaint is 16000
D: Command: status (without parameter)
D: Metadata block 80 bytes
D: Streamtitle found, 78 bytes
D: StreamTitle='Follow us on www.svoboda.org - Radio Svoboda live broadcasting.';

```

h1aji, 2022-02-06:
Confirmed. Works for V2.
I created a ticket to backport this feature for Esp-radio https://github.com/Edzelf/Esp-radio/issues/212

Thanks

closed 2022-02-06 by h1aji