Edzelf/ESP32-Radio #497: CH376S - no tracks

Jefke1, 2022-03-06:
Hi Ed,
I built this radio about 1.5 years ago and it works fine. Now I want to replace the SD card with USB. I copied the mp3 files from sd to usb, but when i boot, read from usb, 0 tracks on usb. what am I doing wrong ??

GPIO18 ??

D: Starting ESP32-radio running on CPU 1 at 240 MHz.  Version Mon, 28 Jun 2021 12:40:00 GMT.  Free memory 307764
D: Display type is NEXTION
D: Partition nvs found, 20480 bytes
D: Read 45 keys from NVS
D: pin_ir set to 35
D: pin_enc_clk set to -1
D: pin_enc_dt set to -1
D: pin_enc_sw set to -1
D: pin_tft_cs set to 15
D: pin_tft_dc set to 2
D: pin_tft_scl set to -1
D: pin_tft_sda set to -1
D: pin_tft_bl set to -1
D: pin_tft_blx set to -1
D: pin_sd_cs set to -1
D: pin_ch376_cs set to 21
D: pin_ch376_int set to 22
D: pin_vs_cs set to 5
D: pin_vs_dcs set to 32
D: pin_vs_dreq set to 4
D: pin_shutdown set to -1
D: pin_shutdownx set to -1
D: pin_spi_sck set to 18
D: pin_spi_miso set to 19
D: pin_spi_mosi set to 23
D: GPIO0 is HIGH
D: GPIO2 is HIGH
D: GPIO4 is HIGH
D: GPIO5 is HIGH
D: GPIO12 is HIGH
D: GPIO13 is HIGH
D: GPIO14 is HIGH
D: GPIO15 is HIGH
D: GPIO16 is HIGH
D: GPIO17 is HIGH
D: GPIO18 is LOW, probably no PULL-UP
D: GPIO19 is HIGH
D: GPIO21 is HIGH
D: GPIO22 is HIGH
D: GPIO23 is HIGH
D: GPIO25 is HIGH
D: GPIO26 is HIGH
D: GPIO27 is HIGH
D: GPIO32 is HIGH
D: GPIO33 is HIGH
D: GPIO34 is LOW, probably no PULL-UP
D: GPIO35 is HIGH
D: GPIO39 is LOW, probably no PULL-UP
D: gpio_00 will execute uppreset = 1
D: gpio_12 will execute upvolume = 1
D: gpio_13 will execute downvolume = 1
D: Enable pin 35 for IR
D: Start display
D: Init Nextion, pins 16,17
D: Nextion command cls BLACK
D: Nextion command page 0
D: Create list with acceptable WiFi networks
D: Added Orange-486AE to list of networks
D: Added SSID2 to list of networks
D: End adding networks
D: Scan Networks
D: Scan completed
D: Number of available networks: 7
D:  1 - Orange-486AE              Signal: -62 dBm, Encryption WPA2_PSK, Acceptable
D:  2 - Orange-1d6e6              Signal: -70 dBm, Encryption WPA2_PSK, 
D:  3 - Orange-82fa0              Signal: -77 dBm, Encryption WPA2_PSK, 
D:  4 - Guest-Orange-82fa0        Signal: -77 dBm, Encryption WPA2_PSK, 
D:  5 - TelenetWiFree             Signal: -89 dBm, Encryption ????, 
D:  6 - telenet-2122642           Signal: -90 dBm, Encryption WPA2_PSK, 
D:  7 - telenet-4E281C7           Signal: -92 dBm, Encryption WPA2_PSK, 
D: End of list
D: Command: clk_dst with parameter 1
D: Command: clk_offset with parameter 1
D: Command: clk_server with parameter pool.ntp.org
D: Command: fs with parameter USB
D: Command: gpio_00 with parameter uppreset = 1
D: Command: gpio_12 with parameter upvolume = 1
D: Command: gpio_13 with parameter downvolume = 1
D: Command: ir_40bf with parameter upvolume = 2
D: Command: ir_c03f with parameter downvolume = 2
D: Command: mqttbroker with parameter none
D: Command: mqttpasswd with parameter *******
D: Command: mqttport with parameter 1883
D: Command: mqttprefix with parameter none
D: Command: mqttuser with parameter none
D: Command: pin_ch376_cs with parameter 21
D: Command: pin_ch376_int with parameter 22
D: Command: pin_enc_clk with parameter -1
D: Command: pin_enc_dt with parameter -1
D: Command: pin_enc_sw with parameter -1
D: Command: pin_ir with parameter 35
D: Command: pin_tft_cs with parameter 15
D: Command: pin_tft_dc with parameter 2
D: Command: pin_vs_cs with parameter 5
D: Command: pin_vs_dcs with parameter 32
D: Command: pin_vs_dreq with parameter 4
D: Command: preset with parameter 6
D: Command: preset_00 with parameter 109.206.96.34:8100
D: Command: preset_01 with parameter airspectrum.cdnstream1.com:8114/1648_128
D: Command: preset_02 with parameter us2.internet-radio.com:8050
D: Command: preset_03 with parameter airspectrum.cdnstream1.com:8000/1261_192
D: Command: preset_04 with parameter airspectrum.cdnstream1.com:8008/1604_128
D: Command: preset_05 with parameter us1.internet-radio.com:8105
D: Command: preset_06 with parameter icecast.vrtcdn.be/stubru_tijdloze-high.mp3
D: Command: preset_07 with parameter 205.164.62.15:10032
D: Command: preset_08 with parameter skonto.ls.lv:8002/mp3
D: Command: preset_09 with parameter 94.23.66.155:8106
D: Command: preset_10 with parameter ihr/IHR_IEDM
D: Command: preset_11 with parameter ihr/IHR_TRAN
D: Command: toneha with parameter 2
D: Command: tonehf with parameter 5
D: Command: tonela with parameter 7
D: Command: tonelf with parameter 11
D: Command: volume with parameter 66
D: Slow SPI, Testing VS1053 read/write registers...
D: Fast SPI, Testing VS1053 read/write registers again...
D: endFillByte is 0
D: Start CH376
D: Flash drive ready
D: Locate mp3 files on USB drive, may take a while...
D: 0 tracks on USB
D: Connect to WiFi
D: Connected to Orange-486AE
D: IP = 192.168.0.108
D: Nextion command page 1
D: Start server for commands
D: Network found. Starting mqtt and OTA
D: MDNS responder started
D: Rotary encoder is disabled (-1/-1/-1)
D: Sync TOD
D: Sync TOD, new value is 16:12:25
D: STOP requested
D: New preset/file requested (6/0) from icecast.vrtcdn.be/stubru_tijdloze-high.mp3
D: New station request
D: Connect to new host icecast.vrtcdn.be/stubru_tijdloze-high.mp3
D: Connect to icecast.vrtcdn.be on port 80, extension /stubru_tijdloze-high.mp3
D: Song stopped correctly after 0 msec
D: Connected to server
D: Nextion t0.txt="ESP32-Radio"
D: Nextion t1.txt=""
D: Nextion t2.txt="6 - Studio Brussel De Tijdloze"
D: Switch to HEADER

Edzelf, 2022-03-06:
Filenames on USB are limited to the 8.3 format.  Try a file with the name ABCDEFGH.MP3.

Jefke1, 2022-03-08:
Thanks Ed,
this was the problem, usb stick formatted with quick format and even disk management (fat32) in win10 this was not the right choice. Formatted with an external program "partition manager". Everything works fine.

closed 2022-03-10 by Jefke1